openhab/org.openhab.binding.zigbee #289: ParseError on startup

peternitro, 2018-11-01:
#273 ## Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |       HUSBZB-1                                                                         |
| openHAB version  | 2.3.0    |
| Hardware         |    Chromebox                                                                            |
| Memory           |     2GB                                                                           |
| Java version     |     openjdk version "1.8.0_181"
OpenJDK Runtime Environment (Zulu 8.31.0.1-linux64) (build 1.8.0_181-b02)
OpenJDK 64-Bit Server VM (Zulu 8.31.0.1-linux64) (build 25.181-b02, mixed mode)                                                                           |
| Devices          | Provide all device types and their addresses                                   |

### Logs

2018-11-01 10:08:45.034 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP dongle initialize done: Initialised false
2018-11-01 10:08:45.035 [DEBUG] [nal.ZigBeeNetworkStateSerializerImpl] - Loading ZigBee network state: Start.
2018-11-01 10:08:45.110 [DEBUG] [nal.ZigBeeNetworkStateSerializerImpl] - Error loading ZigBee state 
com.thoughtworks.xstream.converters.ConversionException:  : ParseError at [row,col]:[209,61]
Message: Character reference "&# :  : ParseError at [row,col]:[209,61]
Message: Character reference "&#
---- Debugging information ----
message             :  : ParseError at [row,col]:[209,61]
Message: Character reference "&#
cause-exception     : com.thoughtworks.xstream.io.StreamException
cause-message       :  : ParseError at [row,col]:[209,61]
Message: Character reference "&#
class               : java.lang.String
required-type       : java.lang.String
converter-type      : com.thoughtworks.xstream.converters.SingleValueConverterWrapper
wrapped-converter   : com.thoughtworks.xstream.converters.basic.StringConverter
path                : /list/ZigBeeNode/endpoints/ZigBeeEndpoint/inputClusters/ZclCluster/attributes/entry[8]/ZclAttribute/lastValue
line number         : 209
class[1]            : com.zsmartsystems.zigbee.zcl.ZclAttribute
converter-type[1]   : com.thoughtworks.xstream.converters.reflection.ReflectionConverter
class[2]            : java.util.HashMap
converter-type[2]   : com.thoughtworks.xstream.converters.collections.MapConverter
class[3]            : com.zsmartsystems.zigbee.dao.ZclClusterDao
class[4]            : java.util.ArrayList
converter-type[3]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
class[5]            : com.zsmartsystems.zigbee.dao.ZigBeeEndpointDao
class[6]            : com.zsmartsystems.zigbee.dao.ZigBeeNodeDao
version             : 1.4.7
-------------------------------
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:79) ~[?:?]
	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:474) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:406) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:257) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.MapConverter.putCurrentEntryIntoMap(MapConverter.java:110) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.MapConverter.populateMap(MapConverter.java:98) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.MapConverter.populateMap(MapConverter.java:92) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.MapConverter.unmarshal(MapConverter.java:87) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshallField(AbstractReflectionConverter.java:474) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.doUnmarshal(AbstractReflectionConverter.java:406) ~[?:?]
	at com.thoughtworks.xstream.converters.reflection.AbstractReflectionConverter.unmarshal(AbstractReflectionConverter.java:257) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
	at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50) ~[?:?]
	at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71) ~[?:?]
[... 50 lines cut ...]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:?]
	at java.lang.Thread.run(Thread.java:748) [?:?]
Caused by: com.thoughtworks.xstream.io.StreamException:  : ParseError at [row,col]:[209,61]
Message: Character reference "&#
	at com.thoughtworks.xstream.io.xml.StaxReader.pullNextEvent(StaxReader.java:73) ~[?:?]
	at com.thoughtworks.xstream.io.xml.AbstractPullReader.readRealEvent(AbstractPullReader.java:148) ~[?:?]
	at com.thoughtworks.xstream.io.xml.AbstractPullReader.readEvent(AbstractPullReader.java:135) ~[?:?]
	at com.thoughtworks.xstream.io.xml.AbstractPullReader.getValue(AbstractPullReader.java:197) ~[?:?]
	at com.thoughtworks.xstream.io.ReaderWrapper.getValue(ReaderWrapper.java:48) ~[?:?]
	at com.thoughtworks.xstream.converters.SingleValueConverterWrapper.unmarshal(SingleValueConverterWrapper.java:49) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
	... 79 more
Caused by: javax.xml.stream.XMLStreamException: ParseError at [row,col]:[209,61]
Message: Character reference "&#
	at com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl.next(XMLStreamReaderImpl.java:604) ~[?:?]
	at com.thoughtworks.xstream.io.xml.StaxReader.pullNextEvent(StaxReader.java:58) ~[?:?]
	at com.thoughtworks.xstream.io.xml.AbstractPullReader.readRealEvent(AbstractPullReader.java:148) ~[?:?]
	at com.thoughtworks.xstream.io.xml.AbstractPullReader.readEvent(AbstractPullReader.java:135) ~[?:?]
	at com.thoughtworks.xstream.io.xml.AbstractPullReader.getValue(AbstractPullReader.java:197) ~[?:?]
	at com.thoughtworks.xstream.io.ReaderWrapper.getValue(ReaderWrapper.java:48) ~[?:?]
	at com.thoughtworks.xstream.converters.SingleValueConverterWrapper.unmarshal(SingleValueConverterWrapper.java:49) ~[?:?]
	at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
	... 79 more

peternitro, 2018-11-01:
Please let me know if I can provide any other information

hsudbrock, 2018-11-01:
Hi @peternitro; thanks for the report; it would be helpful if you could provide the XML file that is being parsed (or at least the excerpt where the parse error occurs, i.e., the part around `[row,col]:[209,61]`).

This is the network state XML file, which is located in the `zigbee` subfolder of the `userdata` folder of your openHAB installation.

cdjackson, 2018-11-01:
Firstly, please upgrade to the latest version. 2.3 is very old, and the latest version of the libraries have some fixes to try and resolve this sort of issue. It is caused by devices sending invalid strings, and it would be really helpful if you can provide a debug logfile of the device initialisation so we can see what is corrupt.

There have been a number of fixes to resolve this though, although I don’t recall if they are included in the library that is in the current 2.4 version of OH or not, but let’s try this for starters.

peternitro, 2018-11-01:
will do - isn't 2.3.0 the latest stable release @cdjackson ? which repo do you suggest I use? `testing` or `unstable` ?

cdjackson, 2018-11-01:
Yes - 2.3 is the last stable, but this is still 4 or 5 months old now and there have been a lot of changes.  I think testing and unstable will be the same so either will be fine.

However, I’m not too confident this will fix the problem, and you probably need to do a further update to get the latest libraries.  Please take a look a the following thread on the forum -:
https://community.openhab.org/t/zigbee-and-z-wave-manual-install-script/50270

I think that this will resolve your issue, however if that doesn’t fix the problem, then let me know and I can take a look at a debug log (see the binding docs for what logging to enable). The XML will not really be very useful as we need to see the raw data received from the device.

cdjackson, 2018-11-16:
As stated above, the XML is not really of any use. Can you please provide the logfile of the device being initialised so I can see exactly what is causing the problem.

Can you also confirm that the issue still exists in 2.4M5 or not?

peternitro, 2018-11-16:
I was using:

2.4.0-SNAPSHOT
Build #1408

just upgraded to:

2.4.0-SNAPSHOT
Build #1427  

and it's still an issue. 

debug log on startup: https://www.amazon.com/clouddrive/share/Y3MOpWcvv8QvVs91A4lbF5y4L75wXNvrhtwJchJrEBT

cdjackson, 2018-11-16:
Unfortunately the log doesn’t contain any transactions with the device, and doesn’t show the downloading of the date code that is causing the problem.

My guess though is that this issue is actually already resolved in the latest libraries. I suspect that there is a null terminator after the date, and this was accounted for a while ago (https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/414).

I would suggest to use the latest libraries - see the link I posted above for a script to do that.

peternitro, 2018-11-16:
The latest libraries are not included in the latest snapshots?

cdjackson, 2018-11-16:
Please see the link I posted a while ago about how to update to the latest libraries.

peternitro, 2018-11-16:
So that's a no? I'm literally using the snapshot built today. 

[root@nexus zigbee]# rpm -qa | grep -i openhab
openhab2-2.4.0.20181116153055-1.noarch

Why do the logs not include the relevant information? Is there another logger I need to drop to DEBUG? I've captured everything from startup until well after the error occurred?

cdjackson, 2018-11-16:
Yes - as I said a while back, you will need to update to get the latest libraries. I said this a couple of weeks back, but I'll post it again for clarity -:

> However, I’m not too confident this will fix the problem, and you probably need to do a further update to get the latest libraries.  Please take a look a the following thread on the forum -:
https://community.openhab.org/t/zigbee-and-z-wave-manual-install-script/50270

> Why do the logs not include the relevant information?

I don't know - there was not communication with the device, but I can't say why.

> I've captured everything from startup until well after the error occurred?

The error is irrelevant. The error is as a result of data that was written when the binding was previously closed. What is needed is to see the data when the binding communicates with the device again - this is not linked to the error at all and will be related to the device and how/when it is used etc.

cdjackson, 2018-12-23:
@peternitro any update on this? Have you tried the latest libraries which I think will resolve this? If not, can you provide the requested log?

peternitro, 2018-12-24:
@cdjackson I believe it's fixed, haven't seen it in a while since upgrading to latest (2.4.0.201812141755) and a few restarts.

I have been seeing this for a while:

`2018-12-23 16:41:57.768 [WARN ] [ve.internal.protocol.ZWaveController] - NODE 5: Restore from config: Error. Data invalid, ignoring config.` 

not sure where that came from but I'll take deeper look soon and open another issue if needed.

Thanks.

cdjackson, 2018-12-24:
Thanks. I suggest we close this.

Note that your other issue is not related to ZigBee at all - the warning is logged through the ZWave binding.

closed 2018-12-24 by cdjackson

peternitro, 2018-12-24:
I noticed... but don't you maintain that binding too? :) https://github.com/openhab/org.openhab.binding.zwave/issues

cdjackson, 2018-12-24:
Yes, I do, but as this is a ZigBee issue list I just wanted to point out that it’s not related to ZigBee...