openhab/org.openhab.binding.zigbee #290: Updated to 2.4.0-M5 from 2.3.0 - no longer able to access UART over USB

nefilim, 2018-11-01:
This happened after updating as requested in #289 
---

### Outline

Replaced the contents of the existing yum repo with the Testing repo and `yum update openhab2`. Restarted OpenHab2 and rebooted. 

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |    HUSBZB-1                                                                            |
| openHAB version  | 2.4.0-M5     |
| Hardware         |                                                                                |
| Memory           |                                                                                |
| Java version     |                                                                                |
| Devices          | Provide all device types and their addresses                                   |

### Logs

```

nefilim, 2018-11-01:
@cdjackson - not sure what else to check? It was working, upgraded with yum and now it's not working, would it suggest something with the upgrade process is awry?

I've updated the original post a few times, checked everything I could think of. Also, selinux has been disabled since before I started playing with OpenHab

nefilim, 2018-11-01:
```

cdjackson, 2018-11-01:
I’m not sure - serial port issues are a common problem unfortunately. Normally it’s related to security, but I’m not really an expert with the low level stuff. I don’t think it’s likely to be related to the binding and I don’t know if the update removed some serial options or security settings maybe?

cdjackson, 2018-11-01:
I would suggest to discuss the serial port access issues on the community forum as it's not a binding bug.

closed 2018-11-01 by cdjackson