openhab/org.openhab.binding.zigbee #301: Channel Consolidation: Unintentionally Disabled, Candidate for Removal?

MichaelRiessTelekom, 2018-11-15:
There seems to be a bug in the channel converter factory: The channels Map has String keys, but is later checked with containsKey() for ChannelTypeUID instances, which never matches. This effectively disables channel consolidation.

Question: Shouldn't this be removed rather than fixed? If any consolidation is necessary, it might be more appropriately done by clients of the binding rather than in the binding.

@cdjackson @kaikreuzer @hsudbrock

https://github.com/openhab/org.openhab.binding.zigbee/blob/db97db8b0ccfbee083aba0a14c88ab610d05064f/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/converter/ZigBeeChannelConverterFactory.java#L109

MichaelRiessTelekom, 2018-11-15:
Sorry for the spam, should have discussed this one more time with my colleagues ;-)

MichaelRiessTelekom, 2018-11-15:
(See issue #300)

closed 2018-11-15 by MichaelRiessTelekom

cdjackson, 2018-11-15:
> Question: Shouldn't this be removed rather than fixed? If any consolidation is necessary, it might be more appropriately done by clients of the binding rather than in the binding.

This was added following previous discussions that concluded that we should not add (for example) a color channel, dimmer channel, and switch channel. Due to inheritance, the color channel already supports the dimmer and onoff types, so adding the extra channels is not necessary.