openhab/org.openhab.binding.zigbee #305: [enhancement] Make polling/reporting interval configurable

weakfl, 2018-11-19:
Add an option to configure the polling/reporting interval per device.

As discussed in #49, this would allow to work around issues with devices which are not properly reporting state changes, resulting in a disparity of the device state in OH and the actual device state.

Two such devices are the Busch-Jaeger 6711 relay insert and 6715 dimmer insert. Both devices only report their state at the default interval of 2 hours but not if a state change is triggered by the wall-mounted switch.

cdjackson, 2018-12-23:
I have mostly coded this, but will wait for #342 to be merged before completing as it will make it easier to reconfigure the reporting once the device initialisation is split from the converter initialisation.

weakfl, 2018-12-23:
Awesome, thanks Chris!

closed 2019-02-08 by hsudbrock

weakfl, 2021-08-28:
@cdjackson It doesn't look like it's still possible to configure this in OH3 or do I miss something? Can I set this through the console or any other way?

cdjackson, 2021-08-28:
I don't think anything has changed in the binding - this still should be available. You can't change polling via any binding console as this is just a binding function so it configured through the normal OH wide configuration system.

Reporting you can potentially change, but I would suggest to avoid doing it manually as it might screw things up.

I don't have a binding running at the moment so can't test this, but certainly from the binding perspective it hasn't changed in OH3

weakfl, 2021-08-28:
I'm only interested in changing the polling interval. With OH2 it was possible to configure polling through HABmin, but it looks like the new UI doesn't support this.

cdjackson, 2021-08-28:
I thought OH3 did support channel configuration, but maybe it's not there. Unfortunately I won't have my system back up for a few weeks I guess, but in any case this hasn't changed in the binding so maybe it's worth discussing in the UI issues?

weakfl, 2021-08-28:
No, you're right, it's available in the channel configuration. Thanks for pointing me in the right direction, I just didn't notice it.

PR #344 (2019-02-08): Add ReportingConfiguration handler
This adds reporting and polling configuration handler. It is incomplete as it requires #342 and possibly a subsequent PR to refactor the LevelControlHandler before it can be completed.

Closes #305