openhab/org.openhab.binding.zigbee #307: USB discovery does not seem to honour persistent device names

puzzle-star, 2018-11-24:
Since "Support discovery of USB dongles #178" USB devices are automatically discovered and some discovery participants allow for auto detection of some devices (i.e. ZigBee Telegesis dongles).

Notwithstanding, USB discovery does not seem to honour persistent device names, and discovered device names may not be consistent between reboots. This is a follow-up on "https://github.com/openhab/org.openhab.binding.zigbee/issues/178#issuecomment-441360319".

@hsudbrock, could you please have a look at this and give your thoughts on whether this would need a fix?

I have seen that the discovery of the USB devices, and naming the different dongles via udev, do not seem to interface very well (at least for me).

The discovery participant is returning the /dev/ttyUSBx (for instance) and using the discovered device name as representationProperty. But this name might change between reboots, so normally, when using different dongles, we symlink to it (i.e. /dev/telegesis-1 -> /dev/ttyUSB0) via udev rules.

With the USB discovery, if for some reason the devname changes between reboots, the dongle becomes unusable. On the other side, if after discovering it I manually change the serial port of the device to the persistent name symlink, it will keep on appearing as new device in the inbox (yes, I can hide it, but it is annoying and maybe misleading for current users).

Changing the representationProperty to the serial number (or maybe vip_pid_serial) might prevent it from appearing again in the inbox (not sure), but would still require to manually change the serial port to the symlink.

Would it make sense to change (this is ESH, not OH):

```
Path devicePath = Paths.get(devDirectory).resolve(serialPortName);`
```

To some thing like (**this is pseudocode**, just to check if this would make sense to you):

```
Path devicePath = resolveDevicePath(devDirectory, serialPortName);
...
Path resolveDevicePath(String devDirectory, String serialPortName) {
    Path devicePath = Paths.get(devDirectory).resolve(serialPortName);
    File devDir = new File(devDirectory);
    for (File dev : devDir.listFiles()) {
       Path devPath = dev.toPath();
       if (Files.isSymbolicLink(devPath) && Files.isSameFile(devPath, devicePath)) return devPath;
    }
    return devicePath;
}
```
I know it is not the most efficient, as it traverses the the full dev directory looking for symlinks to the discovered path, but allows to honour the different dongle naming persistence.

Pedro

hsudbrock, 2018-11-26:
Hi Pedro, thanks for bringing this up! I agree that using the serial port as representation property is not a good idea, because this property does not represent device (for the reason you describe). So, I don't have anything against the representation property being changed to the serial number. (Side note: As far as I recall, some dongles don't provide their serial number via USB, this needs to be taken into account when changing this.)

I have not done anything about it so far, because ESH automatically fixes the configuration when the serial port changes: The serial number of the dongle is encoded in the discovery result's thing UID; in consequence, a dongle will always have the same thing UID in the discovery, regardless of the serial port; when ESH detects a discovery result with the same thing UID as an existing thing, ESH triggers a configuration update of the thing using the configuration provided in the discovery result; as a result, the serial port property is adjusted in the thing's configuration.

Due to this config update mechanism of ESH, I don't think that a scan for symlinks that you propose should be necessary.

> With the USB discovery, if for some reason the devname changes between reboots, the dongle becomes unusable.

I wonder why the ESH mechanism I sketched above does not work. Could it be due to #233, which Chris fixed in #273, and that is not yet included in your installation?

puzzle-star, 2018-11-26:
Hi @hsudbrock ,

Checked and working OK, but it has been a bit more painful than expected.

I have forced a change in the USB name, and it has been correctly detected by the participant. The configuration of the dongle has been updated accordingly...

But the new ttyUSB1 was not in my "-Dgnu.io.rxtx.SerialPorts=...", so it has not worked at all, with the side effect of losing my network state (@cdjackson, is it possible that if the dongle fails to initialize for some reason, the network state is saved with no node, forcing a full rediscovery on next succesful boot?)

So unless "gnu.io.rxtx.SerialPorts" is preloaded with all possible options for device names, it will not work as expected.

Henning, would it then make sense to look for the symlinks as an enhancement?

Also, I am surprised about this: looking whether the list of ports could be automatically set-up, I had a look at the nrjavaserial code, and to my surprise autodetection for Linux (and other OSes) is in place, so I just did a quick test, and:

```
~ #> java -cp /usr/share/openhab2/runtime/system/com/neuronrobotics/nrjavaserial/3.14.0/nrjavaserial-3.14.0.jar:. TestSerialPort
Available port: /dev/ttyAMA0
Available port: /dev/ttyUSB0

~ #> java -Dgnu.io.rxtx.SerialPorts=/dev/telegesis-1 -cp /usr/share/openhab2/runtime/system/com/neuronrobotics/nrjavaserial/3.14.0/nrjavaserial-3.14.0.jar:. TestSerialPort
Available port: /dev/telegesis-1
Available port: /dev/ttyAMA0
Available port: /dev/ttyUSB0
```
Is there somewhere in OH / ESH where this is overriden? It looks like the defined list should not be necessary, but it is not working without it.

puzzle-star, 2019-01-24:
Hi @johanneskropf,

Did you add /dev/ttyUSB0 (and the equivalent for the other dongles), to your "gnu.io.rxtx.SerialPorts"?

The "Serial Error: Port /dev/ttyUSB0 does not exist." normally happens because of this.

johanneskropf, 2019-01-24:
@puzzle-star this shouldn't be necessary as there are Symlinks for each of the three Dongles. But yes I have tried this too, but if I add /dev/ttyUSB0 it doesn't work at all. Too my understanding the Ember Stick is not gonna be on the same tty every boot anyway. That's why there are Symlinks in the first place. Those are correct too. And I am in the right groups with the right permissions and the right users. It works with the zwave and Enocean Stick. Although the Zwave Binding is the only one which respects the Serial port Symlink through a reboot. I have to reenter the Symlink for the Enocean after every reboot but than it works, but the discovery progress does something strange too, see the link in my previous post. The problem with the Zigbee Binding I have is that once I started the discovery the Stick goes offline and there is no way to bring it back but a restart.

puzzle-star, 2019-01-24:
Hi @johanneskropf,

It was not necessary before the USB discovery automatically changed the serial port name when discovering a recognized dongle.

Now, as the USB discovery is changing the name of the serial port to the non-symlink one, it has to be listed in "gnu.io.rxtx.SerialPorts" or it won't work.

@hsudbrock, if the representation property is the serial port, what happens if the serial port name is changed after a reboot, or by reconnecting the dongle? This is why, at least the Linux guys, like myself :), usually give a persistent name to the serial ports, and hence the symlinks. This used to be also the recommended approach when using more than one dongle or in situations where the serial port could change.

Do you think it would be worth either changing the representation property to the serial number of the device, or otherwise allowing the USB discovery to check for the symlinks?

If not I think we will find these issues recurrently, and will be hard to fix if the representation property is the serial port name.

hsudbrock, 2019-01-24:
Hi @puzzle-star; 

you ask:
> Do you think it would be worth either changing the representation property to the serial number of the device, or otherwise allowing the USB discovery to check for the symlinks?

I have nothing against this, see my answer when you opened this issue :)
> Hi Pedro, thanks for bringing this up! I agree that using the serial port as representation property is not a good idea, because this property does not represent device (for the reason you [...]

puzzle-star, 2019-01-24:
Hi @hsudbrock,

I recall, but this is on ESH, and I do not have a full dev environment to make changes on ESH.

I can give it a try to change it myself, but is this the right moment or better wait till ESH is integrated back onto OH?

Also my "IDE" and "compiler" tools at the moment are "vi", "javac" and "jar". Struggling with maven since last week to create a p2 bundle... Is command line building of ESH possible, or do I need to setup again Eclipse IDE?

johanneskropf, 2019-01-24:
So I would have to list ttyUSB1 etc too in Dgnu.io in case the Zigbee dongle is on one of these? After a reboot it was now on  ttyUSB0 and it stays online when I go into discovery. But why does it reset the Serial Port of the Enocean binding when going into discovery? The Enocean Stick is on ttyUSB1. Right now I can't discover devices but that is probably a different problem.

cdjackson, 2019-12-22:
Does anyone have a concrete solution to this problem. If not, I would propose to remove the USB discovery.

hsudbrock, 2019-12-23:
I have ideas for a solution, but, unfortunately, I do not get to experiment how well they work in the foreseeable future :/

Hence, @cdjackson, I agree that maybe it's best to remove the USB discovery for now; I can still create a PR for re-adding a USB discovery which works with persistent device names at some other time, when I have more time for working on that.

puzzle-star, 2019-12-23:
> I have ideas for a solution, but, unfortunately, I do not get to experiment how well they work in the foreseeable future :/

Hi @hsudbrock , at the beginning of the issue (in one of the comments, I posted some code that fixed it. 

https://github.com/openhab/org.openhab.binding.zigbee/issues/307#issue-383982884

I tried in my setup and it worked, but it was several months ago and I have no longer a development 
environment setup to create a PR (and I could not before the end of Christmas)...

Pedro

cdjackson, 2022-06-25:
USB discovery was removed so closing this

closed 2022-06-25 by cdjackson