openhab/org.openhab.binding.zigbee #309: New feature: Channel converter for the IAS tamper attribute

TomTravaglino, 2018-11-26:
We want to introduce a channel which reports that a device has bee tampered.

For the implementation we would
- add a new channel type with the id "ias_tampering_switch"
- create a channel converter which extends the ZigBeeConverterIas

cdjackson, 2018-11-26:
Can I suggest the channel name is just “has_tamper” - I don’t think it needs the term “switch” on the end and it’s not used on other IAS notifications.

hsudbrock, 2018-11-27:
@cdjackson, you probably mean `ias_tamper` (not `has_tamper`), just to avoid misunderstandings :)

cdjackson, 2018-11-27:
Thanks Henning :) - yes - you are correct (I'll blame my Macs email autocorrect!).

TomTravaglino, 2018-11-27:
OK. I'll change the channel type id as suggested.