openhab/org.openhab.binding.zigbee #313: Color tempterature converter uses cold for high values, warm for low values (but should be vice versa)

hsudbrock, 2018-11-30:
In ESH, for the color temperature channel '0% should be the coldest setting (highest Kelvin value)' (see (cf. https://www.eclipse.org/smarthome/documentation/development/bindings/thing-definition.html)).

In the ZigBee binding, this is reversed (0% is the lowest Kelvin value, 100% is the highest Kelvin value).

Maybe this was introduced by my PR https://github.com/openhab/org.openhab.binding.zigbee/pull/287 (for some bulbs - as the goal of that PR was to align how the converter behaves for bulbs that provide min/max Mired values, and for bulbs that do not provide min/max Mired values).

I plan to prepare a fix for this soon. If I overlook a reason why this should be the other way round for this binding, I would be happy about a ping, otherwise I will prepare a (probably small) PR to change this.

cdjackson, 2018-11-30:
I see there are now system channels available for this, so I guess it would be worth changing the channel name to use the system channel at the same time?

hsudbrock, 2018-11-30:
Sounds good.

PR #329 (2018-12-06): Fix color temperature from/to percent conversion in color temperature converter
Fixes #313 by adapting the conversion from Kelvin to percent and vice versa, such that warm color temperatures correspond to high percent values, and cold color temperatures correspond to low percent values.

Also replaces the binding-specific channel type for color temperature with the system channel type, and adds endpoint ids to some log messages.
commit b1b8de59 (2018-12-06): Fix color temperature from/to percent conversion in color temperature converter (#329)

* Fixes #313, by reversing the conversion between Kelvin and percentages
(100% now corresponds to the lowest Kelvin (highest Mired) value, and 0%
to the highest Kelvin (lowest Mired) value).

Also adds info about endpoind ID to log messages, and fixes one log
message.

* Replace binding-specific channel type for color temperature by system
channel type
closed 2018-12-06 by cdjackson