openhab/org.openhab.binding.zigbee #315: Wrong status on Coordinator and Zigbee devices

Hambootje, 2018-11-30:
I used both the CC2531EMK and Ember EM35x Coordinator via the Zigbee binding. 
When a zigbee lightbulb is switched off from the main power, the status of the thing remains online.
The same holds for the coordinator itself. This also show always online status, even if the dongle is removed.

For light bulbs this is not really a problem... You know when they don't work properl. It's different with for example doorsensors. Here you want to keep track that the device is still present and fire a notification when the thing goes offline.

From the log-file I can see that a regular polling cycle is started. If the device is online, the items are updated. If the zigbee-device is offline there are no values returned. I guess in this case the thing status can be set to OFFLINE.

### Configuration
Coordinator | Ember EM35x or CC2531EMK
openHAB version  | 2.4.0.SNAPSHOT
Hardware         |  Raspberry Pi 3+                                                                          
Java version     |    openjdk version "1.8.0_152"                                                                           
Devices            | zigbee:device:013C1ADE:000b57fffe909e96:000B57FFFE909E96_1_dimmer

cdjackson, 2019-01-12:
This is not super easy to resolve. For lightbulbs it is as they are mains powered, however for battery devices, most will not respond to requests, and therefore it is not easy (well, not possible even!) to decide if they are online or not.

Mains devices could be set offline after the polling period (or probably best to be 2x the polling period, or possibly 2x the reporting period). Battery devices I think should not be set offline unless we know that they are the sort that respond when asleep, which is a very short list of devices in my experience.

triller-telekom, 2019-02-06:
I would like to get back on this topic to improve the status handling of zigbee devices in openHAB.

> Mains devices could be set offline after the polling period (or probably best to be 2x the polling period, or possibly 2x the reporting period)

I agree, since devices have to poll the coordinator within that polling period. You are referring to 
EZSP_CONFIG_END_DEVICE_POLL_TIMEOUT and EZSP_CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT which are also mentioned here, right?

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2012/06/29/how_long_can_my_end-jseP

> Battery devices I think should not be set offline unless we know that they are the sort that respond when asleep, which is a very short list of devices in my experience.

This is a certainly a problem if some devices are faulty. But should that prevent us from setting other devices to offline?

IMHO we should set all devices to offline if they do not send any data within a certain timespan, see above. For the faulty devices we could introduce a thing property "alwaysOnline" or something like that and skip the check for those.

What is your opinion on that @cdjackson ?

triller-telekom, 2019-03-11:
@cdjackson: Friendly ping :)

Also I noticed that the Thinghandler sets tha Thing to ONLINE whenever the device provides data anyway. So even if we put a faulty device to OFFLINE, it will recover itself once it provides data.

cdjackson, 2019-03-11:
Sorry @triller-telekom - this dropped off my radar...

> I agree, since devices have to poll the coordinator within that polling period. You are referring to
EZSP_CONFIG_END_DEVICE_POLL_TIMEOUT and EZSP_CONFIG_END_DEVICE_POLL_TIMEOUT_SHIFT which are also mentioned here, right?

No - this is not what I meant. This is end device polling and the upper layers have no visibility of this as it's at the Mac layer, and it's also not applicable for routers (ie mains devices).

By "polling" I meant the polling that the binding does - ie at the application layer. If we poll the device, and it doesn't respond 2 consecutive times, then I would say it gets set OFFLINE. Additionally, given that polling should not be frequent, but reporting should be (relatively speaking), we could say if a device doesn't send a report within 2x the reporting period, then we set it offline. The 2x just allows for lost frames which will happen from time to time and we should avoid false OFFLINEs as much as possible.

> IMHO we should set all devices to offline if they do not send any data within a certain timespan

In general I agree...

> For the faulty devices we could introduce a thing property "alwaysOnline" or something like that and skip the check for those.

Ok, but I think that in general, we should try to make this automatic if we can rather than to expect the user to do it as it will create complaints ;) People won't do this, devices will go offline, and users will be upset.

As I've said - MOST EDs do not respect the requirement to poll within the required time, so MOST will not be pollable (if pollable is a word? ;) ). Anyway, we can still apply the reporting rule - so long as reporting is configured in at least one channel on the device.  The other thing that could be used to provide some input to this is the poll-control system, but at the moment this isn't something that is systematically handled in the ZSS libs. This would allow us to know when supporting EDs are expecting to report in - most newer EDs seem to support this at least, but probably the cheap ones don't.

> Also I noticed that the Thinghandler sets the Thing to ONLINE whenever the device provides data anyway. So even if we put a faulty device to OFFLINE, it will recover itself once it provides data.

Even so, I think we should avoid making things OFFLINE unnecessarily. Once a thing is OFFLINE, it is no longer possible to send commands to it.

---

Does that help? I suspect in general we're thinking along the same lines, and probably if you implement something along these lines we can argue further about the tuning ;)

hanneshofmann, 2019-04-15:
I think https://github.com/openhab/org.openhab.binding.zigbee/pull/412 addressed this issue and the MR has been merged a couple of days ago. Could you please close this @cdjackson?

closed 2019-04-15 by cdjackson