openhab/org.openhab.binding.zigbee #320: Channel factory not working

cdjackson, 2018-12-02:
@tomhoefer there are a lot of reports that after the latest updating devices now have no channels, and there is the following report -:

https://github.com/openhab/org.openhab.binding.zigbee/commit/0c39a4a836e59fd78342055161487099d0c4df9b#commitcomment-31517154

Please can you look at this as a matter of urgency?

hanneshofmann, 2018-12-03:
Thanks for the PR (#324) @puzzle-star, I am not yet confident that the PR may solve the actual problem because the `ZigBeeChannelConverterFactory` _should_ get injected by OSGi.

> I have not been able to find any call to ZigBeeHandlerFactory.setZigBeeChannelConverterFactory()

Since the `ZigBeeHandlerFactory` is an OSGi service component, I believe the `ZigBeeHandlerFactory.setZigBeeChannelConverterFactory` method should be only called by the framework to inject the reference - so I would expect that you may not find any caller of it. Furthermore, the `ZigBeeChannelConverterFactory` should be a static and mandatory reference of the `ZigBeeHandlerFactory` and therefore, it should be never null when the factory is passed through the constructor to the actual handler.

Could you please tell us how to reliable reproduce the NPE stated in https://github.com/openhab/org.openhab.binding.zigbee/commit/0c39a4a836e59fd78342055161487099d0c4df9b#commitcomment-31517154? So far, I could not manage to reproduce it while pairing more than 5 devices with the latest develop.

puzzle-star, 2018-12-03:
Nothing special to reproduce in my case. I just upgraded from 2.4.0.M6 to SNAPSHOT, and all my devices (around 40) failed to initialize with that error (which appeared for every device, even if I only pasted one).

This looks anyway different to the other reported issues in the forum, as my devices appear as offline after this, while in the forum I see they are online but with no channels.

I will go to a clean install, as maybe my null pointers are due to something corrupted in my install.

On the devices appearing without channels reported in the forum, is the "ZigBeeDefaultChannelConverterProvider" added to the OSGI created ZigBeeChannelConverterFactory?

I had to add it also as in my first fix the null pointers disappeared, but then I got the empty channels as in the forum reports.

hanneshofmann, 2018-12-03:
> I will go to a clean install, as maybe my null pointers are due to something corrupted in my install.

I'm assuming that the NPE is the result of an incorrect installation. Please keep us in the loop if you hit the NPE again on a clean installation.

> This looks anyway different to the other reported issues in the forum, as my devices appear as offline after this, while in the forum I see they are online but with no channels.

Ok, thanks for the hint. I will look into the issue that the factory does not provide any channels for some devices.

tomhoefer, 2018-12-03:
Hmmm.. strange ... I could imagine that the default provider is not injected during initial initialization - as quick workaround I would propose to change the deps from 0 .. n to 1 .. n until we find a better solution (I believe that we will run into this issue again at least in our project because we have to provide our own channel converters)

hanneshofmann, 2018-12-03:
>  I could imagine that the default provider is not injected during initial initialization - as quick workaround I would propose to change the deps from 0 .. n to 1 .. n until we find a better solution

I am not sure if that would actually solve the issue. Since the `ZigBeeDefaultChannelConverterProvider` has no external dependencies and it is exposed immediately, it should be correctly bound before the  `ZigBeeThingHandler` gets activated. Furthermore, according to the openHAB community, the channels were still not created after pairing multiple devices and therefore, I think it is not related to the cardinality.

@puzzle-star  Did you manage to reproduce the issue with a clean installation?

hanneshofmann, 2018-12-03:
@cdjackson I am still unable to reproduce the issue. Besides pairing more than 10 devices successfully and testing the implementation by enabling and disabling certain service components, the channels were still correctly created. I don't think that the reported issue relates to @tomhoefer's recent change. I had a detail look into the [change](https://github.com/openhab/org.openhab.binding.zigbee/commit/0c39a4a836e59fd78342055161487099d0c4df9b) and from my perspective, the code is absolutely fine. Maybe I miss something obvious but Thomas' commit has been merged 6 days ago whereas the openHAB community raised their concerns already raised 11 days ago.

puzzle-star, 2018-12-03:
On my side, I have tried reinstalling the bundles from M7 (no full reinstall by now) and everything went online correctly...

I will do a clean reinstall after backing up everything, just in case.

puzzle-star, 2018-12-03:
Definetely working after a clean install (2.4.0.M7). Added a totally new device and everything looks fine.

[image]

So there must have been something broken in my previous upgrade from 2.4.0.M6 to 2.4.0.M7

But wondering then about, i.e. https://community.openhab.org/t/zigbee-thing-channels-disappeared/57498/6?u=puzzle-star, reporting this also from two days ago...

BClark09, 2018-12-03:
Just checking related: I upgraded from `2.4.0.M6` to `2.4.0.M7` and most of the channels for my things have been retained. I am however now missing the switch channels in for all of my Philips Hue bulbs.

cdjackson, 2018-12-03:
In general, if you have a device with a dimmer, it should not have a switch. I guess this is what you are seeing?

BClark09, 2018-12-03:
That's right yes, colour bulbs have only Colour and Colour Temperature.

cdjackson, 2018-12-03:
That sounds correct - that’s the (so called) channel consolidation in action. You can still perform all functions through the single channel.

jjennings308, 2018-12-11:
> That sounds correct - that’s the (so called) channel consolidation in action. You can still perform all functions through the single channel.
> […](#)

If that is correct, then is there an update to the amazon alexa interface, so color bulbs can be control through Alexa?

jjennings308, 2018-12-11:
> This binding is not related to Alexa.

I know the binding is not associated with Alexa.  But I have setup to
control the device with the missing channels.  However, I was able to do
some research and reconfigure how I control the device with a Alexa and all
is good now.

cdjackson, 2018-12-11:
The ZigBee binding is using the standard concepts for these channels, so it should be the same as most other bindings.

jjennings308, 2018-12-11:
When I started using Alexa the color channel was not supported but it is
now. These zigbee components are the only color items I had. All is good
now.

cdjackson, 2018-12-23:
Closing this as I don't think there is an issue with the factory.

closed 2018-12-23 by cdjackson