openhab/org.openhab.binding.zigbee #321: Ubisys S2 input channels do not trigger events

AfromD, 2018-12-02:
Although the input channels of the Ubisys S2 are recognised automatically by the binding, there are no events logged in the event log, rules are not triggered, and Paper UI does not reflect changes.
The debug log does show that events are seen by the binding:
```2018-12-02 13:59:50.957 [DEBUG] [converter.ZigBeeConverterSwitchOnoff] - 001FEE00000115AF: ZigBee attribute reports ZclAttribute [cluster=ON_OFF, id=0, name=OnOff, dataType=BOOLEAN, lastValue=true, lastReportTime=Sun Dec 02 13:59:50 CET 2018]
2018-12-02 13:59:50.961 [DEBUG] [converter.ZigBeeBaseChannelConverter] - 001FEE00000115AF: Channel zigbee:device:c467a1d4:001FEE00000115AF:001FEE00000115AF_1_switch updated to ON
2018-12-02 13:59:50.963 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 001FEE00000115AF: Updating ZigBee channel state zigbee:device:c467a1d4:001FEE00000115AF:001FEE00000115AF_1_switch to ON
2018-12-02 13:59:50.993 [DEBUG] [converter.ZigBeeConverterSwitchOnoff] - 001FEE00000115AF: ZigBee command receiveds ToggleCommand [On/Off: 6241/3 -> 0/1, cluster=0006, TID=0F]
2018-12-02 13:59:51.214 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 001FEE00000115AF: Command for channel zigbee:device:c467a1d4:001FEE00000115AF:001FEE00000115AF_1_switch --> ON [OnOffType]
2018-12-02 13:59:56.612 [DEBUG] [converter.ZigBeeConverterSwitchOnoff] - 001FEE00000115AF: ZigBee attribute reports ZclAttribute [cluster=ON_OFF, id=0, name=OnOff, dataType=BOOLEAN, lastValue=false, lastReportTime=Sun Dec 02 13:59:56 CET 2018]
2018-12-02 13:59:56.614 [DEBUG] [converter.ZigBeeBaseChannelConverter] - 001FEE00000115AF: Channel zigbee:device:c467a1d4:001FEE00000115AF:001FEE00000115AF_1_switch updated to OFF
2018-12-02 13:59:56.618 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 001FEE00000115AF: Updating ZigBee channel state zigbee:device:c467a1d4:001FEE00000115AF:001FEE00000115AF_1_switch to OFF
2018-12-02 13:59:56.681 [DEBUG] [converter.ZigBeeConverterSwitchOnoff] - 001FEE00000115AF: ZigBee command receiveds ToggleCommand [On/Off: 6241/3 -> 0/1, cluster=0006, TID=11]
2018-12-02 13:59:56.903 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 001FEE00000115AF: Command for channel zigbee:device:c467a1d4:001FEE00000115AF:001FEE00000115AF_1_switch --> OFF [OnOffType]
```
Since the input channels are internally configured to trigger the output channels by default, a workaround is to use a rule to filter out changes in output that were not commanded by openhab. [example in the forum](https://community.openhab.org/t/zigbee-binding/15763/1561?u=afromd) The limitation of this workaround is that in default configuration, a change in output is only triggered after a _toggle_ of the input. So this workaround can't evaluate the input channel for its state, it can only be used as a momentary trigger, and the physical switch wired to the input needs to be a momentary switch.

cdjackson, 2018-12-02:
The reason for this not working is that the device is sending the ToggleCommand - not the OnOff command.

This will be easy to fix / add - although it probably needs a little thought on how to maintain state. I believe that there is also the possibility to change the configuration to make this an on/off switch. I will have a look at adding this in the first instance to see if it resolves the issue.

AfromD, 2018-12-02:
Yes configuration can be changed, it is in the technical manual
http://www.ubisys.de/downloads/ubisys-s2-technical-reference.pdf
It is in section 7.7.5
There is a special cluster to use for the configuration where you have to send a parameter according to how you want it to be configured. This can take several options and would probably need some configuration paramters for the Ubisys things.

cdjackson, 2018-12-02:
I won’t implement against their technical manual. I will use the standard documentation and hopefully they have implemented the standard. If not, we’ll worry about that next (although I think it’s unlikely!).

usrstephan, 2018-12-03:
The change of the "switch status" has only to be done in seldom cases.
Could it be an option to integrate some sort of "shell command" to change the cluster parameter "by hand"?
Following ubisys support changing this option is only possible with their gateway, not even with the USB-stick and their software.

cdjackson, 2018-12-03:
Yes - this is a configuration parameter, so it will be displayed as a configuration and you will need to change it manually. This is the same concept as I have already added for other configuration parameters such as dimmer configuration.

cdjackson, 2018-12-23:
Unfortunately Ubisys have used a private cluster to perform the configuration rather than the standard switch configuration cluster.

usrstephan, 2019-01-06:
What a pity!
What do you think is better: Could you explain in one or two sentences what cluster would be standard (so I could ask them to add this in a future firmware version) or is there a possibility to change the cluster value with some sort of "terminal command" in the binding (as it only has to be done in very seldom cases)
Sorry for the delay - but we were on holiday:-)

cdjackson, 2019-01-06:
There is a cluster called OnOffSwitchConfiguration cluster (cluster 7 if I remember correctly) which allows configuration of the OnOff switch. I’ve not looked in huge detail to see what Ubisys implemented that could not be done in the standard class - I would guess they have good reasons for using a custom class, but it’s worth asking them.

Let me think about the terminal command option - something might be possible there.

(and no problem regarding the delay - I hope you had good holidays :) ).

usrstephan, 2019-01-07:
Hi Chris,
yes holidays were great (skiing in Japan) - but too short...
I talked to ubisys today and the developers told me that the standard cluster offered them not enough options. With the special cluster they have 12 different options so they did not use the standard one.
They are planning further firmware improvements within the next months but they will not change the cluster:-(
So I'm waiting for your "terminal command option".

Jaques-Ludwig, 2019-06-18:
Hi,

I want to buy a ubisys S2, because I want to use my Zigbee Lamp with a normal Hardware switch and with openhab as software to switch with openhab app.

The ubisys service told me that this is makeable. I have to put on the hardware switch to an input of the s2 and on the software side I have to connect my lamp to an input of the s2. Therefore I have to configure the S2. With the Philips hue it is not possible. With the ubisys gateway it's possible, but only for configuration its too expensive.

Is it possible to configure the s2 with openhab? If its makeable, is there an instruction how to make it?

Thanks a lot,

Jaques

AfromD, 2019-06-18:
Hi jaques. I am using the s2. It works, but not as well as with the ubisys hub. 
1. There is no way to change settings, which means I had to convert my wall switches to momentary type. It is not possible to change the standard mapping of the inputs with the outputs.
2. The actions on the input channels are not visible in openhab. So I have to watch for the status of the associated output to change.

Jaques-Ludwig, 2019-06-18:
Hm, thats really a shame. The S2 is exactly what I'm looking for. But I'm not willing to pay 350 Euro for the ubisys Gateway only for configuration. And I don't know an alternative.

Is there a chance that configuration of s2 will be able in openhab in the future?

Thanks

Jaques

usrstephan, 2019-06-18:
Hi Jaques,
I use the ubisys devices for several months with OH without problems - except the one you described. It is not possible to change switch status without their bridge. Not even their installation software allows that. (They use special clusters to have more opportunities...). It is also not possible to see the "state" of the inputs in OH.
Chris doesn't see any opportunity to change this soon.
To overcome this problem I use ESP32s (MQTT). With the use of rules under OH many things are possible.

cdjackson, 2019-08-10:
There are two configuration parameters defined - which ones do you need? The ```InputConfigurations``` is not too hard to support - the ```InputActions``` is probably more difficult.

It should soon be possible to set the parameters from the openhab console for the ```InputConfigurations``` attribute.

[image]

There should be zigbee console commands available. The commands required are -:

```nodes``` to list the nodes

then

```write ENDPOINT CLUSTER ATTRIBUTE VALUE TYPE```

where -:

* ```ENDPOINT``` is the endpoint shown in the nodes list (it will have the format 12345/232 - so node address then endpoint ID). The UbiSys setup endpoint is 232.
* ```CLUSTER``` is the cluster ID - ```0xfc00``` for the UbiSys manufacturer specific config cluster
* ```ATTRIBUTE``` is the attribute ID - please read the UbiSys docs. For ```InputConfigurations``` this is 0.
* ```VALUE``` is the value to set - please read the UbiSys docs. Note that this will be an array with one byte per channel - the first element MUST start with square brackets```[``` and the last element must end with the square brackets ```]```
* ```TYPE``` is the datatype. This should be ```DATA_8_BIT``` for this attribute

Note that this is not implemented yet - if it is the ```InputConfigurations``` that is required here, I will look to implement this shortly. If it is ```InputActions``` I will have a look but it might be more problematic due to the data type.

usrstephan, 2019-08-12:
Hi chris,
unfortunately InputActions are needed to change the behavior.
file:///media/mobile/stick/guide.jpg
ubisys confirmed that to me today.
They also advised me to take a special look on the length of the commands and settings as the actual firmware has no check integrated (but it is planned for future versions)-

cdjackson, 2019-08-12:
Do you know what needs to be sent? Have you checked this with UbiSys?

It will take a little more time to get this working but I have managed to read the data and can probably come up with a way to send it - with other things happening, maybe it will be another week or so.

usrstephan, 2019-08-15:
Normally the commands are sent through their app. But they have a tutorial with examples:
[image]
[image]
[image]

cdjackson, 2019-08-26:
I've now tested this using the console in the ZSmartSystems repository. At some point in the near future I hope that this should also be available in OH.

Commands required to read the state of these two attributes -:

```
read 61585/232 0xfc00 0
read 61585/232 0xfc00 1
```

or 

```
read 61585/232 0xfc00 0 1
```

Will respond with -:

```
Attribute 0, type ORDERED_SEQUENCE_ARRAY, value: ZclArrayList [dataType=DATA_8_BIT, values=[0, 0]]
Attribute 1, type ORDERED_SEQUENCE_ARRAY, value: ZclArrayList [dataType=OCTET_STRING, values=[ByteArray [value=00 0D 03 06 00 02], ByteArray [value=01 0D 04 06 00 02]]]
```

To write these attributes -:

```
write 61585/232 0xfc00 0 [0 0] DATA_8_BIT
```

and
 
```
write 61585/232 0xfc00 1 [000D03060002 000D04060002] OCTET_STRING
```

In all of the above examples ```61585``` is the network address of the switch. The value I wrote to attribute 1 is the same value that was already in my switch so it's not consistent with the information above - please change as you see fit. Note that the ```06``` at the start of each array is not needed - this is the length and this will be handled automatically.

A word of warning - there's a statement above somewhere that states the length isn't checked! I managed to write the long string into attribute 0, so error checking here is VERY minimal. I simply changed it back and it doesn't seem to have upset anything, but then I've also not tried to use the switch other than for ZigBee communications. So - **BE CAREFUL** - you have been warned - with power comes responsibility ;)

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/724

usrstephan, 2019-08-26:
Hi Chris,
that actually sounds fantastic - thanks for your great work!!!
Please let me know as soon as I can test it.

cdjackson, 2019-09-20:
This should now be available in openHAB. Note that the commands above need to be started with ```zigbee``` to make them work in the openhab console.

usrstephan, 2019-10-29:
Hi chris,
after a long while I tested it but had no success so far. Is my version too old? Do I need other commands?
[image]

cdjackson, 2019-10-29:
This should be fine - can you check what bundles are loaded - check that the ZigBee console bundles are loaded. 

I can’t easily check at the moment as I’m on a flight to Japan, but please let me know and I’ll try to help work out what’s up...

usrstephan, 2019-10-29:
I think you are right:
254 │ Active │  80 │ 1.2.5                 │ com.zsmartsystems.zigbee.dongle.xbee
259 │ Active │  80 │ 1.2.5                 │ com.zsmartsystems.zigbee.dongle.telegesis
260 │ Active │  80 │ 1.2.5                 │ com.zsmartsystems.zigbee.dongle.ember
262 │ Active │  80 │ 1.2.5                 │ com.zsmartsystems.zigbee
263 │ Active │  80 │ 1.2.5                 │ com.zsmartsystems.zigbee.dongle.cc2531
What's the easiest way to load the console bundles?
I found no options in the script of 5iver

cdjackson, 2019-10-30:
Good question. I think they are built as part of the CI system so you should be able to download them and add them to the addons folder. I will check when I get on my computer.

cdjackson, 2019-10-30:
If you go to the following link, and then click on the console link, and also the console.ember link and download the target files. This will contain the JAR, and also the sources JAR - you only need the main JAR of these two files. Additionally, you will also need the corresponding two files from the ZsmartSystems repository.

I wonder if @openhab-5iver might be able to add these to his fantastic update script file as there are other good features of these files (such as network backup).

cdjackson, 2019-10-30:
And I forgot to paste the link - I’ve been up for around 30 hours now so getting a little tired ;)

https://ci.openhab.org/job/openHAB2-Bundles/lastStableBuild/artifact/bindings/org.openhab.binding.zigbee/ <https://ci.openhab.org/job/openHAB2-Bundles/lastStableBuild/artifact/bindings/org.openhab.binding.zigbee/>

cdjackson, 2019-10-31:
Thanks - I think we're a little early for skiing ;) We're heading into the mountains next week - first we have 5 nights in Tokyo.

The file you need is here -:
https://dl.bintray.com/zsmartsystems/com.zsmartsystems/com/zsmartsystems/zigbee/com.zsmartsystems.zigbee.console/

You just need to get the same version as you have for the other zsmartsystems files you see in your system (I forget what version is used at the moment).

This should give you the commands you need...

usrstephan, 2019-10-31:
console is working now - thanks very much
but I get the next error:
openhab> zigbee read 52876/232 0xfc00 0
Error: Exception during command execution (IllegalArgumentException): A cluster specified by 0xfc00 is not found for endpoint 232
In your citation from the manual it says "it is not yet implemented"
Did these commands work for you? Did you upgrade the firmware of the ubisys device first?
PS: enjoy your holidays first....

cdjackson, 2019-10-31:
Yes, I tested this out a few months ago and it was working. I think you may need to discover the device again as there were some further changes to the way the framework discovers devices, or more correctly - what it puts in it’s internal structures for clusters that it doesn’t know. This information is saved in the XML file in the userdata/zwave folder so probably you should delete the XML for this device and let the binding rediscover it.

I didn’t update the firmware - my firmware is actually very old.

Thanks :)

usrstephan, 2019-11-04:
Did a complete reset of the zigbee devices
Now I get a timeout:
"openhab> zigbee read 46834/232 0xfc00 0
Reading endpoint 46834/232, cluster server cluster  (FC00), attributes Attribute 0
Error executing command: CommandResult [TIMEOUT]"

cdjackson, 2019-11-04:
It’s possible your device may expose different parameters, but a timeout is strange. The endpoint is now found as you don’t get the previous error, and if the attributes weren’t supported, then you should get a response saying that the attribute doesn’t exist.

Is this repeatable (I assume so). What about if you do
zigbee read 46834/1 6 0

usrstephan, 2019-11-04:
I tried it several times - it is repeatable. With your new command I get a similar result
openhab> zigbee read 46834/1 6 0                                                
Reading endpoint 46834/1, cluster server cluster On/Off (0006), attributes On Off
Error executing command: CommandResult [TIMEOUT]

openhab> zigbee read 46834/232 0xfc00 0                                      
Reading endpoint 46834/232, cluster server cluster  (FC00), attributes Attribute 0
Error executing command: CommandResult [TIMEOUT]

openhab> zigbee read 46834/232 0xfc00 0 1                                       
Reading endpoint 46834/232, cluster server cluster  (FC00), attributes Attribute 0, Attribute 1
Error executing command: CommandResult [TIMEOUT]

cdjackson, 2019-11-04:
Strange - is the switch actually working at all?

usrstephan, 2019-11-04:
thanks for your patience. I checked it before - but it was actually offline
after a reboot everything works great!!!

cdjackson, 2019-11-05:
Ok, cool. So does that mean that this issue can now be closed? :)

usrstephan, 2019-11-05:
YES
thanks very much!!!

cdjackson, 2019-11-05:
Great - thanks for confirming and testing...

closed 2019-11-05 by cdjackson