openhab/org.openhab.binding.zigbee #322: Network state file gets easily lost when a power fail or crash happens

puzzle-star, 2018-12-02:
Network state file is not written atomically, and is fully rewritten quite often, so if a power fail or a crash occurs, the network state is easily lost, forcing a full network rediscovery.

Opening this issue to be referenced from the related PR.

puzzle-star, 2018-12-02:
Hopefully fixed the PR. Only the needed commit on it

puzzle-star, 2018-12-04:
Hi @cdjackson,

Any chance to merge this before M8? It should add some stability in case of crash, as rediscovery when the file is lost is painful, specially on big networks.

Best Regards, 
Pedro

cdjackson, 2018-12-04:
I don’t think there will be an M8, but I will try and get it merged ASAP - I’ll look at it in the next couple of days.

cdjackson, 2018-12-05:
I should add that really, if you are having issues with power, you should solve that problem. Any system that has power removed frequently, as you seem to indicate is happening on your system will remain unreliable.

puzzle-star, 2018-12-05:
Agreed, but RasbPi are quite "sensitive" some times, and systems with low memory may end up also in killing the Java process at the worst moment...

So this is an enhancement that would provide some safety against this, hopefully at a very low cost.

The file is quite big and i.e. in my network I see it written several times in a raw, so anything going wrong at that time will garble it. Writing to a temp file and renaming (the best way and most atomically way possible) is more a fail safe operation in this case

puzzle-star, 2018-12-05:
Hi @cdjackson,

Well, I have done some tests under Linux and Windows, and there seems not to be a portable way to do the rename / move atomically in a safe way without a bit of coding, considering the target file might be open (i.e. by a backup).

java.nio.Files.move(orig, dest, ATOMIC_MOVE) will do it correctly (and atomically!) both under Linux and Windows even if the target file exists, but will fail under Windows (not Linux) when the target file is open (curiously enough, it does not fail when the original file is open instead...).

So the only way I found to do this in a portable way, considering the above, is making sure the target file of a rename does not exist if we want the operation to work even if it is open (i.e., by renaming the target file itself to a new unique name beforehand).

Would if be OK for you if I implement it like this?

File save:
- Write the new config to a temporary file ([_name_].xml.new)
- Rename the old config file to a backup unique file name ([_name_].xml -> [_name_].xml.[_timestamp_])
- Rename the new temporary config file to the target file, which does not exist thanks to the previous rename ([_name_].xml.new -> [_name_].xml)
- Remove the backup file ([_name_].xml.[_timestamp_]). We could keep it and remove old ones instead, but is this necessary?

File load:
- If the standard config file ([_name_].xml) does not exist, and a temporary new one ([_name_].xml.new) does, rename the new temporary one into the config file (([_name_].xml.new) -> [_name_].xml).
- Open the standard file normally ([_name_].xml). If it does not exist it is either new or unrecoverable

As the old config name is renamed **after** the new temporary one is written, if the old one does not exist and the new one does, we should assume it is recoverable from there (the old one would exist otherwise).

Your thoughts on this?

Pedro

puzzle-star, 2018-12-09:
Hi @cdjackson,

Just updated the PR with the above.

Now:
- The config file is always complete / readable (if it exists). So it can be backed up as is.
- In case of a previous crash just when renaming the config file, it is always recovered from a complete copy.
- The file can always be renamed and moved, even if open (Linux and Windows). There are some error checks in the code, but we should never see them happening unless there is somethig really wrong in the setup
- No history is kept (not necessary)

I have tested it on Linux (including forcing the file being lost / recovered), but not on Windows, even if there is no "Windows specific" code in the PR, as I do not have OH running under Windows to test.

What I have, anyway, tested under Windows is that the renames work, in the given order, even if the xml is open for a backup.

Pedro

cdjackson, 2018-12-21:
#322 was merged so closing this.

closed 2018-12-21 by cdjackson