openhab/org.openhab.binding.zigbee #325: Transaction timeout while transferring the last chunk of the firmware image

hanneshofmann, 2018-12-03:
While trying to update the device firmware of the Bitron Home 902010/25, the last transaction of the `ImageBlockResponse` has a timeout and therefore, the firmware update gets aborted by the binding. Please note that we have observed that issue not only with the Bitron Home device but also with devices from other vendors (e.g. CentraLite, SengLED,..).

The following log captures the transfer of the last chunk of the firmware image:
````
2018-11-30_13:50:29:174	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ImageBlockCommand [OTA Upgrade: 17784/4 -> 0/1, cluster=0019, TID=C1, fieldControl=2, manufacturerCode=10132, imageType=257, fileVersion=48, fileOffset=211140, maximumDataSize=36, requestNodeAddress=null, blockRequestDelay=0]
2018-11-30_13:50:29:175	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.otaserver.ZclOtaUpgradeServer] - 17784/4 OTA Data: Sending 36 bytes at offset 211140
2018-11-30_13:50:29:177	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ImageBlockResponse [OTA Upgrade: 0/0 -> 17784/4, cluster=0019, TID=D9, status=SUCCESS, manufacturerCode=10132, imageType=257, fileVersion=48, fileOffset=211140, imageData=ByteArray [value=80 02 79 00 02 12 11 C0 82 C0 83 7A 00 79 00 12 23 EA D2 87 90 61 96 74 06 F0 D0 83 D0 82 02 12 11 FF FF FF]]
2018-11-30_13:50:29:178	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=217, commandId=5]
2018-11-30_13:50:29:180	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/1, destinationAddress=17784/4, profile=0104, cluster=25, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=217, payload=09 D9 05 00 94 27 01 01 30 00 00 00 C4 38 03 00 24 80 02 79 00 02 12 11 C0 82 C0 83 7A 00 79 00 12 23 EA D2 87 90 61 96 74 06 F0 D0 83 D0 82 02 12 11 FF FF FF]
2018-11-30_13:50:29:182	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - EzspSendUnicastRequest [type=EMBER_OUTGOING_DIRECT, indexOrDestination=17784, apsFrame=EmberApsFrame [profileId=260, clusterId=25, sourceEndpoint=1, destinationEndpoint=4, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=217], messageTag=217, messageContents=09 D9 05 00 94 27 01 01 30 00 00 00 C4 38 03 00 24 80 02 79 00 02 12 11 C0 82 C0 83 7A 00 79 00 12 23 EA D2 87 90 61 96 74 06 F0 D0 83 D0 82 02 12 11 FF FF FF]
2018-11-30_13:50:29:182	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - TX EZSP queue: 1
2018-11-30_13:50:29:185	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - TX EZSP: EzspSendUnicastRequest [type=EMBER_OUTGOING_DIRECT, indexOrDestination=17784, apsFrame=EmberApsFrame [profileId=260, clusterId=25, sourceEndpoint=1, destinationEndpoint=4, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=217], messageTag=217, messageContents=09 D9 05 00 94 27 01 01 30 00 00 00 C4 38 03 00 24 80 02 79 00 02 12 11 C0 82 C0 83 7A 00 79 00 12 23 EA D2 87 90 61 96 74 06 F0 D0 83 D0 82 02 12 11 FF FF FF]
2018-11-30_13:50:29:188	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - --> TX SPI frame: FE 48 D9 00 34 00 78 45 04 01 19 00 01 04 40 11 00 00 D9 D9 35 09 D9 05 00 94 27 01 01 30 00 00 00 C4 38 03 00 24 80 02 79 00 02 12 11 C0 82 C0 83 7A 00 79 00 12 23 EA D2 87 90 61 96 74 06 F0 D0 83 D0 82 02 12 11 FF FF FF 
2018-11-30_13:50:29:305	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - <-- RX SPI frame: FE 05 D9 80 34 00 59 
2018-11-30_13:50:29:306	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - RX EZSP: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=89]
2018-11-30_13:50:29:306	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=89]
2018-11-30_13:50:29:306	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Unhandled EZSP Frame: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=89]
2018-11-30_13:50:29:671	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - <-- RX SPI frame: FE 14 D9 88 3F 00 78 45 04 01 19 00 01 04 40 11 00 00 59 D9 00 00 
2018-11-30_13:50:29:672	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.internal.spi.SpiFrameHandler] - RX EZSP: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=17784, apsFrame=EmberApsFrame [profileId=260, clusterId=25, sourceEndpoint=1, destinationEndpoint=4, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=89], messageTag=217, status=EMBER_SUCCESS, messageContents=]
2018-11-30_13:50:29:672	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=17784, apsFrame=EmberApsFrame [profileId=260, clusterId=25, sourceEndpoint=1, destinationEndpoint=4, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=89], messageTag=217, status=EMBER_SUCCESS, messageContents=]
2018-11-30_13:50:29:672	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - Unhandled EZSP Frame: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=17784, apsFrame=EmberApsFrame [profileId=260, clusterId=25, sourceEndpoint=1, destinationEndpoint=4, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=89], messageTag=217, status=EMBER_SUCCESS, messageContents=]
2018-11-30_13:50:37:176	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransaction] - Transaction timeout: ImageBlockResponse [OTA Upgrade: 0/0 -> 17784/4, cluster=0019, TID=D9, status=SUCCESS, manufacturerCode=10132, imageType=257, fileVersion=48, fileOffset=211140, imageData=ByteArray [value=80 02 79 00 02 12 11 C0 82 C0 83 7A 00 79 00 12 23 EA D2 87 90 61 96 74 06 F0 D0 83 D0 82 02 12 11 FF FF FF]]
2018-11-30_13:50:59:174	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.otaserver.ZclOtaUpgradeServer] - 17784/4: OTA Error: Timeout - aborting transfer.
2018-11-30_13:50:59:174	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.otaserver.ZclOtaUpgradeServer] - 17784/4 OTA status updated to OTA_UPGRADE_FAILED.
2018-11-30_13:50:59:178	NA	DEBUG:	[symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeThingHandler] - 00124B000420D361: OTA transfer status update OTA_UPGRADE_FAILED, percent=99
2018-11-30_13:50:59:180	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.otaserver.ZclOtaUpgradeServer] - 17784/4 OTA status updated to OTA_CANCELLED.
2018-11-30_13:50:59:184	NA	DEBUG:	[symbolicName=org.openhab.binding.zigbee, className=org.openhab.binding.zigbee.handler.ZigBeeThingHandler] - 00124B000420D361: OTA transfer status update OTA_CANCELLED, percent=99
````

@cdjackson Please let me know if you require further information for your analysis.

cdjackson, 2018-12-03:
Can I suggest to open this in the zsmart systems repository as it doesn't seem to be binding related from a quick glance at the log.

hanneshofmann, 2018-12-03:
Sorry! I will create the issue in ZSS.

closed 2018-12-03 by hanneshofmann