openhab/org.openhab.binding.zigbee #326: Recognize color mode of color bulb

TomTravaglino, 2018-12-03:
This issue is related to the PR https://github.com/eclipse/smarthome/pull/5762

We want to recognize if the color of the bulb has been set via color scheme or via color temperature scheme. In the Hue binding this has been solved by setting the color temperature to `UnDefType.NULL` when the color mode is not 'color temperature' (see linked PR). We would like to do the same in the ZigBee binding.

Here the idea: In the method `ZigBeeConverterColorTemperature.attributeUpdated(ZclAttribute attribute)` when `attribute` is `ColorMode` and its value is not '0x02' (ColorTemperatureMireds) then we would set the channel status to `UnDefType.NULL`. The additional code could look like this 
```
        if (attribute.getCluster() == ZclClusterType.COLOR_CONTROL
                && attribute.getId() == ZclColorControlCluster.ATTR_COLORMODE) {
            
            Integer colorMode = (Integer) attribute.getLastValue();
            if (colorMode != ColorModeEnum.COLORTEMPERATURE.getKey()) {
                updateChannelState(UnDefType.NULL);
            }
        }
```        
But I faced an issue when I investigated this approach. When you switch from color to color temperature and vice versa then `ColorMode` is not updated automatically (I checked it in the console) and so the listener is not called. I tried then to read `ColorMode` with a console command and in that moment its value was updated and the listener was called.
How can we solve this issue? Is it an option to (force-)update `ColorMode` by calling `com.zsmartsystems.zigbee.zcl.ZclCluster.read(int)` when either HSB or the color temperature changes?

AfromD, 2018-12-11:
I have suggested something along the lines of this before, but I wasn't successful
https://github.com/openhab/org.openhab.binding.zigbee/issues/39

I have resorted to dummy color temperature channels, which trigger rules that send everything off as a HSB command, thus keeping the lamps always in color mode. IIRC, I use 45 hue and 0-40 saturation to vary between cold and warm white.

cdjackson, 2018-12-11:
> I have suggested something along the lines of this before, but I wasn't successful

#41 
#46 

I think I saw that since I was told to remove this channel, it has been added to the Hue binding, so this might be justification to add it again.

TomTravaglino, 2018-12-13:
> I think I saw that since I was told to remove this channel, it has been added to the Hue binding, so this might be justification to add it again.

Do you mean this discussion https://github.com/eclipse/smarthome/pull/5762? In the end, the channel was not introduced and instead decided to set the color temperature to "UnDefType.NULL" when the color is set by the color scheme.

> Could the reason be that you have not configured reporting for the color mode attribute?

I activated the reporting of the color mode attribute in the `ZigBeeConverterColorTemperature` by adding this instruction in `initializeConverter()`:

```
        ZclAttribute colorModeAttribute = clusterColorControl
                .getAttribute(ZclColorControlCluster.ATTR_COLORMODE);
        clusterColorControl.setReporting(colorModeAttribute, 1, REPORTING_PERIOD_DEFAULT_MAX, 1);
```

Now the color mode is reported and the color temperature could be set to `UnDefType.NULL`

PR #336 (2019-01-14): Recognize color mode of color bulb
Resolves #326

This is an approach to recognize if the color has been set via color scheme or via color temperature scheme. It corresponds to the solution realized in https://github.com/eclipse/smarthome/pull/5762 which sets the color temperature to UnDefType.NULL when the color mode is not "color temperature".

Your feedback is welcome.
closed 2019-01-14 by cdjackson