openhab/org.openhab.binding.zigbee #348: Bulb not turned on after brightness attribute report.

t-8ch, 2018-12-25:
### Outline

When turning off a Hue bulb  and waiting for ~1 Minute the bulb will not turn on again, when selecting same brightness level as before turning off. The same scenario works when the bulb is turned on faster again.

#### Working scenario:

* Set bulb to `$COLOR` at `$BRIGHTNESS`.
* `bulb switches to correct settings`
* Turn bulb off.
* `bulb shuts off`
* Do not wait.
* Set bulb to `$COLOR` at `$BRIGHTNESS`.
* `bulb restores its settings`

#### Non-working scenario

* Set bulb to `$COLOR` at `$BRIGHTNESS`.
* `bulb switches to correct settings`
* Turn bulb off.
* `bulb shuts off`
* Wait > 1 min
* Set bulb to `$COLOR` at `$BRIGHTNESS`.
* `bulb stays off`

### Cause

When turning off the bulb, the binding remembers this and normally sends a `MoveToLevelWithOnOffCommand` packet.
However this bulb send periodic (1 minute) attribute updates. When the update is received the binding will store this attribute. When trying to turn on the bulb again the binding [compares the new value to the old value](https://github.com/openhab/org.openhab.binding.zigbee/blob/79dfaec91fcad6ea625c485a4e29e92dd7ff9565/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/converter/ZigBeeConverterColorColor.java#L334) to determine if the command would be a NOOP.
Because the current brightness value as reported by the attribute update is the same as the one set via the command, `updateBrightness()` will not be called and the bulb will stay off.

### Possible fixes

1) Remove the check in `handleCommand()` (Tested locally, seems to work)
1) Only update the color characteristics when the bulb is not off.

(Personally I would prefer to keep the behaviour-changing state synchronization between the binding and the physical devices to a minum and go with number 1)

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used | CC2531 stick                                                                            |
| openHAB version  | 2.4.0
| Hardware         | RPi 3B+                                                                               |
| Memory           |                                                                                |
| Java version     |                                                                                |
| Devices          | Provide all device types and their addresses                                   |

cdjackson, 2018-12-25:
Personally, I would agree that removing the check is probably ok, but I will take a better look at this later. I'm not a big fan of filtering state updates in the binding as I think this should be (and is) handled by the framework. However @puzzle-star spent quite some time on this converter so I think it would be good to get his input in case we are missing something...

t-8ch, 2018-12-25:
Sure

puzzle-star, 2018-12-26:
Hi @cdjackson and @t-8ch ,

The issue here, I think, is that the Phillips bulbs are reporting brightness even when OFF. The switch level binding fixes this by only considering brightness updates when ON.

Removing the check for equallity would break implementation for IKEA lights, that do not accept brightness and color change too close in time. This is an issue with the IKEA lights but definetly should not affect the behaviour of the other ones, even if sending updates on brightness when OFF is also strange.

The fix for both issues at the same time, rather than removing the check, would be moving the check on "delayedColorChange" (which is set by default for XY lights, being IKEA the only ones I know of) to the upper "if" clause, so that the check for equalitty is done only for XY bulbs.

The real fix would, anyway, be doing the same as for the switchlevel converter (ignoring brightness updates when OFF), as otherwise the framework will incorrectly think the light is on when receiving an brightness update.

But also @cdjackson, the behaviour of switchlevel channel has been changed recently so that "ON" means "back to previous state" instead of "ON to 100%". You know I was not quite supportive on this because of some side effects, but as it is done, would you think this should be also changed in the color converter to be consistent?

I can refactor a bit the color converter to bring together the above changes, if you like:
- ON to mean "back to previous state"
- Ignore (or cache) brighness updates when OFF
- Make equality check on brighness only happen on XY lights (IKEA fix)

t-8ch, 2018-12-26:
To be explicit: I only remove the condition of the `if` block. The body (including the `Thread.sleep()` stayed.

cdjackson, 2018-12-26:
> The issue here, I think, is that the Phillips bulbs are reporting brightness even when OFF. 

I don't think this is an uncommon case though - I have Osram bulbs that do this as well.

I think that when we originally discussed this, we saw this as well -:

https://github.com/openhab/org.openhab.binding.zigbee/issues/83#issuecomment-356440202

> What is seen in #81 is that a device is reporting both on/off, and level. So a device can seemingly report OFF, but level=254. If these reports coming in the wrong way around, then it's a potential problem...

> would you think this should be also changed in the color converter to be consistent?

Yes, I do plan to do this (I think it was mentioned earlier) - for me, all converters that implement the inherited functionality should implement consistently.

puzzle-star, 2018-12-26:
Hi @t-8ch,

I understand, but then all color changes will make the IKEA lights change brightnes (even if noop), wait for one second, and then change color. The check in brightness makes the color change smoother on all lights (not only IKEA), and definetely better in IKEA lights.

That is why I suggest to move the condition to the first if, so that it affects IKEA lights instead of totally removing it).

But also the channel needs to implement the same fix that was done for the switchlevel channel (nothing to do with IKEA issues), as getting a report on brightness, and accepting it when the lioght is OFF, makes the framwork think the light is ON, which is not true.

BTW, the "delayedColorChange" variable was implemented thinking of doing it a configuration parameter. It takes false as default expect for XY lights. The "configuration" part was never implemented, as the only ligths I know working on XY mode are IKEA, so the default just works.

puzzle-star, 2018-12-26:
> Yes, I do plan to do this (I think it was mentioned earlier)

Then I wait for you to change it, unless you have too much work to do: this is a "simple change", so I could do it if necessary. :-)

t-8ch, 2018-12-26:
What about something similar to the generic button functionality:
A mechanism to override properties for matching things that will influence behavior?
This way we could do direct matches on `Vendor=IKEA` while at the same time allowing users to provide the property manually for devices that are not handled out of the box by the binding.

cdjackson, 2018-12-26:
> On 26 Dec 2018, at 09:47, Pedro <notifications@github.com> wrote:
> 
> Yes, I do plan to do this (I think it was mentioned earlier)
> 
> Then I wait for you to change it, unless you have too much work to do: this is a "simple change", so I could do it if necessary. :-)
> 

I am more than happy for you to update it as part of any fix we agree here - thanks. :)

puzzle-star, 2018-12-26:
> What about something similar to the generic button functionality:
> A mechanism to override properties for matching things that will influence behavior?

This is probably the way to configure it, but this will not fix the issue of some lights reporting brightness when off. The fic for this is making the color channel behave the same as the switchlevel channel, that already addresses this.

puzzle-star, 2018-12-26:
> I am more than happy for you to update it as part of any fix we agree here - thanks. :)

Would these do?

- ON to mean "back to previous state" (as in SwitchLevel converter)
- Ignore (or cache) brightness and color updates when OFF  (as in SwitchLevel converter)
- Make equality check on brighness only happen on XY lights (IKEA fix) (to prevent this branch of the code happening for other lights, as it should not be necessary, even if harmless if the former one is implemented)

cdjackson, 2019-01-12:
Welcome back.

This sounds fine - the only comment is that there is also #336 which might impact this and we might want to merge first. I've not looked at it yet as CI is broken, but this also caches the colour and sets the UNDEF if color mode is CT, so I think it will impact these changes, and therefore probably best merged first.

I'm not sure at the moment why CI is broken here and ZWave (and maybe in other OH repos, but I don't know) - I hope we can resolve that in the coming days.

cdjackson, 2019-01-14:
@puzzle-star CI finally started working again this morning so I've got in and merged #336 while I could :).

puzzle-star, 2019-01-18:
Hi @cdjackson and @t-8ch 

I just submitted a PR (https://github.com/openhab/org.openhab.binding.zigbee/pull/367) for this.

Just one comment. This relies on lastHSB variable (same as with lastLevel in the SwitchLevel converter) to have a more or less up-to-date value of the last know state of the device.

As some devices do not always report prompty (or at all), this may not be true when we recently sent a command.

Would it make sense to update the value of lastHSB / lastLevel also when we process commands and not only when we receive updates? (this has nothing to do with the "autoupdate" feature)

Pedro

t-8ch, 2019-01-18:
The bulb I observed started to report it after one minute, so updating it ourselves seems to be necessary.

puzzle-star, 2019-01-19:
Hi @cdjackson, do you think we should cache the value sent in a command for lastHSB, and not only the reported values ?

I think we should, but the switchLevel channel is not doing so.

I would then prefer submit this one as is to make it consistent with the SwitchLevel cluster, and then check the possibility and implications of caching the command values, and not only attributes reports, for both color and level clusters.

What do you think?

puzzle-star, 2019-02-09:
Hi @cdjackson,

With the last PR the color binding still does not work properly. I am sumbitting a new one for fix the pending issues

puzzle-star, 2019-02-09:
Hi @cdjackson ,

PR submitted. Please let me know if it looks OK for you or if you want me to change something.

It is now working correctly with IKEA color lights and FLS-pp led strips. I have no Philips Hue lights to test.

cdjackson, 2019-08-11:
I believe that this is closed with the above merged PR. If not, please reopen and provide information on what is still to be done

closed 2019-08-11 by cdjackson