openhab/org.openhab.binding.zigbee #357: IKEA TRADFRI remote control switch is not recognized as trigger

lubosz, 2019-01-12:
### Outline

Since I am on OpenHAB 2.4.0 Release Build, I guess I should have this [related smarthome patch](https://github.com/eclipse/smarthome/pull/4373).

I don't have an Ikea gateway, I want to use the remote with my Qivicon ZigBee controller.
The controller works fine with controlling the Ikea Lamp and Osram smart plugs.

### Behaviour

To get the remote into paring mode you need to press the pair button 4 times relatively quickly. The red LED blinks red shortly if you are successful. This was the case for me on firmware `11115720`. 

After that my Qivicon ZigBee controller lists the device as `IKEA of Sweden TRADFRI remote control`.
It shows it first as a generic ZigBee device, but if I wait a bit, I get the proper name.

The remote shows 4 channels in PaperUI. 3 for battery info and a switch.

### Problem

The problem is that the switch is recognized as Actor, not Sensor. So I can set a switch value, which does not do anything, and there are no triggers.
The battery sensors don't output anything though, but the thing is showed as online.
The four secondary buttons are apparently not recognized either.

Nothing shows up in the event log when I press the buttons.

The guy from [this video](https://www.youtube.com/watch?v=MCj34Q-0zfY) made it work with pimatic. Their implementation is documented [here](https://pimatic.org/plugins/pimatic-tradfri/). But it looks that this needs a Ikea Gateway, what I don't want.

The remote can be connected to third party gateways, as you can see here with the Phillips Hue
https://www.youtube.com/watch?v=HFq3xi2fmYY.

Can I make this work through configuration or needs anything to be implemented?

This is also discussed in the [forums](https://community.openhab.org/t/tradfri-remote-support/34695).

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |                                                                                |
| openHAB version  | 2.4.0 Release                                                                  |
| Hardware         |                                                                                |
| Memory           |                                                                                |
| Java version     |                                                                                |
| Devices          | Provide all device types and their addresses                                   |

### Logs

This is how it looks in `org.eclipse.smarthome.core.thing.Thing.json`. I guess there should be a `"kind": "TRIGGER"` in there.
https://gist.github.com/lubosz/fd0f8c2d596a0282dd01952c83a57ee0

A rather long log, where you can see the discovery and addition of the Thing via PaperUI. (in the second half of the log)

https://gist.github.com/lubosz/cd285a0e120e1c11779f717cfc1f09ea

sabefix, 2019-01-12:
> Do any other buttons / remotes work with this binding already?

Have a look over there: https://github.com/openhab/org.openhab.binding.zigbee/pull/284#issue-226550002

cdjackson, 2019-01-14:
> ZigBee is new for me, so I hope this isn´t a dumb question.
> It is possible to simulate a bulb within this binding? I understand that there is the coordinator who manages groups and devices, but is the API from the USB device capable to listen to other devices like a bulb would do it?
> My understing is that the ikea dimmers and remotes are connecting to the bulbs. Therefore they have a paring function which is not the same as pairing them to OH. In my case I need to reset the dimmer (push 4x the pair-button) to connect to OH. Afterwards I can pair (press 10 seconds) to a HUE bulb without loosing the OH connection. The idea would be to connect the dimmer to a virtual bulb within OH to receive the commands from the dimmer.

Yes, you should be able to do this, but I suggest to discuss on the forum where there will be more people who can help get you up and running.

cdjackson, 2019-01-14:
> Thanks for your fast response!
> I thought that the binding needs adaption for this. But I´ll try it in the forum.
>
> Best regards!

Maybe I don’t understand your question fully then, so again best to discuss on the forum.

CaeruleusAqua, 2019-02-10:
It seems that the TRADFRI remote is working in zigbee2mqtt:
https://github.com/Koenkk/zigbee2mqtt/issues/102

BTW: I also want to use the remote directly without the Ikea gateway!

cdjackson, 2019-02-10:
> It seems that the TRADFRI remote is working in zigbee2mqtt:

That is not related to this binding so it doesn't really help here right?

CaeruleusAqua, 2019-02-10:
It is related to this issue and gives hints how to fix it, so indeed, it should be related to this binding!

See e.g. here:
> I have this remote control.
> It is successfully connected to coordinator. But coordinator not receive it's messages.
> I found why! http://faire-ca-soi-meme.fr/domotique/2018/03/14/une-telecommande-zigbee-pour-les-controler-tous/
> It's send message to "group". Coordinator not endDevice in group and do not recived it's message.
> 
> I do not known what to do... May be put coordinator to group? But how?

cdjackson, 2019-02-10:
The point is though that it needs someone to write the code - understanding the issue is only a small part of the problem (and this was already understood). Please feel free to provide a PR to update the binding - it would be appreciated I'm sure.

booxdk, 2019-03-03:
I've just installed a CC2531_coordinaotr since I would like to use the IKEA on/off remote to trigger some actions in openHAB and realized the same as lubosz, that I can pair the remote but don't get any channels. I'm not skilled enough to solve this programming my self and is hoping for someone with the skills to help out. It would be awesome to be able to pair the cheap IKEA remotes for use with openHAB. And of course support for the IKEA motion detector would be nice to have too ;-)

cdjackson, 2019-03-03:
Please discuss on the community forum where there will be more people to help you. The issues list is more for bugs or issues rather than support.

Thanks.

jutoft, 2019-03-04:
> It seems that the TRADFRI remote is working in zigbee2mqtt:
> [Koenkk/zigbee2mqtt#102](https://github.com/Koenkk/zigbee2mqtt/issues/102)
> 
> BTW: I also want to use the remote directly without the Ikea gateway!

The button works by sending commands to a zigbee group.

Yes it should work with zigbee2mqtt. But it does require the user to sniff zigbee traffic to get the groupid the button transmits to. And then set it in the item's configuration after.

jhron, 2019-04-13:
I just tried to pair Ikea remote control with Openhab and when I press buttons I can see following output in debug log.

Main Button
`
21:43:29.699 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 17 44 81 39 5C 06 00 EB 5D 01 01 00 00 00 8B 43 5D 00 00 03 01 62 02 EB 5D 0B 4D)
21:43:29.700 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=23, apiId=44 81, data=FE 17 44 81 39 5C 06 00 EB 5D 01 01 00 00 00 8B 43 5D 00 00 03 01 62 02 EB 5D 0B 4D, checksum=4D, error=false
21:43:29.701 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=24043/1, destinationAddress=0/1, profile=0104, cluster=6, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=01 62 02]
21:43:29.702 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=98, commandId=2]
21:43:29.703 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ToggleCommand [On/Off: 24043/1 -> 0/1, cluster=0006, TID=62]
21:43:29.704 [DEBUG] [.converter.ZigBeeConverterSwitchOnoff] - D0CF5EFFFEF2B6E8: ZigBee command receiveds ToggleCommand [On/Off: 24043/1 -> 0/1, cluster=0006, TID=62]
`

Level UP
`
21:44:37.061 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1B 44 81 39 5C 08 00 EB 5D 01 01 00 0D 00 CD 79 60 00 00 07 01 65 06 00 2B 05 00 EB 5D 0B 2A)
21:44:37.062 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=27, apiId=44 81, data=FE 1B 44 81 39 5C 08 00 EB 5D 01 01 00 0D 00 CD 79 60 00 00 07 01 65 06 00 2B 05 00 EB 5D 0B 2A, checksum=2A, error=false
21:44:37.063 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=24043/1, destinationAddress=0/1, profile=0104, cluster=8, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=01 65 06 00 2B 05 00]
21:44:37.064 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=101, commandId=6]
21:44:37.066 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: StepWithOnOffCommand [Level Control: 24043/1 -> 0/1, cluster=0008, TID=65, stepMode=0, stepSize=43, transitionTime=5]
`

Level Down
`
21:44:50.416 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1B 44 81 39 5C 08 00 EB 5D 01 01 00 0F 00 CE 1C 61 00 00 07 01 66 02 01 2B 05 00 EB 5D 0B 49)
21:44:50.418 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=27, apiId=44 81, data=FE 1B 44 81 39 5C 08 00 EB 5D 01 01 00 0F 00 CE 1C 61 00 00 07 01 66 02 01 2B 05 00 EB 5D 0B 49, checksum=49, error=false
21:44:50.419 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=24043/1, destinationAddress=0/1, profile=0104, cluster=8, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=01 66 02 01 2B 05 00]
21:44:50.420 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=102, commandId=2]
21:44:50.420 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: StepCommand [Level Control: 24043/1 -> 0/1, cluster=0008, TID=66, stepMode=1, stepSize=43, transitionTime=5]
`

Left
`
21:45:27.809 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1D 44 81 39 5C 05 00 EB 5D 01 01 00 1A 00 3B E5 62 00 00 09 05 7C 11 67 07 01 01 0D 00 EB 5D 0B 19)
21:45:27.810 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=29, apiId=44 81, data=FE 1D 44 81 39 5C 05 00 EB 5D 01 01 00 1A 00 3B E5 62 00 00 09 05 7C 11 67 07 01 01 0D 00 EB 5D 0B 19, checksum=19, error=false
21:45:27.812 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=24043/1, destinationAddress=0/1, profile=0104, cluster=5, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=05 7C 11 67 07 01 01 0D 00]
21:45:27.813 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=true, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=4476, sequenceNumber=103, commandId=7]
21:45:27.814 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - No command type found for CLUSTER_SPECIFIC_COMMAND, cluster=5, command=7, direction=CLIENT_TO_SERVER
21:45:27.815 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - Incoming message did not translate to command.
`

Right
`
21:45:40.517 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1D 44 81 39 5C 05 00 EB 5D 01 01 00 0A 00 5D 80 63 00 00 09 05 7C 11 68 07 00 01 0D 00 EB 5D 0B 05)
21:45:40.519 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=29, apiId=44 81, data=FE 1D 44 81 39 5C 05 00 EB 5D 01 01 00 0A 00 5D 80 63 00 00 09 05 7C 11 68 07 00 01 0D 00 EB 5D 0B 05, checksum=05, error=false
[... cut ...] 104, commandId=7]
21:45:40.523 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - No command type found for CLUSTER_SPECIFIC_COMMAND, cluster=5, command=7, direction=CLIENT_TO_SERVER
21:45:40.525 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - Incoming message did not translate to command.
`

In additional it is able to differ between press and hold of buttons - different log output.

But it does not produce any channel events in Openhab. How could be those buttons events propagated to Openhab?

goddib, 2019-05-13:
@jhron have you made any progress? I suppose the good news here is that openHAB is receiving things.

I'm also trying to get the TRADFRI remote to work with the zigbee binding and have yet to make progress. Can the debug messages be used in rules?

TheLexus, 2019-06-06:
With the information from above and the tutorial at https://github.com/openhab/org.openhab.binding.zigbee/pull/284#issuecomment-442210447 it should be possible. Unfortunately I cannot do it atm, maybe I try it in a few weeks. 

What is missing, and maybe @jhron can add it, the remote sends different commands on long presses (at least the dimmer buttons do). To create a complete configuration we need that information too.

TheLexus, 2019-06-06:
@jhron  I tried to create the file based on the tutorial but there are a few problems. First, Left and Right seems to be the same. At least clusterId and CommandId are the same at the log excerpt above.

And I need the vendor and modelid (should be visible at the paperui!).

goddib, 2019-06-06:
I've tried to create the necessary files (without longpress and still accepting that left/right will be the same) but did not manage to compile them/load them into openhab.

discovery.txt
`philips_sml001,vendor=Philips,modelId=SML001
philips_rwl021,vendor=Philips,modelId=RWL021
smartthings_motionv4,vendor=SmartThings,modelId=motionv4
bitron-video-902010-23,vendor=Bitron Home,modelId=902010/23
bitron-video-av2010-34,vendor=Bitron Video,modelId=AV2010/34
xiaomi_lumisensorht,modelId=lumi.sensor_ht
xiaomi_lumisensor-motion,modelId=lumi.sensor_motion
xiaomi_lumiremoteb286acn01,modelId=lumi.remote.b286acn01
xiaomi_lumisensor86sw2,modelId=lumi.sensor_86sw2
xiaomi_lumisensor-switchaq2,modelId=lumi.sensor_switch.aq2
innr-rc-110,vendor=innr,modelId=RC 110
osram-switch-4x-eu,vendor=OSRAM,modelId=Switch 4x EU-LIGHTIFY
ikea-tradfri-remote,vendor=IKEA of Sweden,modelId=TRADFRI remote control`

ikea-tradfri-remote.xml
`<?xml version="1.0" encoding="UTF-8"?>
<thing:thing-descriptions xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
	xmlns:thing="http://eclipse.org/smarthome/schemas/thing-description/v1.0.0" bindingId="zigbee"
	xsi:schemaLocation="http://eclipse.org/smarthome/schemas/thing-description/v1.0.0 http://eclipse.org/smarthome/schemas/thing-description-1.0.0.xsd">
	<thing-type id="ikea-tradfri-remote" listed="false">
		<label>IKEA Tradfri Remote</label>
		<channels>
			<channel id="buttonCentral" typeId="system.button">
				<label>Central Button</label>
				<description>Central On/Off Button</description>
				<properties>
					<property name="zigbee_endpoint">1</property>
					<property name="zigbee_shortpress_cluster_id">0x0006</property>
					<property name="zigbee_shortpress_command_id">0x02</property>
				</properties>
			</channel>
			<channel id="buttonTop" typeId="system.button">
				<label>Top Button</label>
				<description>Top Button (Brightness Up)</description>
				<properties>
					<property name="zigbee_endpoint">2</property>
					<property name="zigbee_shortpress_cluster_id">0x0008</property>
					<property name="zigbee_shortpress_command_id">0x06</property>
				</properties>
			</channel>
			<channel id="buttonBottom" typeId="system.button">
				<label>Bottom Button</label>
				<description>Bottom Button (Brightness Down)</description>
				<properties>
					<property name="zigbee_endpoint">3</property>
					<property name="zigbee_shortpress_cluster_id">0x0008</property>
					<property name="zigbee_shortpress_command_id">0x02</property>
				</properties>
			</channel>
			<channel id="buttonLeft" typeId="system.button">
				<label>Left Button</label>
				<description>Left Button (Arrow Left)</description>
				<properties>
					<property name="zigbee_endpoint">4</property>
					<property name="zigbee_shortpress_cluster_id">0x05</property>
					<property name="zigbee_shortpress_command_id">0x07</property>
				</properties>
			</channel>
			<channel id="buttonRight" typeId="system.button">
				<label>Right Button</label>
				<description>Right Button (Arrow Right)</description>
				<properties>
					<property name="zigbee_endpoint">5</property>
					<property name="zigbee_shortpress_cluster_id">0x05</property>
					<property name="zigbee_shortpress_command_id">0x07</property>
				</properties>
			</channel>
		</channels>
		<config-description>
			<parameter name="zigbee_macaddress" type="text" readOnly="true" required="true">
				<label>MAC Address</label>
			</parameter>
		</config-description>
	</thing-type>
</thing:thing-descriptions>`

Maybe this helps? I'll gladly test any implementation if somebody tells me the steps I need to take to load this!

bkummel, 2019-06-10:
Hi, I do have an IKEA Tradfri remote and an Ember-based Zigbee stick. I'm willing to help figuring out the right codes for the buttons. However, when setting the log levels to DEBUG, I get such an overwhelming amount of logging, that it is difficult to filter out what messages could be from the Tradfri remote. Is there a smarter way to do this?

cdjackson, 2019-06-10:
You can use the logviewer on my website.

https://www.cd-jackson.com/index.php/openhab/zigbee-log-viewer

bartkummel, 2019-07-09:
I have added my Tradfri remote as a Thing and was trying to get some info from the debug logging. However, I don't see any received `CLUSTER_SPECIFIC_COMMAND` in my log. I searched my logfiles with grep using several keywords, such as:
- `CLUSTER_SPECIFIC_COMMAND` - I only find transmitted commands of this type
- the `zigbee_networkaddress` of my remote - I only see messages of the device being discovered and added to a neighbour table.
- `ToggleCommand` - I see nothing at all.

Of course, I pressed a lot of buttons short and long before searching the logs. And, as mentioned, I do see messages of the device being discovered. When I accidentally removed the battery and placed it again, I see the device is rejoining. So the remote is communicating with OpenHAB, but I don't see any commands being received by OpenHAB that I can relate to button presses. 

Is there any additional action I have to do before being able to see those messages in the log? @jhron - what did you do to get the log messages you posted above?

jhron, 2019-07-09:
> Is there any additional action I have to do before being able to see those messages in the log? @jhron - what did you do to get the log messages you posted above?

I did increase log level of following:
```
log:set debug org.openhab.binding.zigbee
log:set debug com.zsmartsystems.zigbee
log:set info com.zsmartsystems.zigbee.dongle.ember.internal.ash
```
Unfortunately I am quite busy last few months so I had to "abandon" activity in this matter :( at the moment.

bartkummel, 2019-07-10:
@jhron That's exactly what I did. There must be something else...

jhron, 2019-07-10:
> @jhron That's exactly what I did. There must be something else...

Then try change info to debug here:
`log:set debug com.zsmartsystems.zigbee.dongle.ember.internal.ash`

I can not try it by my self because at the moment because I had to (to move on) switch from native openhab binding to zigbee2mqtt (here Ikea button works ok) and I am parsing mqtt topics by rules in openhab.

bartkummel, 2019-07-31:
@jhron Tried that, but it only logs "ASH frames". No idea what those are, but they don't look like the examples you posted on April 13th.

petero-dk, 2019-11-15:
I anyone still working or interested in getting this to work? I have just gotten the controller paired with openhab through the zigbee binding

I have a telegesis dongle, and set the debugging level with
```
 log:set debug org.openhab.binding.zigbee
 log:set debug com.zsmartsystems.zigbee
 log:set info com.zsmartsystems.zigbee.dongle.telegesis.internal.ash
```

The log file is attached 
[openhab.log](https://github.com/openhab/org.openhab.binding.zigbee/files/3852715/openhab.log)

What can I do to help get this to work?

bakkerv, 2019-12-17:
Very interested in getting this to work. I've bought the new Tradfri wireless dimmer (704.085.95), which is  paired with OpenHAB.
I do get On/Off events (short pressed) and would like to use the dimmer functionality the be used on OpenHAB for custom dimming.

Some zigbee data:
Up longpress (On button):
```20:47:23.850 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: MoveWithOnOffCommand [Level Control: 60055/1 -> 0/1, cluster=0008, TID=6C, moveMode=0, rate=83]```
And Release:
`20:47:26.086 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: Stop2Command [Level Control: 60055/1 -> 0/1, cluster=0008, TID=6D]`

Down longpress (Off button):
`20:48:04.174 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: MoveCommand [Level Control: 60055/1 -> 0/1, cluster=0008, TID=6F, moveMode=1, rate=83]`
And release:
`20:48:09.367 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: Stop2Command [Level Control: 60055/1 -> 0/1, cluster=0008, TID=70]`

Ideally, I would like to use this device similar to be the rawrocker described in the items (https://www.openhab.org/docs/configuration/items.html), which I know from my enocean rockers.
My EnOcean rocker sends different signals when a button is pressed and releases, and by using the rawrocker-to-dimmer profile I can directly use the rocker to a dimmer item.
I know the zigbee dimmer's behavior is a bit more difficult, since it sends a On/Off command on a short press and different command on a long press. This does not match the `rawbutton`, `button` or `rawrocker` channel types and corresponding triggers.

I am more than willing to develop this myself, but have no clue where to start :)

cdjackson, 2019-12-17:
Please provide all 3 log entries that are received for each command - not just the RX CMD entry.

bakkerv, 2019-12-17:
On:
`21:39:31.801 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=60055/1, destinationAddress=0/1, profile=0104, cluster=6, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=01 02 01]
21:39:31.816 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=2, commandId=1]
21:39:31.829 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: OnCommand [On/Off: 60055/1 -> 0/1, cluster=0006, TID=02]
`

Off:
`21:39:47.822 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=60055/1, destinationAddress=0/1, profile=0104, cluster=6, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=01 03 00]
21:39:47.841 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=3, commandId=0]
21:39:47.855 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: OffCommand [On/Off: 60055/1 -> 0/1, cluster=0006, TID=03]
`

Long On:
`21:43:50.442 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=60055/1, destinationAddress=0/1, profile=0104, cluster=8, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=01 18 05 00 53]
21:43:50.452 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=24, commandId=5]
21:43:50.463 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: MoveWithOnOffCommand [Level Control: 60055/1 -> 0/1, cluster=0008, TID=18, moveMode=0, rate=83]
21:43:52.184 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=60055/1, destinationAddress=0/1, profile=0104, cluster=8, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=01 19 07]
21:43:52.195 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=25, commandId=7]
21:43:52.212 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: Stop2Command [Level Control: 60055/1 -> 0/1, cluster=0008, TID=19]
`

Long Off:
`
21:44:12.090 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=60055/1, destinationAddress=0/1, profile=0104, cluster=8, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=01 1C 01 01 53 00 00]
21:44:12.100 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=28, commandId=1]
21:44:12.114 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: MoveCommand [Level Control: 60055/1 -> 0/1, cluster=0008, TID=1C, moveMode=1, rate=83]
21:44:13.432 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=60055/1, destinationAddress=0/1, profile=0104, cluster=8, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=01 1D 07]
21:44:13.450 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=29, commandId=7]
21:44:13.464 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: Stop2Command [Level Control: 60055/1 -> 0/1, cluster=0008, TID=1D]
`

Silent-Hunter, 2020-11-14:
I too would be willing to help test things.

NightDragon1, 2021-09-14:
Has there been any update on this item, maybe for OH3?

nailyk-fr, 2021-09-26:
I am discovering zigbee as well and also have this device (and others switches, CWS bulbs, TRV, etc.) that I would like to work with direct zigbee binding in openhab. 
Can you please point in the right direction into the code in order to create the proper channels for the button ? 
I have seen the https://github.com/openhab/org.openhab.binding.zigbee/pull/284 which might already help but any advice is welcome :slightly_smiling_face: 

Thanks in advance !