openhab/org.openhab.binding.zigbee #359: Centralite 3310-G Temperature/Humidity, Humidity on odd cluster

izzy84075, 2019-01-13:
I just picked up some Centralite 3310-G Temperature/Humidity sensors, as the Xiaomi ones I had previously were very spotty. These ones seem more reliable, but they've got one problem... The Humidity reading is on a nonstandard cluster. As far as I can tell from skimming through [SmartThings device handlers](https://github.com/SANdood/SmartThings-Fixed/blob/master/SmartSense-Temp-Humidity%20Sensor%20Custom#L158), the Humidity reading is on cluster 0xFC45, Attribute 0, and is in the normal format for relative humidity.

I'm not quite sure if it's the ZCL standard Relative Humidity cluster, just on a different number, though. None of the SmartThings device handlers try to handle anything except attribute 0.

Is there any way to add a cluster-redirect or something, to tell it that this 0xFC45 cluster can be treated like a 0x0405 cluster?

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |  GoControl HUSBZB-1                                                                              |
| openHAB version  | 2.4.0     |
| Hardware         |   Way overkill                                                                             |
| Memory           |    8GB                                                                            |
| Java version     |      Zulu-8                                                                          |
| Devices          | Centralite 3310-G                                   |

izzy84075, 2019-01-14:
https://drive.google.com/open?id=1k38QcCxAOiHMbX1tx8Sy8m3PveF6Uwb4 Here's the entry in the Zigbee XML. Doesn't appear to show this odd cluster at all, not sure if that's a shortcoming in the device itself or the binding not looking for it...

hsudbrock, 2019-01-15:
Hi @izzy84075 ; a cluster with ID `0xfc45` is a manufacturer-specific cluster, for which there is currently not much support in the underlying zsmartsystems ZigBee library. I.e., just treating this cluster like the `0x0405` cluster will not work directly, as the device only reacts on commands that contain Centralite's manufacturer code.

I am currently working on adding support for such clusters to the underlying library; I will try to make the Centralite device that you mention here one application scenario for this here in the binding :)

hsudbrock, 2019-01-15:
> Doesn't appear to show this odd cluster at all, not sure if that's a shortcoming in the device itself or the binding not looking for it...

Currently, the underlying ZigBee library will only list clusters that it 'knows* how to deal with, that is the reason that you cannot see the manufacturer-specific humidity cluster of the device.

izzy84075, 2019-01-15:
Ah, I hadn't caught the manufacturer code dependency... It's hard trying to catch up on all the details of Zigbee, haha!

If there's anything I can do to help, please let me know! I'm not great at Java, but I'm always looking to learn. I do a lot of embedded work, so Zigbee is definitely something I'd like to learn more about.