openhab/org.openhab.binding.zigbee #380: Changing of Bridge should make Things go to OFFLINE.GONE

triller-telekom, 2019-02-05:
PaperUI supports the changing of the bridge for a thing.

With this feature a user can change for example from a Telegesis zigbee dongle to an Ember dongle and select the new bridge on the previously paired devices.

According to the current code those devices will initialize and set themselves to ONLINE although they are not even paired to the new dongle yet. This is because all things will be set to ONLINE if there is no Exception during initialization.

The correct status for such devices would be OFFLINE.GONE.

With the status OFFLINE.GONE it is possible for the framework to detect that the device is not working and could offer a possibility to re-pair the device without loosing all its (thing) configuration, names, etc.

One way to implement this would be:

On pairing a new device, the IEEE address of the bridge will be set as a property on the thing. Once the thing initializes, it compares the current `getBridge().getIEEEAddress()` with the one in its property and if there is a mismatch, the device can be set to OFFLINE.GONE.

I think setting the device to OFFLINE.GONE if we cannot talk to it anymore would be an improvement in the status handling of zigbee devices.

triller-telekom, 2019-02-08:
@cdjackson May I ask you about your opinion on this issue?

cdjackson, 2019-02-08:
Sorry - there are a lot things to look at at the moment as I have been away on holiday. I will take a look at this next week.

cdjackson, 2019-02-17:
> On pairing a new device, the IEEE address of the bridge will be set as a property on the thing. Once the thing initializes, it compares the current `getBridge().getIEEEAddress()` with the one in its property and if there is a mismatch, the device can be set to OFFLINE.GONE.

I guess this is ok - although it does feel wrong that we need to record the address of the bridge just to detect if the bridge changed, when it is changed through ESH in the first place. Maybe ESH should be managing this better - ie if you change the bridge on a thing, then it should set some state/flag/property on the thing so that bindings know this as this clearly isn't a ZigBee specific problem.

triller-telekom, 2019-02-20:
Thank you for your feedback.

Yes, I agree with you, that handling the changing of a bridge would be best done inside the ESH framework somehow, would be cleaner. But this would also be a larger change to the framework.

So would you be fine with a solution I sketched above? And if there is a proper solution in the framework, we could still swap to that one?

If so, I would provide a PR which sets the thing to OFFLINE.GONE if the bridge IEEE address changes.

cdjackson, 2019-02-20:
Why is it a larger change in the framework? Isn't it even easier there as it just needs to check when saving the thing if the bridge changes, and if so set a flag?

triller-telekom, 2019-02-20:
Because in the framework this should be a general handling of all possible cases and side effects of a bridge change. Maybe the framework shouldn't even try to initialize the handler of a thing at all if its bridge has changed. I wouldn't just introduce a "dumb flag" called "bridgeChanged=true", but rather expect the framework to provide a more general and possibly extensible mechanism to handle this.

Also the core framework ESH is dead: https://www.eclipse.org/lists/smarthome-dev/msg00195.html

So I guess if the framework should take care of it, it should be somehow implemented in the new openhab-core framework.

That's why I am saying if you are OK with the solution I wrote in the beginning of this thread, I would propose to go that road. WDYT?

cdjackson, 2019-02-20:
> 
> Also the core framework ESH is dead: https://www.eclipse.org/lists/smarthome-dev/msg00195.html <https://www.eclipse.org/lists/smarthome-dev/msg00195.html>
> 

So because the core framework is dead, we now need to put all these sorts of functions into every binding? I really hope that is not where this is all heading as that’s a disaster. I assumed that core framework development has moved to openHAB, and as this is an openHAB binding, it should be handled in the core framework still?

Or am I wrong and it’s completely dead?

triller-telekom, 2019-02-20:
You are not wrong, it is not dead. Openhab-core is still alive, though I am not sure if the zigbee binding is already building against it or still against ESH...

To stay on topic: In https://github.com/openhab/org.openhab.binding.zigbee/issues/380#issuecomment-464449046 you said its OK to implement the sketched solution. Now you state the opposite.

Do you thing that this check for a property so annoying in the binding, that from sunday to today you have changed your mind?

triller-telekom, 2019-02-20:
Thinking about this again: In the zigbee case a change of a bridge leads to a non-functional thing.

But for other bindings this might not necessarily be true. So the core framework cannot set the thing to OFFLINE.GONE in all cases for all bindings, because there might be a binding where the thing will still be ONLINE.

hsudbrock, 2019-02-20:
>  though I am not sure if the zigbee binding is already building against it or still against ESH...

@triller-telekom The binding is building against the openhab core framework

cdjackson, 2019-02-20:
> To stay on topic: In #380 (comment) <https://github.com/openhab/org.openhab.binding.zigbee/issues/380#issuecomment-464449046> you said its OK to implement the sketched solution. Now you state the opposite.

I don’t really state the opposite - I just think this should be dealt with in the core. The core allows this function which can’t easily be used, and now to work around these core issues we are adding things into the binding that also won’t be used. I know that Qivicon may want to use this, but I doubt that normal OH users can due to the issues with changing coordinators and having to resolve the mess of keys and counters.

triller-telekom, 2019-02-20:
> I doubt that normal OH users can due to the issues with changing coordinators and having to resolve the mess of keys and counters.

Right, and this is why if a user changes the bridge, the thing should be OFFLINE.GONE to reflect that it cannot be used anymore.

Did you see my other comment regarding the general implementation in the core framework? https://github.com/openhab/org.openhab.binding.zigbee/issues/380#issuecomment-465613770

cdjackson, 2019-02-20:
> Did you see my other comment regarding the general implementation in the core framework? #380 (comment) <https://github.com/openhab/org.openhab.binding.zigbee/issues/380#issuecomment-465613770>

Yes, I guess it needs a bit more thought before implementing anything.

triller-telekom, 2019-02-21:
A general solution in the framework definitely needs more thought, I agree.

But what about the **feature for zigbee**, that users see that if they change the bridge that their things are in state OFFLINE.GONE?

hsudbrock, 2019-02-28:
I had a chat on using the `OFFLINE.GONE` thing status detail in the ZigBee binding with @hanneshofmann yesterday. We came to the conclusion that one should probably use this status detail when there is no node in the ZigBee network for the thing (rather than tying this status detail to an ESH operation like "changing the bridge"). I think that this captures as good as possible what is [documented for this status detail](https://www.eclipse.org/smarthome/documentation/concepts/things.html#status-details):
> The Thing has been removed from the bridge or the network to which it belongs and is no longer available for use. "

This would cover the case described by @triller-telekom (and even improve it): After changing the bridge, the status detail would be set to `OFFLINE.GONE` (unless the device has already joined the network of the new bridge, in which case the `ZigBeeNetworkManager` of the new bridge would already provide a node for the thing and this status detail would correctly not be set).

This could also cover other cases where the device has left the network. E.g., some devices send a `Leave Request` command when they are being factory reset. If the ZSS library would remove the corresponding node (which it currently does not, at least in my test), then this could also result in status detail `OFFLINE.GONE`.

Considering implementation, one could set the status detail to `GONE` [here](https://github.com/openhab/org.openhab.binding.zigbee/blob/0cd602d8e4e9b675a63e9a9bfbe20c35ee99e9dd/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/handler/ZigBeeThingHandler.java#L222) and [here](https://github.com/openhab/org.openhab.binding.zigbee/blob/0cd602d8e4e9b675a63e9a9bfbe20c35ee99e9dd/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/handler/ZigBeeThingHandler.java#L768) in the `ZigBeeThingHandler`.

WDYT, @cdjackson and @triller-telekom ?

triller-telekom, 2019-02-28:
Sounds like a good approach to me, since it even covers more cases than just changing the bridge.

cdjackson, 2019-02-28:
This is fine with me.

Note that the ZSS does not remove nodes from the network - this needs to be done at application level. This is because when there is a rejoin, we see a "Node Left" notificaiton, and then a "Node Joined" - if the node is deleted, then it needs to be rediscovered. Given that this is an issue for EDs, and that most EDs can't be rediscovered, it leaves devices ununsable when they are really still on the network.

Maybe there are better ways to handle this, but for now devices are not removed for this reason.

PR #401 (2019-03-01): Set thing status to offline/gone if there is no node for a thing
This resolves #380 

As discussed in #380 , this changes the thing status to `OFFLINE` with detail `GONE` if (a) there is no node for the thing on node initialization in the thing handler, and (b) if the thing handler learns that the node was removed (and removal was not triggered by the openHAB core, in which case the thing status would be `REMOVING`).

As you mentioned [here](https://github.com/openhab/org.openhab.binding.zigbee/issues/380#issuecomment-468263410), Chris, case (b) should not happen currently, as the ZSS library  [... cut]
closed 2019-03-01 by cdjackson