openhab/org.openhab.binding.zigbee #382: Adding support for Xiaomi Water Leak Sensor

iva-k, 2019-02-06:
Hello!

I've been trying to add support for Xiaomi Water Leak Sensor SJCGQ11LM after finding that it didn't have any Items added when I pair it with the CC2531 and add the generic Thing under ZigBee binding like other items I've paired before so it was pretty useless like that.

I managed to get it working after spending some time scouring the github by adding the ias_water channel under cluster 1280, so that bit is working fine it seems!

However i see that the sensor should be providing some battery alarm and voltage. I tried adding the battery_alarm and  battery_voltage channels but they never recieve any data from the sensor. I've tried setting the cluster to 0 (this seems to be the correct one), 1, 3. The most i have gotten out of it is that sometimes voltage changes from null to 0.0V

Any idea what i might be doing wrong? I have attached the xml file generated for the sensor in zigbee folder when it is added as a Thing.

Thanks,
Ivan

[zigbee-water-sensor.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/2838408/zigbee-water-sensor.txt)

Hunv, 2019-08-06:
Just saw this issue and like to link it to the same topic at the OpenHab Community: https://community.openhab.org/t/zigbee-binding-add-xiaomi-aqara-water-leak-sensor/76529/9

achille2k, 2019-08-15:
1. Please append this string in : \src\main\resources\discovery.txt
xiaomi_lumisensor-wleakaq1,vendor=LUMI,modelId=lumi.sensor_wleak.aq1
2. unzip this attachment
[lumisensor-wleakaq1.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/3505646/lumisensor-wleakaq1.zip)
And add file lumisensor-wleakaq1.xml  to folder  \src\main\resources\ESH-INF\thing\xiaomi\
3. Build and deployment

Vegardbeider, 2021-09-30:
Any progress here? I just picked up this sensor. I was able to add it once and it said it was online but not updating any of the channenls. But now I cant get it to discover at all. I see there is a Xaomi Water Leak sensor to add manually do I need to do that? What information do I need to provide than?

cdjackson, 2021-09-30:
This issue is still open, so I don’t think there has been any updates.

iva-k, 2022-12-21:
Moved to Zigbee2Mqtt and never looked back 😊

closed 2022-12-21 by iva-k

magx2, 2024-01-12:
@iva-k Can you link the issue?