openhab/org.openhab.binding.zigbee #398: Binding may try to initialise ZigBee when initialisation is in progress

cdjackson, 2019-02-27:
As discussed [here](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/518#issuecomment-466916649) there is a bug in the binding where the reconnect job may perform a reconnection while the initialisation is already in progress.

@triller-telekom can you take a look at this?

triller-telekom, 2019-03-01:
The problem I see in the linked thread is the following:

The reconnectJob is started whenever the ZigbeeTransport state changes to `OFFLINE`. This changing to `OFFLINE` also happens if someone calls `ZigbeeNetworkManager.startup(true)`.

We could argue that this "someone" should not be allowed to call `ZigbeeNetworkManager.startup(true)` and leave the initialization of the `ZigbeeNetworkManager` in the hands of the `ZigbeeCoordinatorHandler`.

But then we would need a possibility to tell the `ZigbeeCoordinatorHandler` via a `public` method that he should restart the `ZigbeeNetworkManager`. A possible use case would be an automatic reconfiguration after a restore from a backup. If the `ZigbeeCoordinatorHandler` is in charge of restarting the `ZigbeeNetworkManager`, it can prevent the reconnectJob from being started because it knows the ZigbeeTransport state is `OFFLINE` because it has trigged a restart. Lets call this "binding-only solution".

One other approach could be that the `ZigbeeNetworkManager` offers a callback registration mechanism where the `ZigbeeCoordinatorHandler` can subscribe itself as a listener and gets informed about the state of the `ZigbeeNetworkManager` and not of the ZigbeeTransport state. This way the `ZigbeeCoordinatorHandler` would know that someone had triggered a restart of the `ZigbeeNetworkManager` and can ignore an OFFLINE until the restart has been finished. This would be a ZSS+binding solution, sicne both repositories have to be touched.

@cdjackson WDYT?

cdjackson, 2019-03-01:
> One other approach could be that the `ZigbeeNetworkManager` offers a callback registration mechanism where the `ZigbeeCoordinatorHandler` can subscribe itself as a listener and gets informed about the state of the `ZigbeeNetworkManager` and not of the ZigbeeTransport state

I think that while the callback is looking at the Transport state, it is actually not the state of the transport - it is the state of the manager. It is updated when the initialisation starts, or fails, and when the manager goes online.  I think that this is what you are asking for?

Probably this should not use the ```ZigBeeTransportState``` as the enumeration here, but historically that is how it was.

Please take a look to see if that does what you are asking - if so, maybe we look to refactor this to ```ZigBeeNetworkState``` - there are a few breaking changes flowing into 1.2 of the library, so we could also look to incorporate this at the same time.

triller-telekom, 2019-03-01:
If the `ZigbeeTransportState` here https://github.com/openhab/org.openhab.binding.zigbee/blob/master/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/handler/ZigBeeCoordinatorHandler.java#L875 is the state of the `ZigbeeNetworkManager`, then this is the state I am looking for, right.

However, the `ZigBeeCoordinatorHandler` receives this OFFLINE is someone restarts the ZigbeeNetworkManager and then starts the reconnectJob.

So you suggest that if one wants to reinitialize the networkmanger we should let the coordinator do that, so it knows that the intermediate OFFLINE state is due to a reinitialisation of the networkmanger which it just had triggered?

Then we would need the public method on the coordinator to offer a network restart, would this be Ok?

cdjackson, 2019-03-01:
> However, the `ZigBeeCoordinatorHandler` receives this OFFLINE is someone restarts the ZigbeeNetworkManager and then starts the reconnectJob.

I'm not sure I follow this point? When you start the network manager, the state is ```UNINITIALIZED```. When you start the initialisation, the state in ```INITIALIZING```. When the initialisation finishes (after the ```startup``` method is called it will go to ```ONLINE```. Can you describe what you are looking for?

> Then we would need the public method on the coordinator to offer a network restart, would this be Ok?

Sorry - I don't understand this either. I assume here you are talking about the CoordinatorHandler? Why does it need a public method to restart the network? You already have this sort of function in your restart timer implementation - I think that all that is needed is to not restart if initialisation is already in progress? You either need to check the state of the network and not do another reinitialisation if the state is INITIALIZING, or, you need to remember in your reinitialisation code in the coordinator handler that this is called already and not do it again?

Or maybe I'm missing something (sorry if that's the case).

triller-telekom, 2019-03-01:
> I'm not sure I follow this point? When you start the network manager, the state is `UNINITIALIZED`. When you start the initialisation, the state in `INITIALIZING`. When the initialisation finishes [...]

That is what I am assuming as well. But during the `formNetwork` call the ZigbeeTransportState changes to OFFLINE and the Coordinator is notified about this. I am still totally confused about the state of the NetworkManager and the underlying ZigbeeTransport. Are they really the same? I don't think so.

The CoordinatorHandler is a listener to the ZigbeeTransportState, which I think is different from the ZigbeeNetworkManagerState.

This is the first problem we should solve here, otherwise there will be even more confusion.

cdjackson, 2019-03-01:
> But during the `formNetwork` call the ZigbeeTransportState changes to OFFLINE

So we could stop this. We could make a transition of the network state to OFFLINE while it is INITIALIZING an illegal transition in this case.

> Are they really the same? I don't think so.

No. And a user (ie user of the library - so developer) should not need to care about the transport really. This should all be wrapped up by the overall network state.

> The CoordinatorHandler is a listener to the ZigbeeTransportState,

Well, the enumeration that is used is the TransportState, but as I said above, this is just the enumeration. It should use a separate enum to avoid confusion, but if you look at the listener, it is ```ZigBeeNetworkStateListener``` and it is meant to be the network state.

---

If we prevent the network state changing to OFFLINE as a result of the transport going OFFLINE during INITIALIZATION, does that help you? We would still allow transition to OFFLINE from INITIALISATING, but only if the overall initialisation sequence fails at the network manager level.

triller-telekom, 2019-03-01:
Thank you for the explanations, now I got it. And we really should adjust the enum naming to avoid further confusion of others!

> If we prevent the network state changing to OFFLINE as a result of the transport going OFFLINE during INITIALIZATION, does that help you?

Yes it does help!

> We would still allow transition to OFFLINE from INITIALISATING, but only if the overall initialisation sequence fails at the network manager level.

Of course, makes sense.

cdjackson, 2019-03-01:
I have raised an issue here -: https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/554 and will look at this over the next couple of days.

Thanks.

cdjackson, 2019-08-10:
I believe that this is now resolved as we changed the way the network state is handled in ZSS so closing.

closed 2019-08-10 by cdjackson