openhab/org.openhab.binding.zigbee #399: Speedup discovery by reading multiple attributes with single request

hsudbrock, 2019-02-28:
Currently, when discovering a new device, the discovery queries several attributes each with a single `Read Attribute` command (see [here](https://github.com/openhab/org.openhab.binding.zigbee/blob/0cd602d8e4e9b675a63e9a9bfbe20c35ee99e9dd/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/discovery/ZigBeeNodePropertyDiscoverer.java#L80)).

Since recently, the ZSS ZigBee library supports reading multiple attributes with a single request (see [here](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/530)).

It would be nice to use this feature for speeding up the discovery process by requesting all the attributes at once (with a fallback to using multiple `Read Attribute` requests for devices that cannot put all attribute values into a single response).

I could implement this, @cdjackson - so in case you already started with this please give me a ping :)

cdjackson, 2019-02-28:
I've not looked at implementing this in the property discoverer - but have started implemented the same thing in the polling consolidation...

Feel free to implement - I hope to get the new transaction manager merged in the coming days and will then do an update for the OH TP with these features.

hsudbrock, 2019-02-28:
Sounds good, I will implement this then...

PR #464 (2019-07-28): Read multiple attributes at once during discovery
This replaces #400 and resolves #399 by using a multi-attribute read request in the `ZigBeeNodePropertyDiscoverer`. I did some smaller refactorings in that class as well, and used separate commits for those to ease the review.

Please note that this PR needs the new ZSS version before it can be merged (the new version that supports multi-attribute-read commands).

The basic idea is to do a multi-attribute read first, and then read each attribute separately from the ZSS layer as is already done before this PR. If the multi-attribute read succeed [... cut]
closed 2019-07-28 by cdjackson