openhab/org.openhab.binding.zigbee #404: Make device leave network if unapproved discovery result is removed

hsudbrock, 2019-03-06:
During discovery (cf. [`ZigBeeDiscoveryService`](https://github.com/openhab/org.openhab.binding.zigbee/blob/master/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/discovery/ZigBeeDiscoveryService.java)), a device may join the ZigBee network, which results in a discovery result.

The "normal" case is that the discovery result is approved, thereby creating a thing. This issue is about the not so "normal" case that the discovery result is never approved, but removed by the user (e.g., because the user decides against creating the thing, for whatever reason).

Currently, this will remove the discovery result, but the device is still in the network, and the ZSS library still manages a `Node` for the device. In consequence, when another discovery is started, a discovery result is again created for the device.

I think that, if a discovery result is removed and there is no thing corresponding to the result, then the binding should ask the device to leave the network (and remove the node explicitly if the device does not answer to the `Leave` command).

From the implementation side, I would handle this with an `InboxListener`, along the following lines:
```java
    @Override
    public void thingRemoved(Inbox source, DiscoveryResult result) {
        ThingUID thingUID = result.getThingUID();
        ThingUID bridgeUID = result.getBridgeUID();
        if (isZigBeeBridge(bridgeUID) && noThingExistsForDevice(thingUID)) {
            getHandler(bridgeUID).removeDeviceFromNetwork(thingUID.getId());
        }
    }
```

WDYT, @cdjackson - do you have any thoughts about including such a feature in the binding?

cdjackson, 2019-03-06:
I’m not keen on this - it will prevent some use cases from working.

For example, if someone wanted to include a remote control into a network, or, a secondary controller (eg another HA box), your proposal here (if I understand correctly) would remove this if it didn’t create a thing. Assume we have a STB that has a ZigBee interface, that you wanted to use as a controller to control your lights. This STB will not be included as a thing in the network, so you would remove it. But it still needs to participate in the network, so we prevent this use case.

(I should add - I have exactly this use case here).

Or am I misunderstanding something?

hsudbrock, 2019-03-06:
> For example, if someone wanted to include a remote control into a network, or, a secondary controller (eg another HA box), your proposal here (if I understand correctly) would remove this if it didn’t create a thing. 

It would remove the device from the network if one would remove the discovery result without creating a thing, correct.

I see your use case (which I did not see before), thanks for letting me know! I think that it therefore is not useful to include this in the binding. Solutions can still add their own `InboxListener` to get what I described.

I will close the issue.

closed 2019-03-06 by hsudbrock