openhab/org.openhab.binding.zigbee #406: Xiaomi Cube not recognized with channels

schniddsel, 2019-03-09:
Java Exception:
`2019-03-09 19:32:59.326 [ERROR] [tworkManager$AFMessageListenerFilter] - Error AF message listener notify.

java.lang.NullPointerException: null

	at com.zsmartsystems.zigbee.serialization.DefaultDeserializer.readZigBeeType(DefaultDeserializer.java:72) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.zcl.field.AttributeReport.deserialize(AttributeReport.java:100) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.zcl.ZclFieldDeserializer.deserialize(ZclFieldDeserializer.java:73) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.zcl.clusters.general.ReportAttributesCommand.deserialize(ReportAttributesCommand.java:88) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveZclCommand(ZigBeeNetworkManager.java:808) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveCommand(ZigBeeNetworkManager.java:731) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.ZigBeeDongleTiCc2531.notify(ZigBeeDongleTiCc2531.java:316) ~[234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.ZigBeeNetworkManager$AFMessageListenerFilter.receivedAsynchronousCommand(ZigBeeNetworkManager.java:1288) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.notifyAsynchronousCommand(CommandInterfaceImpl.java:382) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.handlePacket(CommandInterfaceImpl.java:160) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:116) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at java.lang.Thread.run(Thread.java:748) [?:?]`

Xiaomi Cube is discovered, but has no channels. Qnly thing i recognize on turning the cube is the excepting from above. Is this a known issue?

cdjackson, 2019-03-09:
This is probably not an exception from the binding - it's probably from the lower layers which is a separate project. Please provide a debug log showing the issue and we might then move it to the appropriate place.

schniddsel, 2019-03-09:
I have another debug log from detecting the device:
`2019-03-09 19:15:37.687 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00027DF930: Node updated - ZigBeeNode [IEEE=00158D00027DF930, NWK=91AF, Type=END_DEVICE]

2019-03-09 19:20:43.859 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00027DF930: Node updated - ZigBeeNode [IEEE=00158D00027DF930, NWK=91AF, Type=END_DEVICE]

2019-03-09 19:24:56.599 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00027DF930: Node updated - ZigBeeNode [IEEE=00158D00027DF930, NWK=91AF, Type=END_DEVICE]

2019-03-09 19:30:12.450 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00027DF930: Node updated - ZigBeeNode [IEEE=00158D00027DF930, NWK=91AF, Type=END_DEVICE]

2019-03-09 19:35:10.960 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00027DF930: Node updated - ZigBeeNode [IEEE=00158D00027DF930, NWK=91AF, Type=END_DEVICE]

2019-03-09 19:39:08.869 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00027DF930: Discovery: Starting discovery for existing device

2019-03-09 19:39:08.885 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 00158D00027DF930: Starting ZigBee device discovery

2019-03-09 19:39:08.890 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00027DF930: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:stick

2019-03-09 19:39:08.901 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00027DF930: ZigBee node property discovery start

2019-03-09 19:39:08.903 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00027DF930: ZigBee node property discovery using 37295/1

2019-03-09 19:39:16.913 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00027DF930: OTA firmware failed

2019-03-09 19:39:16.918 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00027DF930: ZigBee node property discovery complete: {zigbee_logicaltype=END_DEVICE, zigbee_powerlevel=FULL, modelId=lumi.sensor_cube.aqgl01, vendor=LUMI, zigbee_networkaddress=37295, zigbee_powermode=RECEIVER_ON_IDLE, zigbee_powersources=[DISPOSABLE_BATTERY], zigbee_powersource=DISPOSABLE_BATTERY, hardwareVersion=1, zigbee_stkversion=2, zigbee_datecode=20160704, zigbee_zclversion=1}

2019-03-09 19:39:16.926 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00027DF930: Update ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:stick, label 'LUMI lumi.sensor_cube.aqgl01'

2019-03-09 19:42:50.764 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00027DF930: Node updated - ZigBeeNode [IEEE=00158D00027DF930, NWK=91AF, Type=END_DEVICE]

2019-03-09 19:48:37.763 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00027DF930: Node updated - ZigBeeNode [IEEE=00158D00027DF930, NWK=91AF, Type=END_DEVICE]

2019-03-09 19:53:01.886 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00027DF930: Node updated - ZigBeeNode [IEEE=00158D00027DF930, NWK=91AF, Type=END_DEVICE]

2019-03-09 19:54:06.804 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00027DF930: Discovery: Starting discovery for existing device

2019-03-09 19:54:06.818 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 00158D00027DF930: Starting ZigBee device discovery

2019-03-09 19:54:06.821 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00027DF930: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:stick

2019-03-09 19:54:06.825 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00027DF930: ZigBee node property discovery start

2019-03-09 19:54:06.832 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00027DF930: ZigBee node property discovery using 37295/1

2019-03-09 19:54:14.838 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00027DF930: OTA firmware failed

2019-03-09 19:54:14.842 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00027DF930: ZigBee node property discovery complete: {zigbee_logicaltype=END_DEVICE, zigbee_powerlevel=FULL, modelId=lumi.sensor_cube.aqgl01, vendor=LUMI, zigbee_networkaddress=37295, zigbee_powermode=RECEIVER_ON_IDLE, zigbee_powersources=[DISPOSABLE_BATTERY], zigbee_powersource=DISPOSABLE_BATTERY, hardwareVersion=1, zigbee_stkversion=2, zigbee_datecode=20160704, zigbee_zclversion=1}

2019-03-09 19:54:14.846 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00027DF930: Update ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:stick, label 'LUMI lumi.sensor_cube.aqgl01'`

cdjackson, 2019-03-09:
I need to see a debug log that shows the data received from the dongle, as well as the exception - all in one place. It doesn't look like you have enabled all the debug logging - please refer to the binding documentation.

Without this log with the data, and the exception, all together, it's hard to know what caused the exception.

schniddsel, 2019-03-09:
OK....now i think i got it. Activatet all debug logs. So here the output from Receiving data until the exception occurs:

`2019-03-09 20:54:28.806 [ERROR] [tworkManager$AFMessageListenerFilter] - Error AF message listener notify.

java.lang.NullPointerException: null

	at com.zsmartsystems.zigbee.serialization.DefaultDeserializer.readZigBeeType(DefaultDeserializer.java:72) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.zcl.field.AttributeReport.deserialize(AttributeReport.java:100) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.zcl.ZclFieldDeserializer.deserialize(ZclFieldDeserializer.java:73) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.zcl.clusters.general.ReportAttributesCommand.deserialize(ReportAttributesCommand.java:88) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveZclCommand(ZigBeeNetworkManager.java:808) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveCommand(ZigBeeNetworkManager.java:731) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.ZigBeeDongleTiCc2531.notify(ZigBeeDongleTiCc2531.java:316) ~[234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.ZigBeeNetworkManager$AFMessageListenerFilter.receivedAsynchronousCommand(ZigBeeNetworkManager.java:1288) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.notifyAsynchronousCommand(CommandInterfaceImpl.java:382) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.handlePacket(CommandInterfaceImpl.java:160) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:116) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at java.lang.Thread.run(Thread.java:748) [?:?]

2019-03-09 20:54:28.813 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)

2019-03-09 20:54:29.178 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- ZToolPacket (FE 05 45 C4 7E 35 01 A9 BD DA)

2019-03-09 20:54:29.182 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=5, apiId=45 C4, data=FE 05 45 C4 7E 35 01 A9 BD DA, checksum=DA, error=false

2019-03-09 20:54:29.187 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x45c4

2019-03-09 20:54:29.194 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 23 44 81 00 00 0C 00 7E 35 03 01 00 5E 00 A7 7D D5 00 00 0F 18 12 0A 05 FF 21 90 01 55 00 39 86 EB 41 41 A9 BD 1C BE)

2019-03-09 20:54:29.198 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=35, apiId=44 81, data=FE 23 44 81 00 00 0C 00 7E 35 03 01 00 5E 00 A7 7D D5 00 00 0F 18 12 0A 05 FF 21 90 01 55 00 39 86 EB 41 41 A9 BD 1C BE, checksum=BE, error=false

2019-03-09 20:54:29.204 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=13694/3, destinationAddress=0/1, profile=0104, cluster=12, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=18 12 0A 05 FF 21 90 01 55 00 39 86 EB 41 41]

2019-03-09 20:54:29.208 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=18, commandId=10]

2019-03-09 20:54:29.210 [ERROR] [tworkManager$AFMessageListenerFilter] - Error AF message listener notify.

java.lang.NullPointerException: null

	at com.zsmartsystems.zigbee.serialization.DefaultDeserializer.readZigBeeType(DefaultDeserializer.java:72) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.zcl.field.AttributeReport.deserialize(AttributeReport.java:100) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.zcl.ZclFieldDeserializer.deserialize(ZclFieldDeserializer.java:73) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.zcl.clusters.general.ReportAttributesCommand.deserialize(ReportAttributesCommand.java:88) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveZclCommand(ZigBeeNetworkManager.java:808) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveCommand(ZigBeeNetworkManager.java:731) ~[233:com.zsmartsystems.zigbee:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.ZigBeeDongleTiCc2531.notify(ZigBeeDongleTiCc2531.java:316) ~[234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.ZigBeeNetworkManager$AFMessageListenerFilter.receivedAsynchronousCommand(ZigBeeNetworkManager.java:1288) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.notifyAsynchronousCommand(CommandInterfaceImpl.java:382) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.handlePacket(CommandInterfaceImpl.java:160) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:116) [234:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]

	at java.lang.Thread.run(Thread.java:748) [?:?]`

cdjackson, 2019-03-09:
Thanks - that looks better… But can you also provide the log entries from just before the exception - maybe 10 lines or so before will hopefully be ok.

schniddsel, 2019-03-09:
For sure i can do that. Unfortunately formatting is a bit messed up. Sorry for that. Here is the log with some lines before the Error:

`
2019-03-09 21:39:47.884 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D0002BF8A4B: Node SVC Discovery: running

2019-03-09 21:39:47.888 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementLqiRequest [0/0 -> 49162/0, cluster=0031, TID=D5, startIndex=0]

2019-03-09 21:39:47.891 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=49162/0, profile=0000, cluster=49, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=213, payload=00 00]

2019-03-09 21:39:47.895 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=12, apiId=24 01, data=FE 0C 24 01 0A C0 00 00 31 00 D5 30 1F 02 00 00 2A, checksum=2A, error=false) 

2019-03-09 21:39:48.019 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)

2019-03-09 21:39:50.014 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction timeout: ManagementLqiRequest [0/0 -> 55722/0, cluster=0031, TID=D4, startIndex=0]

2019-03-09 21:39:50.021 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D0002EC80C3: Node SVC Discovery: ManagementLqiRequest response CommandResult [TIMEOUT]

2019-03-09 21:39:50.027 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D0002EC80C3: Node SVC Discovery: request NEIGHBORS failed. Retry 11, wait 23490ms before retry.

2019-03-09 21:39:50.188 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D0002EBD65C: Node SVC Discovery: running

2019-03-09 21:39:50.193 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementLqiRequest [0/0 -> 4906/0, cluster=0031, TID=D6, startIndex=0]

2019-03-09 21:39:50.199 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=4906/0, profile=0000, cluster=49, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=214, payload=00 00]

2019-03-09 21:39:50.203 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=12, apiId=24 01, data=FE 0C 24 01 2A 13 00 00 31 00 D6 30 1F 02 00 00 DA, checksum=DA, error=false) 

2019-03-09 21:39:50.326 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)

2019-03-09 21:39:51.718 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 000B57FFFEE64564: Polling...

2019-03-09 21:39:51.724 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 000B57FFFEE64564: Polling zigbee:device:stick:000b57fffee64564:000B57FFFEE64564_1_dimmer

2019-03-09 21:39:51.729 [DEBUG] [.zsmartsystems.zigbee.zcl.ZclCluster] - readSync request: ZclAttribute [cluster=ON_OFF, id=0, name=OnOff, dataType=BOOLEAN, lastValue=true, lastReportTime=Sat Mar 09 21:38:46 CET 2019]

2019-03-09 21:39:51.733 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [On/Off: 0/0 -> 48553/1, cluster=0006, TID=D7, identifiers=[0]]

2019-03-09 21:39:51.738 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=215, commandId=0]

2019-03-09 21:39:51.742 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/1, destinationAddress=48553/1, profile=0104, cluster=6, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=215, payload=00 D7 00 00 00]

2019-03-09 21:39:51.749 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=15, apiId=24 01, data=FE 0F 24 01 A9 BD 01 01 06 00 D7 30 1F 05 00 D7 00 00 00 12, checksum=12, error=false) 

2019-03-09 21:39:51.915 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)

2019-03-09 21:39:51.920 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- AF_DATA_CONFIRM (FE 03 44 80 00 01 D7 11)

2019-03-09 21:39:51.923 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: AF_DATA_CONFIRM(Endpoint=1, Status=SUCCESS(0), TransID=215)

2019-03-09 21:39:51.931 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1C 44 81 00 00 06 00 A9 BD 01 01 00 63 00 72 53 58 00 00 08 18 D7 01 00 00 00 10 01 A9 BD 1D 0F)

2019-03-09 21:39:51.935 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=28, apiId=44 81, data=FE 1C 44 81 00 00 06 00 A9 BD 01 01 00 63 00 72 53 58 00 00 08 18 D7 01 00 00 00 10 01 A9 BD 1D 0F, checksum=0F, error=false

2019-03-09 21:39:51.940 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=48553/1, destinationAddress=0/1, profile=0104, cluster=6, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=18 D7 01 00 00 00 10 01]

2019-03-09 21:39:51.943 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=215, commandId=1]

2019-03-09 21:39:51.947 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReadAttributesResponse [On/Off: 48553/1 -> 0/1, cluster=0006, TID=D7, records=[ReadAttributeStatusRecord [attributeDataType=BOOLEAN, attributeIdentifier=0, status=SUCCESS, attributeValue=true]]]

[... cut ...]
2019-03-09 21:40:21.356 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction timeout: ManagementLqiRequest [0/0 -> 13694/0, cluster=0031, TID=DC, startIndex=0]

2019-03-09 21:40:21.367 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00027DF930: Node SVC Discovery: ManagementLqiRequest response CommandResult [TIMEOUT]

2019-03-09 21:40:21.372 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00027DF930: Node SVC Discovery: request NEIGHBORS failed. Retry 12, wait 8520ms before retry.
`

cdjackson, 2019-03-09:
Thanks - this is fine and I see the issue.

cdjackson, 2019-03-09:
Closed by https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/570

closed 2019-03-09 by cdjackson

cdjackson, 2019-03-10:
If you run the latest snapshot if the libraries, this exception should no longer occur.

https://community.openhab.org/t/zigbee-binding/15763/2106

schniddsel, 2019-03-11:
Thanks for the quick update. After installing the 2.5.0 Snapshot and manually installing your 1.1.11 SNAPSHOT. The Cube is discovered like before and the correct Device-Type is detected. Sadly, there are still no Channels available on the Thing. 
Second Problem is, that the Binding is receiving data from the cube when it is spinned - this time without an Exception - but it says "unknown attribute [various number] in cluster 12"  

Here the logs on receiving data:

`2019-03-11 12:33:40.510 [DEBUG] [nal.ZigBeeNetworkStateSerializerImpl] - Saving ZigBee network state: Done.

2019-03-11 12:33:41.057 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction complete: ZigBeeTransaction [queueTime=38183, state=FAILED, sendCnt=2, command=ManagementLqiRequest [0/0 -> 30180/0, cluster=0031, TID=98, startIndex=0]]

2019-03-11 12:33:41.058 [DEBUG] [transaction.ZigBeeTransactionManager] - removeTransactionListener: ZigBeeTransaction [queueTime=38185, state=FAILED, sendCnt=2, command=ManagementLqiRequest [0/0 -> 30180/0, cluster=0031, TID=98, startIndex=0]]

2019-03-11 12:33:41.059 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerRemoved: 1 outstanding

2019-03-11 12:33:41.062 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 00158D00025EE5D6: transactionComplete FAILED 0

2019-03-11 12:33:41.063 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 00158D00025EE5D6: transactionComplete exceeded retries 2

2019-03-11 12:33:41.064 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction terminated: ZigBeeTransaction [queueTime=38191, state=FAILED, sendCnt=2, command=ManagementLqiRequest [0/0 -> 30180/0, cluster=0031, TID=98, startIndex=0]]

2019-03-11 12:33:41.066 [DEBUG] [transaction.ZigBeeTransactionManager] - 30180/0: Sending ZigBeeTransaction [queueTime=10022, state=WAITING, sendCnt=0, command=ManagementLqiRequest [0/0 -> 30180/0, cluster=0031, TID=9A, startIndex=0]]

2019-03-11 12:33:41.068 [DEBUG] [transaction.ZigBeeTransactionManager] - addTransactionListener: ZigBeeTransaction [queueTime=10023, state=WAITING, sendCnt=0, command=ManagementLqiRequest [0/0 -> 30180/0, cluster=0031, TID=9A, startIndex=0]]

2019-03-11 12:33:41.069 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerAdded: 2 outstanding

2019-03-11 12:33:41.071 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementLqiRequest [0/0 -> 30180/0, cluster=0031, TID=9A, startIndex=0]

2019-03-11 12:33:41.073 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=30180/0, profile=0000, cluster=0031, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=D4, payload=00 00]

2019-03-11 12:33:41.074 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=12, apiId=24 01, data=FE 0C 24 01 E4 75 00 00 31 00 D4 30 1F 02 00 00 70, checksum=70, error=false) 

2019-03-11 12:33:41.202 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)

2019-03-11 12:33:43.197 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00025EE5D6: Node SVC Discovery: ManagementLqiRequest response CommandResult [TIMEOUT]

2019-03-11 12:33:43.200 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00025EE5D6: Node SVC Discovery: request NEIGHBORS failed after 13 attempts.

2019-03-11 12:33:43.204 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00025EE5D6: Node SVC Discovery: complete

2019-03-11 12:33:43.207 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00158D00025EE5D6: Node 30180 update

2019-03-11 12:33:43.211 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 00158D00025EE5D6: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=1, interTransactionDelay=50, maxRetries=2]

2019-03-11 12:33:43.214 [DEBUG] [nal.ZigBeeNetworkStateSerializerImpl] - Saving ZigBee network state

2019-03-11 12:33:43.216 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00025EE5D6: Node updated - ZigBeeNode [state=UNKNOWN, IEEE=00158D00025EE5D6, NWK=75E4, Type=END_DEVICE]

2019-03-11 12:33:43.225 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 00158D00025EE5D6: Starting ZigBee device discovery

2019-03-11 12:33:43.244 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00025EE5D6: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:stick

2019-03-11 12:33:43.247 [DEBUG] [nal.ZigBeeNetworkStateSerializerImpl] - Saving ZigBee network state: Start.

2019-03-11 12:33:43.255 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00025EE5D6: ZigBee node property discovery start

2019-03-11 12:33:43.259 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00025EE5D6: ZigBee node property discovery using 30180/1

==> /var/log/openhab2/events.log <==

2019-03-11 12:33:43.261 [me.event.ThingUpdatedEvent] - Thing 'zigbee:device:stick:00158d00025ee5d6' has been updated.

==> /var/log/openhab2/openhab.log <==

2019-03-11 12:33:43.265 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 00158D00025EE5D6: ZigBee node property discovery complete: {zigbee_logicaltype=END_DEVICE, zigbee_powerlevel=FULL, modelId=lumi.weather, vendor=LUMI, zigbee_networkaddress=30180, zigbee_powermode=RECEIVER_ON_IDLE, zigbee_powersources=[DISPOSABLE_BATTERY], zigbee_powersource=DISPOSABLE_BATTERY, hardwareVersion=30, zigbee_stkversion=2, zigbee_datecode=20161129, zigbee_zclversion=1}

[... cut ...] ttributeDataType=UNSIGNED_16_BIT_INTEGER, attributeIdentifier=65285, attributeValue=500, Attribute Report: attributeDataType=FLOAT_32_BIT, attributeIdentifier=85, attributeValue=88.16000366210938]] 

2019-03-11 12:34:20.891 [DEBUG] [.zsmartsystems.zigbee.zcl.ZclCluster] - 58079/3: Unknown attribute 65285 in cluster 12

2019-03-11 12:34:20.895 [DEBUG] [.zsmartsystems.zigbee.zcl.ZclCluster] - 58079/3: Unknown attribute 85 in cluster 12`

Please let me know, when i can help you with some more logs.

cdjackson, 2019-03-11:
No - I do not expect any channels. The cluster is not currently supported at all. The current changes were simply to stop the exception.

The following issue needs to be resolved before the cluster is supported, then, once that is done, someone will need to write support for this device.
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/570

schniddsel, 2019-03-11:
Ok thanks a lot. You may have enough to do...but i would like to know, where the devices are specified? I mean in the code. I would like to learn some dev-specs. It would be great, if you could show me some code, or the places where i can find that code. Thanks a lot! I really appreciate your work!

cdjackson, 2019-03-11:
They aren't directly specified as such (normally anyway). The binding attempts to detect the features a device reports, and then to provide channels for these features. This means that the binding is "device independent".

So, when I said "support for this device" - what I probably should have said was "channels that reflect this device".

Now, for non-standard devices (which this might fall into) it may be necessary to write a complete new device handler. This would be a standard OH thingHandler so you can refer to standard ESH/OH documentation.