openhab/org.openhab.binding.zigbee #410: Support for power consumption measurement using Smart Energy Metering cluster

hsudbrock, 2019-03-12:
The binding already supports power measurement based on the electrical measurements cluster (cf. `ZigBeeConverterMeasurementPower`). Some devices (like, e.g., SmartPlugs from BitronVideo) provide power measurements only based on the smart energy metering cluster.  This issue is for adding support for power measurement based on the smart energy metering cluster (which will be introduced into the ZSS ZigBee library with https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/527).

I plan on providing a PR for this.

cdjackson, 2019-03-12:
Is your plan to add this to the existing converter?

I'm not sure what is best here, so I'm interested in your thoughts as this is something I keep thinking about. There are numerous energy channels available in different clusters (probably the SE clusters have literally dozens - different phases etc).

I don't have a strong feeling for how to do this, but at the risk of complicating things, it might be worth considering this "duplication"?

hsudbrock, 2019-03-12:
> Is your plan to add this to the existing converter?

This would be my preference, unless it makes the converter too complex. But, to be honest, I will need to play around with this a little in the code first, before I can say what I would actually prefer. But I will comment here again as soon as I have done that.

falmanna, 2019-12-10:
Is this about the metering cluster (0x0702)?
I just received a zgibee energy meter that has the 0x0702 cluster. Unfortunatly, the binding didn't recognize any channel. 

By [looking](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/e2d03ddb6729445cf2ba8acd1cb61f90b67c75a0/com.zsmartsystems.zigbee.autocode/src/main/resources/0702_Metering.xml) at the ZSS lib, it seems that they are supported, so I guess this cluster is not handled here yet?

I believe implementing issue #414 might also help in integrating metering devices?

cdjackson, 2021-04-25:
I will close this as it's a bit too generic. There are a number of channels that support the metering cluster. If more are required, then we should name them specifically and open a new issue.

closed 2021-04-25 by cdjackson