openhab/org.openhab.binding.zigbee #415: RuntimeException thrown when ZigBee stick is unplugged.

YordanDZhelev, 2019-03-21:
In case of unplugged zigbee stick the binding still tries to open it, which results in the following `RuntimeException` 
```
java.lang.RuntimeException: No such device in open
    at gnu.io.RXTXCommDriver.getCommPort(getCommPort:835)
    at gnu.io.CommPortIdentifier.open(open:383)
    at org.openhab.binding.zigbee.handler.ZigBeeSerialPort.open(open:136)
    at org.openhab.binding.zigbee.handler.ZigBeeSerialPort.open(open:112)
    at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialiseEzspProtocol(initialiseEzspProtocol:878)
    at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialize(initialize:310)
    at com.zsmartsystems.zigbee.ZigBeeNetworkManager.initialize(initialize:334)
    at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.initialiseZigBee(initialiseZigBee:422)
    at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.access$6(access$6:382)
    at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler$1.run(run:532)
    at java.util.concurrent.Executors$RunnableAdapter.call(call)
    at java.util.concurrent.FutureTask.runAndReset(runAndReset)
    at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(access$301)
    at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(run)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(runWorker)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(run)
    at java.lang.Thread.run(run)
```
The same error was observed in two different variants:
```
java.lang.RuntimeException: No such file or directory in open
    at gnu.io.RXTXCommDriver.getCommPort(getCommPort:835)
    at gnu.io.CommPortIdentifier.open(open:383)
    at org.openhab.binding.zigbee.handler.ZigBeeSerialPort.open(open:136)
```
and 
```
java.lang.RuntimeException: No such device in open
    at gnu.io.RXTXCommDriver.getCommPort(getCommPort:835)
    at gnu.io.CommPortIdentifier.open(open:383)
    at org.openhab.binding.zigbee.handler.ZigBeeSerialPort.open(open:136)
```
Thi could be fixed by catching the `RuntimeException` thrown at `org.openhab.binding.zigbee.handler.ZigBeeSerialPort.open(open:136)` and logging the appropriate error message.

triller-telekom, 2019-03-21:
> and logging the appropriate error message.

This is not an "error" per se, its just a intentional retry. So I would only log it at trace level, if at all.

hsudbrock, 2019-03-21:
I understand that you don't want to log too excessively when trying to reconnect after the USB dongle has been plugged out. Are you sure that this would not suppress useful logging in other situations? I see in #416 that you suggest to only trace-log in case of a `RuntimeException` - my expectation would be that runtime exceptions can also occur in other, really erroneous situations. How can you be sure that a `RuntimeException` only happens in the scenario "dongle was removed intentionally" that you describe above?

triller-telekom, 2019-03-21:
Skipping the stacktrace it a good point. But this still leaves the error message in the logs quite often.

@YordanDZhelev Do you know how often the `RuntimeException` occurs?

If it isn't too often I am fine with leaving it at debug level.

YordanDZhelev, 2019-03-21:
We could log it as error like, e.g. `logger.error("Serial Error: Device no longer connected to Port {}.", portName);` What do you think? This way will not lose the real errors caused by something else than intentional stick removal.

triller-telekom, 2019-03-21:
I would vote for level "warning" since this is not a non recoverable error. The system can recover from it.

So how about logging something like:

```
logger.warn("Serial Error {}: Device no longer connected to Port {}. This might be because the stick was unplugged.", e.getMessage(), portName);
```

hsudbrock, 2019-03-21:
I think it is good to omit the stack trace, and to log a warning. I still think that it is dangerous to assume that a `RuntimeException` always signifies that the device is no longer connected to the port - I would hence not write `Device no longer connected to Port {}` as you suggest, @triller-telekom .

cdjackson, 2019-03-21:
Won't this get logged every "few seconds" - ie doesn't the reconnection continue to try and reconnect periodically? If so, we presumably will get this log entry at this rate, and if so, I would not want this at either ```WARN``` or ```ERROR``` - and think it should be ```DEBUG```. Otherwise it will fill the logs, and I know that users will complain. 

If it logs once, then I'm happy with ```WARN```.

I also agree that we shouldn't log that the device is no longer connected - just that it can't be opened. I don't think we should really log things like "this might be because" - there are a bunch of possible reasons and I think the logs should state facts about what has happened and not speculate on the cause where it can't be known.

t-8ch, 2019-03-21:
If it is logged at debug would it make sense to keep the stacktrace?
For errors like NullPointerException, FileNotFound or PermissionDenied I think it would help very much having this information.

cdjackson, 2019-03-21:
No - I don't believe that these sort of exceptions should have stack trace logged - just the cause. It is clear where and why they occur and having the stack trace doesn't provide any further information.

We are not talking about NPE or a general statement not to log the stack trace here (where I agree that the stack trace is essential) - we are talking about being unable to open a serial port - the exceptions should be caught explicitly as is currently done now for most of these errors.

closed 2019-04-09 by cdjackson