openhab/org.openhab.binding.zigbee #420: Build fails due to checkstyle issue (after recent bum of sat-plugin to 0.6.1 in openhab pom)

hsudbrock, 2019-04-01:
The checkstyle issue can be seen, e.g., in this build: https://ci.openhab.org/job/PR-org.openhab.binding.zigbee/568/consoleFull (which is not for `master` branch, but it is reproducible on `master` branch as well).

The error message is: `Failed to execute goal org.openhab.tools.sat:sat-plugin:0.6.1:checkstyle (default) on project org.openhab.binding.zigbee: Unable to execute mojo: An error has occurred in Checkstyle report generation. Failed during checkstyle execution: Failed during checkstyle configuration: unable to parse configuration stream: Property ${checkstyle.forbiddenPackageUsageCheck.forbiddenPackages} has not been set`.

This is probably not ass issue for most other bindings, as for the `openhab-addons` project such checkstyle properties are set (here: https://github.com/openhab/openhab2-addons/blob/master/tools/checkstyle.properties).

hsudbrock, 2019-04-01:
I will have another look this afternoon to see what needs to be done here. (Or are you maybe already investigating this, @cdjackson ?)

hsudbrock, 2019-04-01:
The following PR for the `static-code-analysis` project should resolve the issue: https://github.com/openhab/static-code-analysis/pull/329 However, it is quite some months old, and I don't know if it is intended to be merged...

cdjackson, 2019-04-01:
There was a PR on the ZWave binding from @wbourn yesterday that I think has fixed the problem with ZWave. As CI is running very slowly (ie it's taking 6 or 7 hours) it takes a long time to ru and can time outn, but it did pass in the end.

cdjackson, 2019-04-01:
See https://github.com/openhab/org.openhab.binding.zwave/pull/1158

hsudbrock, 2019-04-01:
Thanks for the pointer, Chris. As a quick fix, we could do a similar thing as for the Z-Wave binding. But: The approach used for Z-Wave is note 100% ideal for the ZigBee binding, because we have a multi-module project, and referencing the checkstyle config from the top-level pom will prevent building successfully from within a submodule.

There might be workarounds (like the one here described for checkstyle: https://maven.apache.org/plugins/maven-checkstyle-plugin/examples/multi-module-config.html), but I haven't checked those yet.

cdjackson, 2019-04-01:
I’ve not looked at how this is done in the openhab repo, but isn’t there the same issue there with multiple modules?

Sorry for the hassle with this - it was changed in openhab without any notification to maintainers - if you have a proposal to resolve it, it would be great?

hsudbrock, 2019-04-01:
I had hoped to get feedback on the non-merged PR from the code analysis project. I will create a PR for the binding pom (maybe today, probably tomorrow) as I did not get any feedback...

PR #421 (2019-04-04): Add checkstyle configuration to fix the build
This resolves #420.

As in the `openhab-addons2` project, I use the [`directory-maven-plugin`](https://github.com/openhab/openhab2-addons/blob/54eaec83e34604321075554c5bd6f7b2981cc227/poms/tycho/pom.xml#L132) to capture the directory with the checkstyle configuration.

The checkstyle configuration is like for the Z-Wave binding (see https://github.com/openhab/org.openhab.binding.zwave/pull/1158, but with the check for the copyright header adapted to match https://github.com/openhab/org.openhab.binding.zigbee/blob/master/src/etc/header.txt.

Maybe t [... cut]
commit 87c496de (2019-04-04): Add checkstyle configuration to fix the build (#421)

This resolves #420.
closed 2019-04-04 by cdjackson