openhab/org.openhab.binding.zigbee #422: Ember: Config setting zigbee_txpower not implemented?

hsudbrock, 2019-04-03:
I just stumbled upon the config setting `zigbee_txpower` in the ember coordinator config description (see [here](https://github.com/openhab/org.openhab.binding.zigbee/blob/9492e9a79a4956388eab58a08025b786a0d20d84/org.openhab.binding.zigbee.ember/ESH-INF/thing/controller_ember.xml#L78-L86)).

My impression is that this is in the config description, but currently not implemented. In particular, this setting is not included in the [`EmberConfiguration`](https://github.com/openhab/org.openhab.binding.zigbee/blob/master/org.openhab.binding.zigbee.ember/src/main/java/org/openhab/binding/zigbee/ember/internal/EmberConfiguration.java).

Do I overlook something? Would it be useful if I created a PR to remove this setting?

cdjackson, 2019-04-03:
As far as I know this is fully implemented a long time ago. Please don’t remove it - many people use this.

hsudbrock, 2019-04-03:
Hm, I see that the `zigbee_powermode` is fully implemented for the Ember. Are you sure about the `zigbee_txpower` ?

hsudbrock, 2019-04-03:
This here is the current set of attributes in the `EmberConfiguration`:

```
public class EmberConfiguration {
    public String zigbee_port;
    public Integer zigbee_baud;
    public Integer zigbee_flowcontrol;
    public Integer zigbee_powermode;
    public Integer zigbee_childtimeout;
    public Integer zigbee_concentrator;
}```

hsudbrock, 2019-04-03:
Ah, sorry Chris, I guess this is done in the base class for all coordinators, I will have another look (and Eclipse's full text search seems to have tricked me before...)

closed 2019-04-03 by hsudbrock

cdjackson, 2019-04-03:
I would need to check the naming, maybe it is the same, but I’m almost 100% sure this is implemented and was tested by many people.

hsudbrock, 2019-04-03:
I think you are right - I restarted my Eclipse to be completely sure that it is not there - and after that restart I got other search results. Probably I had a wrong search setting before, sorry for the hassle.

cdjackson, 2019-04-03:
No worries :)