openhab/org.openhab.binding.zigbee #423: Missing Service Capability

amitjoy, 2019-04-05:
`org.openhab.binding.zigbee` should provide `org.openhab.binding.zigbee.converter.ZigBeeChannelConverterFactory` service capability in its manifest.

cdjackson, 2019-04-05:
Please provide information on your issue, and your configuration, versions etc and any logs!

amitjoy, 2019-04-08:
It is actually not an issue but an improvement. In our project, we are using bndtools for development and we deal with several different types of gateways. For preparing the list of bundles for any specific type of gateway, we use OSGi Resolver that reads OSGi Metadata and prepares the deployment package required for any specific type of gateway. Since we use openHAB ZigBee as well, we have client bundles that use `org.openhab.binding.zigbee.converter.ZigBeeChannelConverterFactory` using OSGi DS. If a SCR Component uses a services dependency, bndtools automatically generates metadata specifying that the consumer bundle depends on a service requirement. The same applies to the client bundles that use `org.openhab.binding.zigbee.converter.ZigBeeChannelConverterFactory`, that means, the client bundles have a service requirement of `org.openhab.binding.zigbee.converter.ZigBeeChannelConverterFactory`. But there is no bundle that provides the required metadata for this service. Since `org.openhab.binding.zigbee` provides the implementation of `org.openhab.binding.zigbee.converter.ZigBeeChannelConverterFactory`, it should include the following in its manifest:

```
Provide-Capability: osgi.service;objectClass:List<String>="org.openhab.binding.zigbee.converter.ZigBeeChannelConverterFactory"
```

hanneshofmann, 2019-04-10:
As the `Provide-Capability` manifest header is not used nor required within the openHAB binding development, I suggest to not add them in the ZigBee binding. 

If an integrator of the binding uses bndtools and needs further manifest headers, the integrator need to add them himself (e.g. using the [Maven BND plugin](http://felix.apache.org/documentation/subprojects/apache-felix-maven-bundle-plugin-bnd.html) or handle these as an internal on top commit).

I believe there is no need to adapt the binding. Therefore, could you please close this @cdjackson?

cdjackson, 2019-04-10:
Thanks @hanneshofmann

closed 2019-04-10 by cdjackson