openhab/org.openhab.binding.zigbee #428: Xiaomi MiJia Smart Wireless Switch (WXKG01LM) - OFFLINE Node has not completed discovery

achille2k, 2019-04-24:
I installed OH 2.5.0 Snapshot Zigbee binding with CC2531 Coordinator. With Aqara Motion and Luminance Sensor (WSDCGQ11LM) and Aqara Temperature and Humidity Sensor (WSDCGQ11LM) work perfectly fine.
But I add Xiaomi MiJia Smart Wireless Switch (WXKG01LM), in this Paper UI thing show message "OFFLINE Node has not completed discovery". Sorry For My Bad English!

This Hide Properties
zigbee_devices 	[]
zigbee_lastupdate 	
zigbee_neighbors 	[]
zigbee_networkaddress 	60921
zigbee_routes 	[]
zigbee_stkcompliance 	0

OH 2.5.0 Snapshot Zigbee:

286 │ Active   │  80 │ 2.5.0.201904161359     │ org.openhab.binding.zigbee
287 │ Active   │  80 │ 2.5.0.201904150817     │ org.openhab.binding.zigbee.console
288 │ Active   │  80 │ 2.5.0.201904150817     │ org.openhab.binding.zigbee.xbee
289 │ Active   │  80 │ 2.5.0.201904150817     │ org.openhab.binding.zigbee.cc2531
290 │ Active   │  80 │ 2.5.0.201904150817     │ org.openhab.binding.zigbee.console.ember
291 │ Active   │  80 │ 2.5.0.201904150817     │ org.openhab.binding.zigbee.console.telegesis
292 │ Active   │  80 │ 2.5.0.201904150817     │ org.openhab.binding.zigbee.ember
293 │ Active   │  80 │ 2.5.0.201904150817     │ org.openhab.binding.zigbee.telegesis
294 │ Active   │  80 │ 1.1.6                  │ com.zsmartsystems.zigbee
295 │ Active   │  80 │ 1.1.6                  │ com.zsmartsystems.zigbee.dongle.telegesis
296 │ Active   │  80 │ 1.1.6                  │ com.zsmartsystems.zigbee.dongle.ember
297 │ Active   │  80 │ 1.1.6                  │ com.zsmartsystems.zigbee.console.ember
298 │ Active   │  80 │ 1.1.6                  │ com.zsmartsystems.zigbee.console.telegesis
299 │ Active   │  80 │ 1.1.6                  │ com.zsmartsystems.zigbee.console
300 │ Active   │  80 │ 1.1.6                  │ com.zsmartsystems.zigbee.dongle.xbee
301 │ Active   │  80 │ 1.1.6                  │ com.zsmartsystems.zigbee.dongle.cc2531

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used | CC2531                                                                         |
| openHAB version  | 2.4     |
| Hardware         | Raspberry Pi 3+                                                                |
| Memory           | 1Gb                                                                            |
| Java version     | 1.8.0_201                                                                      |
| Devices          | Xiaomi MiJia Smart Wireless Switch (WXKG01LM)                                  |

### Logs

2019-04-24 02:40:58.441 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 40 44 81 00 00 00 00 6C 50 01 01 00 63 00 64 9E FD 00 00 2C 18 62 0A 01 FF 42 25 01 21 BD 0B 04 21 A8 13 05 21 B4 00 06 24 01 00 00 00 00 64 29 C3 0B 65 21 7C 20 66 2B 0E 87 01 00 0A 21 00 00 6C 50 1D F1)
2019-04-24 02:40:58.446 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=64, apiId=44 81, data=FE 40 44 81 00 00 00 00 6C 50 01 01 00 63 00 64 9E FD 00 00 2C 18 62 0A 01 FF 42 25 01 21 BD 0B 04 21 A8 13 05 21 B4 00 06 24 01 00 00 00 00 64 29 C3 0B 65 21 7C 20 66 2B 0E 87 01 00 0A 21 00 00 6C 50 1D F1, checksum=F1, error=false
2019-04-24 02:40:58.459 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=20588/1, destinationAddress=0/1, profile=0104, cluster=0, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=18 62 0A 01 FF 42 25 01 21 BD 0B 04 21 A8 13 05 21 B4 00 06 24 01 00 00 00 00 64 29 C3 0B 65 21 7C 20 66 2B 0E 87 01 00 0A 21 00 00]
2019-04-24 02:40:58.462 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=98, commandId=10]
2019-04-24 02:40:58.466 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [Basic: 20588/1 -> 0/1, cluster=0000, TID=62, reports=[Attribute Report: attributeDataType=CHARACTER_STRING, attributeIdentifier=65281, attributeValue=^A!�^K^D!�^S^E!�]]
2019-04-24 02:40:58.470 [DEBUG] [.zsmartsystems.zigbee.zcl.ZclCluster] - 20588/1: Unknown attribute 65281 in cluster 0
2019-04-24 02:47:11.226 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 31 44 81 00 00 00 00 F9 ED 01 01 00 00 00 6F 63 10 00 00 1D 1C 34 12 06 0A 02 FF 4C 06 00 10 01 21 28 0C 21 A8 13 24 01 00 00 00 00 21 14 05 20 59 F9 ED 1D 8B)
2019-04-24 02:47:11.229 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=49, apiId=44 81, data=FE 31 44 81 00 00 00 00 F9 ED 01 01 00 00 00 6F 63 10 00 00 1D 1C 34 12 06 0A 02 FF 4C 06 00 10 01 21 28 0C 21 A8 13 24 01 00 00 00 00 21 14 05 20 59 F9 ED 1D 8B, checksum=8B, error=false
2019-04-24 02:47:11.233 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=60921/1, destinationAddress=0/1, profile=0104, cluster=0, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=1C 34 12 06 0A 02 FF 4C 06 00 10 01 21 28 0C 21 A8 13 24 01 00 00 00 00 21 14 05 20 59]
2019-04-24 02:47:11.235 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=true, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=4660, sequenceNumber=6, commandId=10]
2019-04-24 02:47:11.237 [ERROR] [tworkManager$AFMessageListenerFilter] - Error AF message listener notify.
java.lang.NullPointerException: null
        at com.zsmartsystems.zigbee.serialization.DefaultDeserializer.readZigBeeType(DefaultDeserializer.java:72) ~[294:com.zsmartsystems.zigbee:1.1.6]
        at com.zsmartsystems.zigbee.zcl.field.AttributeReport.deserialize(AttributeReport.java:100) ~[294:com.zsmartsystems.zigbee:1.1.6]
        at com.zsmartsystems.zigbee.zcl.ZclFieldDeserializer.deserialize(ZclFieldDeserializer.java:73) ~[294:com.zsmartsystems.zigbee:1.1.6]
        at com.zsmartsystems.zigbee.zcl.clusters.general.ReportAttributesCommand.deserialize(ReportAttributesCommand.java:88) ~[294:com.zsmartsystems.zigbee:1.1.6]
        at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveZclCommand(ZigBeeNetworkManager.java:808) ~[294:com.zsmartsystems.zigbee:1.1.6]
        at com.zsmartsystems.zigbee.ZigBeeNetworkManager.receiveCommand(ZigBeeNetworkManager.java:731) ~[294:com.zsmartsystems.zigbee:1.1.6]
        at com.zsmartsystems.zigbee.dongle.cc2531.ZigBeeDongleTiCc2531.notify(ZigBeeDongleTiCc2531.java:316) ~[301:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.ZigBeeNetworkManager$AFMessageListenerFilter.receivedAsynchronousCommand(ZigBeeNetworkManager.java:1288) [301:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.notifyAsynchronousCommand(CommandInterfaceImpl.java:382) [301:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.handlePacket(CommandInterfaceImpl.java:160) [301:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]
		at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:116) [301:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]
        at java.lang.Thread.run(Thread.java:748) [?:?]
2019-04-24 02:47:11.242 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 31 44 81 00 00 00 00 F9 ED 01 01 00 00 00 51 64 10 00 00 1D 1C 34 12 06 0A 02 FF 4C 06 00 10 01 21 28 0C 21 A8 13 24 01 00 00 00 00 21 14 05 20 59 F9 ED 1D B2)
2019-04-24 02:47:11.244 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=49, apiId=44 81, data=FE 31 44 81 00 00 00 00 F9 ED 01 01 00 00 00 51 64 10 00 00 1D 1C 34 12 06 0A 02 FF 4C 06 00 10 01 21 28 0C 21 A8 13 24 01 00 00 00 00 21 14 05 20 59 F9 ED 1D B2, checksum=B2, error=false
2019-04-24 02:47:11.249 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=60921/1, destinationAddress=0/1, profile=0104, cluster=0, addressMode=null, radius=0, apsSecurity=false, apsCounter=0, payload=1C 34 12 06 0A 02 FF 4C 06 00 10 01 21 28 0C 21 A8 13 24 01 00 00 00 00 21 14 05 20 59]
[... cut ...]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.notifyAsynchronousCommand(CommandInterfaceImpl.java:382) [301:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.impl.CommandInterfaceImpl.handlePacket(CommandInterfaceImpl.java:160) [301:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]
        at com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:116) [301:com.zsmartsystems.zigbee.dongle.cc2531:1.1.6]
        at java.lang.Thread.run(Thread.java:748) [?:?]

cdjackson, 2019-04-24:
Please upgrade to a newer version of the libraries. If you’re unsure how to do this there is a script that is discussed in the forum. (Zwave and ZigBee update script)

achille2k, 2019-04-24:
Thank you very much. I've updated and everything working fine!

closed 2019-04-24 by cdjackson