openhab/org.openhab.binding.zigbee #429: Cluster x not found for received endpoint command

achille2k, 2019-04-26:
### Outline

I have added successfully device Xiaomi Mi Smart Home Door / Window Sensor. But channel SwitchOnoff cannot update value. When I trace log and I see message "Cluster 6 not found for received endpoint command". Help me please.
Thanks so much.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |      CC2531                                                                          |
| openHAB version  | 2.4     |
| Hardware         |         Raspberry Pi 3+                                                                       |
| Memory           |     1Gb                                                                           |
| Java version     |    1.8.0_201                                                                            |
| Devices          | Xiaomi Mi Smart Home Door / Window Sensor (MCCGQ01LM)                      |

### Logs

2019-04-26 21:50:35.733 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1B 44 81 00 00 06 00 02 67 01 01 00 12 00 03 70 21 00 00 07 18 FA 0A 00 00 10 01 02 67 1D 7B)
2019-04-26 21:50:35.736 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=27, apiId=44 81, data=FE 1B 44 81 00 00 06 00 02 67 01 01 00 12 00 03 70 21 00 00 07 18 FA 0A 00 00 10 01 02 67 1D 7B, checksum=7B, error=false
2019-04-26 21:50:35.740 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=26370/1, destinationAddress=0/1, profile=0104, cluster=0006, addressMode=null, radius=0, apsSecurity=false, apsCounter=00, payload=18 FA 0A 00 00 10 01]
2019-04-26 21:50:35.743 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=250, commandId=10]
2019-04-26 21:50:35.747 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [On/Off: 26370/1 -> 0/1, cluster=0006, TID=FA, reports=[Attribute Report: attributeDataType=BOOLEAN, attributeIdentifier=0, attributeValue=true]]
2019-04-26 21:50:35.750 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ReportAttributesCommand [On/Off: 26370/1 -> 0/1, cluster=0006, TID=FA, reports=[Attribute Report: attributeDataType=BOOLEAN, attributeIdentifier=0, attributeValue=true]]
2019-04-26 21:50:35.753 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ReportAttributesCommand [On/Off: 26370/1 -> 0/1, cluster=0006, TID=FA, reports=[Attribute Report: attributeDataType=BOOLEAN, attributeIdentifier=0, attributeValue=true]] ZigBeeTransaction [queueTime=6977, state=DISPATCHED, sendCnt=1, command=IeeeAddressRequest [0/0 -> 65533/0, cluster=0001, TID=85, nwkAddrOfInterest=41352, requestType=0, startIndex=0]]
2019-04-26 21:50:35.756 [DEBUG] [.zsmartsystems.zigbee.ZigBeeEndpoint] - 26370/1: Cluster 6 not found for received endpoint command

cdjackson, 2019-04-26:
It’s hard to say why the is happening with such a small log. Please provide the network XML, or better still the initialisation log when the device was included into the network. Probably it wasn’t completed properly and the series weren’t discovered completely.

achille2k, 2019-04-27:
Please check network XML
[lumisensor-magnet.xml.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/3123414/lumisensor-magnet.xml.txt)

And log when device pair with network (I see msg "Error AF message listener notify. java.lang.NullPointerException: null") and inbox of paper UI very slow show device with label vendor and modelId
[log.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/3123415/log.txt)

cdjackson, 2019-04-27:
You should ensure you are running a recent version of the binding. 2.4 is very old now and a lot of new features were added

achille2k, 2019-04-27:
Sorry I installed binding latest version 2.5

	Xiaomi Mi Smart Home Door / Window Sensor (MCCGQ01LM)
OH 2.5.0 Snapshot Zigbee:
286 │ Active │ 80 │ 2.5.0.201904261422 │ org.openhab.binding.zigbee
287 │ Active │ 80 │ 2.5.0.201904150817 │ org.openhab.binding.zigbee.console
288 │ Active │ 80 │ 2.5.0.201904150817 │ org.openhab.binding.zigbee.xbee
289 │ Active │ 80 │ 2.5.0.201904150817 │ org.openhab.binding.zigbee.cc2531
290 │ Active │ 80 │ 2.5.0.201904150817 │ org.openhab.binding.zigbee.console.ember
291 │ Active │ 80 │ 2.5.0.201904150817 │ org.openhab.binding.zigbee.console.telegesis
292 │ Active │ 80 │ 2.5.0.201904150817 │ org.openhab.binding.zigbee.ember
293 │ Active │ 80 │ 2.5.0.201904150817 │ org.openhab.binding.zigbee.telegesis
316 │ Active │ 80 │ 1.1.11.SNAPSHOT │ com.zsmartsystems.zigbee
317 │ Active │ 80 │ 1.1.11.SNAPSHOT │ com.zsmartsystems.zigbee.dongle.ember
318 │ Active │ 80 │ 1.1.11.SNAPSHOT │ com.zsmartsystems.zigbee.dongle.telegesis
319 │ Active │ 80 │ 1.1.11.SNAPSHOT │ com.zsmartsystems.zigbee.console
320 │ Active │ 80 │ 1.1.11.SNAPSHOT │ com.zsmartsystems.zigbee.console.ember
321 │ Active │ 80 │ 1.1.11.SNAPSHOT │ com.zsmartsystems.zigbee.dongle.xbee
322 │ Active │ 80 │ 1.1.11.SNAPSHOT │ com.zsmartsystems.zigbee.console.telegesis
323 │ Active │ 80 │ 1.1.11.SNAPSHOT │ com.zsmartsystems.zigbee.dongle.cc2531

cdjackson, 2019-04-27:
Ok, great.

I see two things in the log - this device appears to be using some manufacturer specific features which are not supported at this time. Secondly, I’d need to see a log during the initialisation to see what is happening, and this log doesn’t provide that. The device was already included in the network I believe as there were no discovery operations performed.

achille2k, 2019-04-28:
Hi cdjackson,
you see two things in the log, because one thing of Coordinator (00124B001938A85F). ieeeAddr device is: 00158D0002EA2DD0

Today I can't include devce in the network, because Node Discovery not completed. yesterday device can get vendor is LUMI. but today cannot. It seems that the error message "Error AF message listener notify. java.lang.NullPointerException: null" caused this problem.
Logs
[log.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/3124570/log.txt)

cdjackson, 2019-04-28:
When I said I saw two things , I meant two points - ie the two pints I made. I was not referring to things in the sense of devices.

achille2k, 2019-04-28:
Sorry for my bad english :(

cdjackson, 2019-04-28:
It’s not your bad English - just my confusing English ;)

cdjackson, 2019-04-28:
The exception is caused by the use of some non-standard attributes which are not currently supported. This will be added in future, but at the moment the support for this is simply not available.

achille2k, 2019-04-28:
I think some thing went wrong with lower lib, and now I'll try understand it. I used a lot of device (Xiaomi) made from china. With Xiaomi Aqara brand (LUMI vendor) working perfect, but Xiaomi Mijjia brand (LUMI vendor) cannot completed discovery.

cdjackson, 2019-04-28:
Yes - it’s as I said - the low library doesn’t support manufacturer specific commands at the moment. This is being added, but it’s not available at the moment. This device is using non-standard features that are not supported.

achille2k, 2019-04-28:
Thank you so much for quick your responce!

cdjackson, 2019-04-28:
Hopefully in a few weeks this general support should be added, then we can take a look at the specific issue with this device.

cdjackson, 2020-01-03:
This should be closed by #517 which should now dynamically add any clusters that are unknown.

closed 2020-01-03 by cdjackson