openhab/org.openhab.binding.zigbee #432: NPE in some converters when calculating updated attributes

TomTravaglino, 2019-05-03:
Caused by #427 and #409 

I'm experiencing a NPE in the method `ZigBeeConverterMeasurementPower.attributeUpdated(...)` thrown by this statement:
```
BigDecimal valueInWatt = BigDecimal.valueOf(value * multiplier / divisor);
```
The reason is that `multiplier` and `divisor` are both null. This happens because since #427 these variables are initialized in the method `initializeDevice()` which since #409 is only invoked on the very first initialization and then only `initializeConverter()` will be called on startup. Such variables should be better initialized in the method `initializeConverter()`.

Other converters which currently use to initialize variables in `initializeDevice()` will need to be changed too (didn't check them all).

cdjackson, 2019-05-03:
Please can you provide a PR for this?

cdjackson, 2019-05-03:
Duplicate of #430

PR #434 (2019-05-04): Refactoring of some converters to always initialize variables
Resolves #432 

Some converters used to initialize variables on device initialization in
order to handle attributes updates. After the initilization has been
been split into a device and a converter initialization these variables
were no longer initialized when only the converter was initialized. This
made the initialization run into NPEs. This commit ensures that the
initialization is properly done when only the converter is initialized.
closed 2019-05-04 by cdjackson