openhab/org.openhab.binding.zigbee #436: Read device configuration and put it into thing configuration

TomTravaglino, 2019-05-08:
My case: I have a thing type for motion sensor. The thing type has a config description containing properties for sensitivity, such as `PIRUnoccupiedToOccupiedThreshold`. When the thing is created its config contains the properties from the config description but their values are different from the corresponding attributes on the device. In order to sync them I would like to read the attributes from the device and update the thing config on the very first thing initialization.

cdjackson, 2019-08-10:
@TomTravaglino is this something you are working on and we should expect a PR?

TomTravaglino, 2019-08-20:
We solved that in our solution. We can close this.

closed 2019-08-20 by TomTravaglino