openhab/org.openhab.binding.zigbee #439: warning_device is defined as readOnly but needs to be r/w

DirkSagwitz, 2019-05-13:
`	
	<channel-type id="warning_device">
...
		<state readOnly="true" />
...
         </channel-type>
`

cdjackson, 2019-05-13:
Please provide more information - why do you want this, what device etc.

DirkSagwitz, 2019-05-13:
devices are e.g. bitronvideo 902010/24, 902010/24a, 902010/29, 902010/24a
this channel is intended to control the siren e.g. to swich it on/off but as it is read only you cannot control it

hsudbrock, 2019-05-13:
The channel for warning devices is intended to be used by sending commands to the channel, and not by updating the state of the channel. This is why the state is marked as read-only.

The reason for this is that, from the ZigBee cluster library perspective, warning devices do not provide a state (like, e.g., "warning is on" or "warning is off"). It is only possible to trigger a warning by sending a command to the warning device.

From the openHAB perspective, IMHO, it would be best to model this with a channel that has no state - but  that's, to my knowledge, currently not possible. Hence, the channel has a state which is simply not intended to be written.

Does this sound plausible, or is there a reason why the channel state should be writable that I do not see?

DirkSagwitz, 2019-05-13:
@hsudbrock  it's not the channel which indicated whether there is a warning or not, it's the siren channel which we need to be able to control. How can we control the siren if it this channel is set to read only? I do not understand ...

hsudbrock, 2019-05-13:
> How can we control the siren if it this channel is set to read only?

The channel is not set to read only, it is the channel state that is set to read only. (As far as I know, it is not a concept in openHAB to set channels to read-only, as one cannot read or write a channel.)

Starting a warning is done by sending a command to the item that is linked to the channel for the warning device. (There is even a command description provider that provides possible commands, quite similar to the state description providers that are used in some bindings.)

As background to clarify this a little more: It is possible to interact with an item in (at least) two ways: (a) Updating the item state and (b) sending a command to the item (compare, e.g., the REST API for items, the docs from the formerly used ESH are here: https://www.eclipse.org/smarthome/rest/index.html). For warning devices, you can use (b). (Why not to use (a) is what I tried to explain above.)

DirkSagwitz, 2019-05-13:
@hsudbrock as you know we use item type and tagging to derive the functionality which is provided by the item. In this case this mechanism no longer works as readOnly items provide sensor values and r/w items provide actor items :-(

hsudbrock, 2019-05-13:
@DirkSagwitz I argued above why I think that the current modeling of the warning device channel state as read-only is sensible. Do you think that my argument is not ok, and, hence, the state description should be changed in the openHAB ZigBee binding? In this case, why do you think that the argument is not ok? Or do you think that my argument is ok, and the issue is due to the mechanism that you describe? In this case, why not simply change the state description in your integration of the binding (without touching the openHAB ZigBee binding)?

afuechsel, 2019-05-14:
From an enduser perspective it is irrelevant whether or not the item has a state or even a modifiable state. A UI normally only sends commands to items and it does this for all items, that are *not* read-only. Now we have an item that is read-only but we still could send commands to it. This is contradictory to the original concept. 

From a technical POV I can follow the explanation from @hsudbrock but this would still be a flaw in the concept. To make it clear, we would need to extend the read-only concept into a three-state-logic: 

* `read-write`: Item has a state, that could be read and written to; item would accept commands
* `read-only`: Item has state that could be read but not written to; item would not accept commands
* `write-only`: Item does not have any state; Item would only accept commands (fire and forget) 

@kaikreuzer wdyt?

kaikreuzer, 2019-05-14:
I'm not sure I grasped all details of the discussion. So this channel does not have a state, but only accepts commands?
For such situations, we have usually a "command" channel, which accepts String commands and which handles ALL commands that the device might offer. autoupdate is set to false, so that the state always stays NULL (and it absolutely ok to read this state as it anyhow does not contain any information, so no need for "write-only").
The cleaner solution, which had been introduced last summer is to define ThingActions for such functionality. This is helpful for everything that you anyhow wouldn't want to display in a UI per se, but rather only use it from within rules.

hsudbrock, 2019-05-14:
Thanks @kaikreuzer , this more or less sums up what we discussed last year when I implemented this. I decided for a "command" channel (instead of the cleaner thing actions) back then, as thing actions were not supported by the solution that @DirkSagwitz is working on. The currently implemented "command" channel is mostly as you describe above (with the exception that `autoupdate` is not set to `false`, but this could be added to make it nicer).

As the channel accepts String commands and has no useful state, I still think that it is useful to set the state to read-only in the state description (but I am happy to be convinced otherwise).

Why not simply remove this state description in your solution, @DirkSagwitz ? (If I recall correctly, it contains some other modifications to the channel types as well, where this modification could simply be added.)

cdjackson, 2019-08-10:
It seems from Hennings comments that this can be closed. If not, please reopen and update this with further information.

closed 2019-08-10 by cdjackson