openhab/org.openhab.binding.zigbee #440: ASH: ERROR received (code 81). Disconnecting.

openhab-5iver, 2019-05-13:
OH S1589
```
293 │ Active   │  80 │ 2.5.0.201905111054    │ org.openhab.binding.zigbee.cc2531
294 │ Active   │  80 │ 2.5.0.201905111054    │ org.openhab.binding.zigbee.xbee
295 │ Active   │  80 │ 1.1.11.SNAPSHOT       │ com.zsmartsystems.zigbee.dongle.telegesis
296 │ Active   │  80 │ 1.1.11.SNAPSHOT       │ com.zsmartsystems.zigbee
297 │ Active   │  80 │ 1.1.11.SNAPSHOT       │ com.zsmartsystems.zigbee.dongle.ember
298 │ Active   │  80 │ 1.1.11.SNAPSHOT       │ com.zsmartsystems.zigbee.dongle.xbee
299 │ Active   │  80 │ 1.1.11.SNAPSHOT       │ com.zsmartsystems.zigbee.dongle.cc2531
300 │ Active   │  80 │ 2.5.0.201905111054    │ org.openhab.binding.zigbee.telegesis
301 │ Active   │  80 │ 2.5.0.201905111054    │ org.openhab.binding.zigbee
302 │ Active   │  80 │ 2.5.0.201905111054    │ org.openhab.binding.zigbee.ember
```
I realize this error in the title is zsmartsystems, but I have been using these libraries without issue for quite a while, so I put the issue here. All of my devices are becoming unresponsive at least once each day. This has been happening since being migrated to BND. At first, it appeared to be that communication to the serial port was lost. The symptoms were the same and there were logs about losing serial. After adding my ports to EXTRA_JAVA_OPTS (which resolved the ports from disappearing from the dropdown), I started to get errors like in the attached log. I usually restart OH when this happens, but I found this morning that restarting all of the zigbee stuff seems to get it going again. However, after doing this today, zigbee stopped working within an hour and I had debug turned on. Things go bad at 07:19:31.604.

[zigbee.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/3172772/zigbee.zip)

cdjackson, 2019-05-13:
The error is cased by the NCP. There is only 1 way out of this, and that is to restart the connection. That should be done automatically (although I'm not sure if it's in the latest binding, but I thought it was).

openhab-5iver, 2019-05-18:
Were you able to determine if the automatic restart code is in the latest binding? If not, that could explain things. If it is, then what does this mean? Bad hardware? I have a spare stick to try, if you think this is a possibility.

cdjackson, 2019-05-18:
Yes, I think it’s included. It was merged back in Jan/Feb timeframe if I remember correctly. It might not trigger properly though without more recent drivers.

openhab-5iver, 2019-05-24:
This is either an OHC issue (which unfortunately, may only affect Zigbee) or my hardware (which I haven't tried swapping out yet). I downgraded to...

```
302 │ Active │  80 │ 1.1.9                 │ com.zsmartsystems.zigbee
303 │ Active │  80 │ 1.1.9                 │ com.zsmartsystems.zigbee.dongle.ember
307 │ Active │  80 │ 2.5.0.201901280820    │ org.openhab.binding.zigbee
308 │ Active │  80 │ 2.5.0.201901280820    │ org.openhab.binding.zigbee.ember
```
... and in a couple hours the same thing happened...

```
2019-05-23 22:19:32.957 [WARN ] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: ERROR received (code 81). Disconnecting.
2019-05-23 22:19:34.894 [WARN ] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: ERROR received (code 81). Disconnecting.
2019-05-23 22:19:40.424 [WARN ] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: ERROR received (code 81). Disconnecting.
2019-05-23 22:19:45.076 [WARN ] [org.openhab.binding.zigbee.handler.ZigBeeThingHandler] - 000D6F00040494AA: Polling aborted due to exception 
java.util.concurrent.RejectedExecutionException: Task java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask@44aab016 rejected from java.util.concurrent.ScheduledThreadPoolExecutor@6e111783[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 21628]
        at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2063) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:830) ~[?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor.delayedExecute(ScheduledThreadPoolExecutor.java:326) ~[?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor.schedule(ScheduledThreadPoolExecutor.java:533) ~[?:?]
        at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.sendCommand(ZigBeeDongleEzsp.java:553) ~[?:?]
        at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendCommand(ZigBeeNetworkManager.java:706) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.send(ZigBeeTransactionManager.java:85) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.send(ZigBeeTransaction.java:129) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendTransaction(ZigBeeTransactionManager.java:74) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendTransaction(ZigBeeNetworkManager.java:1469) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.ZigBeeNode.sendTransaction(ZigBeeNode.java:798) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.ZigBeeEndpoint.sendTransaction(ZigBeeEndpoint.java:502) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.zcl.ZclCluster.send(ZclCluster.java:158) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.zcl.ZclCluster.read(ZclCluster.java:174) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.zcl.ZclCluster.read(ZclCluster.java:184) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.zcl.ZclCluster.readSync(ZclCluster.java:233) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at com.zsmartsystems.zigbee.zcl.clusters.ZclElectricalMeasurementCluster.getActivePower(ZclElectricalMeasurementCluster.java:518) ~[302:com.zsmartsystems.zigbee:1.1.9]
        at org.openhab.binding.zigbee.internal.converter.ZigBeeConverterMeasurementPower.handleRefresh(ZigBeeConverterMeasurementPower.java:97) ~[307:org.openhab.binding.zigbee:2.5.0.201901280820]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$2.run(ZigBeeThingHandler.java:491) [307:org.openhab.binding.zigbee:2.5.0.201901280820]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:?]
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:?]
        at java.lang.Thread.run(Thread.java:748) [?:?]
2019-05-23 22:19:45.979 [WARN ] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: ERROR received (code 81). Disconnecting.
2019-05-23 22:19:51.437 [WARN ] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: ERROR received (code 81). Disconnecting.
2019-05-23 22:19:56.887 [WARN ] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: ERROR received (code 81). Disconnecting.
2019-05-23 22:20:02.693 [WARN ] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - ASH: ERROR received (code 81). Disconnecting.
```

openhab-5iver, 2019-05-29:
Short followup... I had been using a manual install of the binding. After removing it and installing through addons.cfg, the binding hasn't had an issue. Maybe it's working because the libraries are at 1.1.6 or maybe the manual install was causing the problem.

openhab-5iver, 2019-05-31:
Another update... not using a manual install did not correct this. The same exception is occurring about once a day. Restarting OH for other things was masking the issue.

cdjackson, 2019-05-31:
It seems that in this situation, a restart is not solving the issue and we need to perform a reset at the ASH layer either before taking the system offline, or probably when the network is re-initialised.

Let’s close this and please open an issue on the ZSS repository to perform the ASH reset.

cdjackson, 2019-05-31:
Actually, on further investigation, the library already performs an ASH layer reset on startup, but it is rejected by the NCP. The docs for the reset command states -:

• Purpose: Requests the NCP to perform a software reset (valid even if the NCP is in the FAILED state).

The NCP is in the FAILED state, and yet it doesn’t reset.

```
2019-05-13 07:19:38.907 [DEBUG] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameRst []
2019-05-13 07:19:38.908 [DEBUG] [com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameError [version=2, errorCode=81, Error: Exceeded maximum ACK timeout count]
```

I’m not sure what can be done from here. I might need to raise this with Silabs.

closed 2019-05-31 by cdjackson

cdjackson, 2019-06-16:
@openhab-5iver were you planning on opening an issue on this in the ZSS repository? It's slow going with Silabs on this one, but it might be that I just need to retry the RESET command as apparently the first reset might just return the same error, and the second reset is where it will have reset from the first reset! I'm trying to get Silabs to confirm this but it would be good if you could open an issue as discussed above so we don't loose track of this.

openhab-5iver, 2019-06-16:
I took this...

> Let’s close this and please open an issue on the ZSS repository to perform the ASH reset.

> Actually, on further investigation, the library already performs an ASH layer reset on startup, but it is rejected by the NCP.

... to mean that an issue for ZSS would not be needed. Sounds like that was not the case, so I'll open one :smile:!

I have some newer information that may be of interest to you. This error has stopped after applying this workaround... https://github.com/openhab/openhab-core/pull/861. Discussed more in this issue... https://github.com/openhab/openhab-core/issues/829. Also, the startup and performance for zwave is much improved (nothing empirically tested, but noticeable). I haven't tried doing a network heal yet with this patch (https://github.com/openhab/org.openhab.binding.zwave/issues/1178).

So, the issue with zigbee may have been caused or exacerbated by openhab-core. The only issue I'm currently having is that I can't use a manual install of the zigbee binding, as it never finds the port, similar to what is/was occuring with zwave https://github.com/openhab/org.openhab.binding.zwave/issues/1165. I had hoped https://github.com/openhab/openhab-core/issues/805 would help with this. It looks like I'm/we are not able to use manual installations, but that may only effect people using both bindings or someone has put a curse on my system. I'd like to try rebuilding or reproducing in a test environment, but my attention is on automation.

I had a bulb drop yesterday and tried getting a manual zigbee and not manual zwave setup working today, but nothing I tried would work. The last result I got, trying to use non-manual installations of both bindings but 1.1.11 zigbee libraries (bulbs worked great with this one), was...

```
2019-06-16 14:44:58.369 [ERROR] [org.openhab.core.karaf.internal.FeatureInstaller] - Failed installing 'openhab-persistence-jdbc-mariadb, openhab-misc-openhabcloud, openhab-binding-network, openhab-binding-darksky, openhab-binding-caldav-command1, openhab-ui-habpanel, openhab-misc-ruleengine, openhab-transformation-jsonpath, openhab-binding-zwave, openhab-binding-zigbee, openhab-binding-wifiled, openhab-binding-feed, openhab-ui-restdocs, openhab-binding-mqtt, openhab-ui-basic, openhab-transformation-map, openhab-ui-classic, openhab-binding-kodi, openhab-binding-openweathermap, openhab-binding-caldav-personal1, openhab-binding-astro, openhab-ui-habmin, openhab-ui-paper, openhab-transformation-scale': Uses constraint violation. Unable to resolve resource org.openhab.binding.zigbee.ember [org.openhab.binding.zigbee.ember/2.5.0.201906161432] because it is exposed to package 'com.zsmartsystems.zigbee' from resources com.zsmartsystems.zigbee [com.zsmartsystems.zigbee/1.1.6] and com.zsmartsystems.zigbee [osgi.identity; osgi.identity="com.zsmartsystems.zigbee"; type="osgi.bundle"; version:Version="1.1.11.SNAPSHOT"] via two dependency chains.

Chain 1:
  org.openhab.binding.zigbee.ember [org.openhab.binding.zigbee.ember/2.5.0.201906161432]
    import: (&(osgi.wiring.package=com.zsmartsystems.zigbee)(version>=1.1.0)(!(version>=2.0.0)))
     |
    export: osgi.wiring.package: com.zsmartsystems.zigbee
  com.zsmartsystems.zigbee [com.zsmartsystems.zigbee/1.1.6]

Chain 2:
  org.openhab.binding.zigbee.ember [org.openhab.binding.zigbee.ember/2.5.0.201906161432]
    import: (osgi.wiring.package=org.openhab.binding.zigbee.handler)
     |
    export: osgi.wiring.package=org.openhab.binding.zigbee.handler; uses:=com.zsmartsystems.zigbee
  org.openhab.binding.zigbee [org.openhab.binding.zigbee/2.5.0.201906161432]
    import: (&(osgi.wiring.package=com.zsmartsystems.zigbee)(version>=1.1.0)(!(version>=2.0.0)))
     |
    export: osgi.wiring.package: com.zsmartsystems.zigbee
  com.zsmartsystems.zigbee [osgi.identity; osgi.identity="com.zsmartsystems.zigbee"; type="osgi.bundle"; version:Version="1.1.11.SNAPSHOT"]
```

cdjackson, 2019-06-17:
> I have some newer information that may be of interest to you. This error has stopped after applying this workaround... openhab/openhab-core#861 <https://github.com/openhab/openhab-core/pull/861>. Discussed more in this issue... openhab/openhab-core#829 <https://github.com/openhab/openhab-core/issues/829>. Also, the startup and performance for zwave is much improved (nothing empirically tested, but noticeable). I haven't tried doing a network heal yet with this patch (openhab/org.openhab.binding.zwave#1178 <https://github.com/openhab/org.openhab.binding.zwave/issues/1178>).
> So, the issue with zigbee may have been caused or exacerbated by openhab-core. The only issue I'm currently having is that I can't use a manual install of the zigbee binding, as it never finds the port, similar to what is/was occuring with zwave openhab/org.openhab.binding.zwave#1165 <https://github.com/openhab/org.openhab.binding.zwave/issues/1165>. I had hoped openhab/openhab-core#805 <https://github.com/openhab/openhab-core/issues/805> would help with this. It looks like I'm/we are not able to use manual installations, but that may only effect people using both bindings or someone has put a curse on my system. I'd like to try rebuilding or reproducing in a test environment, but my attention is on automation.

Sorry for the confusion. I opened an issue with Silabs a couple of weeks back. As it’s not connected to one of their “important” customers, it gets slow priority… 

Even if it has stopped, I’d like to understand it as it doesn’t seem right. Silabs support keep focussing on the reason for getting into the error state in the first place, but given that could be anything, I’m more focussed on how to recover from it given the docs states that the reset should recover from the failed state.

Thanks.

openhab-5iver, 2019-06-17:
Thank you Chris. Let me know if there's anything I can do to help or to reproduce the issue. I have some backups, but should be easy to setup a test environment too. 

You replied though email, so missed this edit. Do you have any plans on moving zigbee off of gnu.io? It's a long shot, but all the craziness started after the serial change for zwave. Maybe things will go back to normal if both bindings used the same one?

thekev, 2019-09-12:
+1 same issue. Restart does not seem to be happening.

cdjackson, 2019-09-12:
Unfortunately this is not something that can be resolved. Unless your dongle has a hardware reset and we know how to use that reset then there is no reliable reset mechanism as Silabs have advised that the ASH reset should not be relied upon.

schrieveslaach, 2019-12-04:
I did have the same issue with husbzb-1 stick. Every two to three days I had to reboot the system because of the ASH error described here. But since two weeks my OpenHAB distributions uses following script every 15 minutes and works just fine:

```bash
#!/bin/bash

ZIGBEE_STATUS=$(curl http://localhost:8080/rest/things/<put your stick's things id here> -H "Accept: application/json" | jq .statusInfo.statusDetail -r)
if [ "$ZIGBEE_STATUS" = "COMMUNICATION_ERROR" ]
then
	systemctl reboot
fi
```
Interestingly my OpenHAB is now running for two weeks without any issue. Do you think that is just a coincidence or do you think that the status request avoids the communication error?

cdjackson, 2019-12-04:
I’m not sure. The error itself can be recovered by the binding, but what was seen here is that sometimes the NCP gets into a state where the reset command will not work. This is confirmed by Silabs. In this condition, Silabs say that the only way out is a hard reset, which these sticks do not have. Therefore, I suspect that the script doesn’t help, but  it can’t hurt of course :)

openhab-5iver, 2019-12-05:
Using a recent OH snapshots (and their zigbee bi dings and libs), I have not seen this issue for a month or two.

TheNON75, 2020-03-27:
Lucky you, I do face to the same issue or at least error code and symptoms, but with a bitronvideo (ember) stick.
A difference is, disabling and then enabling the stick revolves the issue, but that is only a workaround

cdjackson, 2020-03-27:
Sorry, but I haven't seen your debug logs showing the problem that you have? Where are these please?

tknaller, 2020-03-27:
exactly the same here

i think it has to do with system load/disk load as it happens less often since i moved the download program to a drive different from openhab...but it still does

cdjackson, 2020-03-27:
I'm sorry, but there really is no point in posting silly comments like this if you are not prepared to provide information requested that could help to solve any issue!!!

tknaller, 2020-03-27:
i simply stated that i have the same issue like you with the same environment and behaviour and had a new idea/observation lately

what's with the attitude?

tknaller, 2020-03-27:
ah awesome github bug, showed TheNON75's comment as coming from from cdjackson :/

will enable debug logging and report back as soon as the ash error happens again!

cdjackson, 2020-03-27:
Sorry, but if people want to help, then please post logs showing the problem. There's no point in posting silly comments like "lucky you - I still have a problem" if people are not prepared to provide information to help.

It's not attitude - I'm simply trying get people to provide information that can help me to help you.

tknaller, 2020-03-27:
here is the openhab.log: https://termbin.com/tgcq

to find something in there i think it is useful to use the timestamps from events.log: https://termbin.com/5pfa

hope this helps

cdjackson, 2020-03-27:
Thanks. Can you describe the problem please as I don't see the error reported in this log.

Why do you say that the events log is useful - normally this is not needed and I would not look at it, but I'm trying to understand what you are reporting - is it the same issue as reported here (ie error code 81)?

TheNON75, 2020-03-27:
@cdjackson i have a regular OH log available with default logging levels. Could it be helpful or should I set something to debug level?

tknaller, 2020-03-27:
well actually this is the only relevant line of the events.log:
`2020-03-27 21:13:00.845 [hingStatusInfoChangedEvent] - 'zigbee:coordinator_ember:01382ACB' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR)`
because this is when stuff doesn't work anymore. 

and i understand why you couldn't find anything: the termbin has been cut off :/
hopefully this works better: https://gist.github.com/tknaller/13613638f17ac603065364d30f5ffba4
here that timestamp and the error can be found :)

as TheNON75 said the ember-thing is offline at that state. when i disable the thing in paper ui and enable it again it starts working again.

and as i stated before it appears to have something to do with cpu & disk load on the docker host (for me: synology - docker on esata ssd) - as long as everything is idle the ember never goes offline, but as soon as something is going on (especially large copies/downloads - everything on the synology is slow then!) the thing goes offline.

hope this is more helpful than my last try!

cdjackson, 2020-03-27:
Sorry but the default logging will not show anything at all - it needs debug logging as per the binding docs.

cdjackson, 2020-03-27:
@tknaller the log file does not cover this period which is why it doesn't have the error. The log finishes at 2107, and you report the error at 2113.

tknaller, 2020-03-27:
> @tknaller the log file does not cover this period which is why it doesn't have the error. The log finishes at 2107, and you report the error at 2113.

right, but the gist covers it: https://gist.github.com/tknaller/13613638f17ac603065364d30f5ffba4

TheNON75, 2020-03-27:
Ok, done. I will report if the issue appears.

However what Knaller wrote lets me assume at least that it is not an rpi specific issue

cdjackson, 2020-03-27:
> right, but the raw gist covers it:

No - as I said the log doesn't cover it - unless you've updated it, neither of them runs more than a minute or two and certainly not up to the time you mention.

tknaller, 2020-03-27:
last few lines of https://gist.githubusercontent.com/tknaller/13613638f17ac603065364d30f5ffba4/raw/f70edad47d44d00c4afc27fd106cd2b0fdd25ac0/zigbee.log
are
```
2020-03-27 21:13:08.705 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Adding supported server cluster 0006
2020-03-27 21:13:08.710 [DEBUG] [stems.zigbee.internal.ClusterMatcher] - ClusterMatcher adding server cluster 0006
2020-03-27 21:13:08.711 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Adding supported server cluster 0008
2020-03-27 21:13:08.712 [DEBUG] [stems.zigbee.internal.ClusterMatcher] - ClusterMatcher adding server cluster 0008
2020-03-27 21:13:08.713 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Adding supported server cluster 0402
2020-03-27 21:13:08.714 [DEBUG] [stems.zigbee.internal.ClusterMatcher] - ClusterMatcher adding server cluster 0402
2020-03-27 21:13:08.715 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Adding supported client cluster 0000
2020-03-27 21:13:08.716 [DEBUG] [stems.zigbee.internal.ClusterMatcher] - ClusterMatcher adding client cluster 0000
2020-03-27 21:13:08.717 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Adding supported client cluster 0019
2020-03-27 21:13:08.718 [DEBUG] [stems.zigbee.internal.ClusterMatcher] - ClusterMatcher adding client cluster 0019
2020-03-27 21:13:08.719 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Adding supported server cluster 0000
2020-03-27 21:13:08.720 [DEBUG] [stems.zigbee.internal.ClusterMatcher] - ClusterMatcher adding server cluster 0000
2020-03-27 21:13:08.721 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - ZigBee Initialise: Previous device configuration was: channel=CHANNEL_11, PanID=45135, EPanId=9809C53465138525
2020-03-27 21:13:08.722 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Config: zigbee_trustcentremode=TC_JOIN_SECURE
2020-03-27 21:13:08.723 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - TX EZSP: EzspSetPolicyRequest [policyId=EZSP_TRUST_CENTER_POLICY, decisionId=EZSP_ALLOW_PRECONFIGURED_KEY_JOINS]
2020-03-27 21:13:08.724 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
2020-03-27 21:13:08.725 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=1, ackNum=1, reTx=false, data=7D 00 55 00 01]
2020-03-27 21:13:08.730 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=1, ackNum=2, reTx=false, data=7D 80 55 00]
2020-03-27 21:13:08.731 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: Frame acked and removed AshFrameData [frmNum=1, ackNum=1, reTx=false, data=7D 00 55 00 01]
2020-03-27 21:13:08.731 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspSetPolicyResponse [status=EZSP_SUCCESS]
2020-03-27 21:13:08.732 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=2, notRdy=false]
2020-03-27 21:13:08.733 [DEBUG] [systems.zigbee.dongle.ember.EmberNcp] - EzspSetPolicyResponse [status=EZSP_SUCCESS]
2020-03-27 21:13:08.743 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - TX EZSP: EzspSetConcentratorRequest [enable=true, concentratorType=EMBER_HIGH_RAM_CONCENTRATOR, minTime=60, maxTime=3600, routeErrorThreshold=8, deliveryFailureThreshold=8, maxHops=0]
2020-03-27 21:13:08.744 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
2020-03-27 21:13:08.746 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=2, ackNum=2, reTx=false, data=7E 00 10 01 F9 FF 3C 00 10 0E 08 08 00]
2020-03-27 21:13:08.758 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=2, ackNum=3, reTx=false, data=7E 80 10 00]
2020-03-27 21:13:08.761 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: Frame acked and removed AshFrameData [frmNum=2, ackNum=2, reTx=false, data=7E 00 10 01 F9 FF 3C 00 10 0E 08 08 00]
2020-03-27 21:13:08.764 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspSetConcentratorResponse [status=EZSP_SUCCESS]
2020-03-27 21:13:08.765 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=3, notRdy=false]
2020-03-27 21:13:08.764 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - EzspSetConcentratorResponse [status=EZSP_SUCCESS]
2020-03-27 21:13:08.765 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - ZigBeeNetworkManager startup: reinitialize=false, networkState=INITIALISING
2020-03-27 21:13:08.766 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Dongle: Startup - reinitialize=false
2020-03-27 21:13:08.767 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Adding Endpoint: ProfileID=0104, DeviceID=0050
2020-03-27 21:13:08.768 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Adding Endpoint: Input Clusters   0000
2020-03-27 21:13:08.769 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Adding Endpoint: Output Clusters  0000
2020-03-27 21:13:08.770 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - TX EZSP: EzspAddEndpointRequest [endpoint=1, profileId=0104, deviceId=0050, appFlags=0, inputClusterList=0000, outputClusterList=0000]
2020-03-27 21:13:08.772 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
2020-03-27 21:13:08.773 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=3, ackNum=3, reTx=false, data=7F 00 02 01 04 01 50 00 00 01 01 00 00 00 00]
2020-03-27 21:13:08.780 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=3, ackNum=4, reTx=false, data=7F 80 02 00]
[... cut ...]
2020-03-27 21:13:08.805 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=5, ackNum=5, reTx=false, data=81 00 69]
2020-03-27 21:13:08.805 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspStackStatusHandler [status=EMBER_NETWORK_UP]
2020-03-27 21:13:08.806 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - Ember: Link State change to true, initialised=true, networkStateUp=true
2020-03-27 21:13:08.807 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - Ember: Link State change to true ignored.
```

cdjackson, 2020-03-27:
Thanks - this link provides the full log.

cdjackson, 2020-03-27:
There's something very wrong here since there is 20 seconds of the log missing - or the computer froze up maybe - it's hard to tell. This gets the system out of sync between the NCP and the host, and this can't be recovered from.

tknaller, 2020-03-27:
as i am apparantly really blind - can you tell me where the gap is? 

maybe i grepped to much, but i did `cat openhab.log | egrep -i "zigbee|ash|ember"` and was thinking that must suffice...

cdjackson, 2020-03-27:
```
2020-03-27 21:12:44.443 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=2, ackNum=1, reTx=false, data=28 90 3F 00 6A 9E 00 00 05 00 00 00 40 11 00 00 F1 8A 66 00]
2020-03-27 21:12:44.444 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspMessageSentHandler [type=EMBER_OUTGOING_DIRECT, indexOrDestination=40554, apsFrame=EmberApsFrame [profileId=0000, clusterId=0005, sourceEndpoint=0, destinationEndpoint=0, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=F1], messageTag=8A, status=EMBER_DELIVERY_FAILED, messageContents=]
2020-03-27 21:12:44.445 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=8A state=RX_NAK
2020-03-27 21:12:44.445 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=8A, state=RX_NAK, outstanding=2
2020-03-27 21:12:58.230 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction complete: ZigBeeTransaction [ieeeAddress=000D6F0014176F34 queueTime=43754, state=FAILED, sendCnt=2, command=ActiveEndpointsRequest [0000/0 -> 9E6A/0, cluster=0005, TID=8A, nwkAddrOfInterest=9E6A]]
2020-03-27 21:12:58.262 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=3, notRdy=false]
2020-03-27 21:12:58.236 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D0004236E61: Polling...
2
```

tknaller, 2020-03-28:
thank you!

ah yes well this hang might have been noticable even on the ssh connection, there was no other log which may have been grepped out neither. 
this happens while synology is under heavy disk usage. i don't know where to start looking for the culprit of that as it doesn't really matter to me if the system hangs for a short period of time - except for this binding giving up :( 
any pointers?

so if it happens - is there really no chance for the binding to recover? 
how about:
1. store the error somewhere so we can be sure the thing is offline because of this error
1. check if binding is online every 60 seconds
1. if it is in error state & the reason is this error (step 1) then take it offline & online (reinitialize)

or something similar

cdjackson, 2020-03-28:
Unfortunately once this error occurs, there is no possibility to recover. As described above, the error is that the NCP needs to be reset, and it's only possible to do this via a hardware reset. The binding does already try to reconnect and you can see this in your log -

```
2020-03-27 21:13:03.875 [DEBUG] [ding.zigbee.handler.ZigBeeSerialPort] - Serial port [/dev/ttyUSB0] is initialized.
2020-03-27 21:13:03.876 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: Connect
```

However the error condition persists, and there's no way out of this without a hard reset.

The only thing I can look at is to stop the error occuring in the first place, but if you say that this will occur on your hardware when it becomes busy, I'm not sure what I can do.

tknaller, 2020-03-28:
thank you for your investigation and effort.

what i don't quite understand is: you say a hardware reset is required but pressing a button in paper ui is sufficient (disable & enable), which isn't really hardware now or where am i wrong?

hardware reset for me would be something that has got physically to do with the ember stick, like unplugging and repluggig it

cdjackson, 2020-03-28:
Really? That is surprising since the binding already reinitialises completely - as you see in the log. However the error persists.

I discussed this at length with Silabs and they told me that a hardware reset was required and so far, this has also been the experience of others.

Please provide the log that shows this reset working.

cdjackson, 2020-03-28:
As mentioned, I've discussed this a lot with Silabs when this issue was first raised. Since I know the often people think this is either not true, or that it is wrong, I paste one of the messages here for visibility.

[image]

Unfortunately it is possible that when the NCP gets into this state, the only way out is a hard reset. Since as far as I know the current dongles on the market do not provide such a reset line, it is impossible for this to be done without powering down the device (ie removing it from the USB).

TheNON75, 2020-03-28:
Hi Chris,

I’ve installed my bitron dongle some months ago. In the beginning I had this issue approx once a month but in the last week 3 times or so. 
First time I did a restart of my rpi, but later only disable-enable, which always solved the situation. 
Now I am waiting for another occurrence so I can provide a log for analyzing

tknaller, 2020-03-28:
here's all of my openhab.lof https://gist.githubusercontent.com/tknaller/f2edf66fea3e18d32e73d16424b848b6/raw/17b2ddf4e65cad6d8562daf8bfb3f93104f2354e/openhab.log

disable/enable in paper ui:
```
2020-03-27 21:43:26.076 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - ZigBee network [zigbee:coordinator_ember:01382ACB] closed.
2020-03-27 21:43:27.113 [DEBUG] [ng.zigbee.ember.handler.EmberHandler] - Initializing ZigBee Ember serial bridge handler.
```

cdjackson, 2020-03-28:
So the problem you see is not this one. The binding does restart, and it reinitialises the NCP ok. This can also happen - it depends on how the NCP fails.

I suspect that the issue with the restart is this -:
https://github.com/openhab/org.openhab.binding.zigbee/pull/477

cdjackson, 2020-03-29:
@tknaller once #564 is merged it should help with your problem.

tknaller, 2020-03-29:
awesome thank you! which docker image will get this update? i am currently on openhab:2.5.3-snapshot with autoupdate enabled (ouroboros), do i need to switch to 2.5.4-snapshot?

cdjackson, 2020-03-29:
There are no updates of 2.5.3 - that version is already released so cannot be changed now. The current snapshot is 2.5.4-snapshot.

tknaller, 2020-03-29:
ah great thanks!

TheNON75, 2020-03-30:
Sounds great!
Is there a chance that it resolves my similar issue with bitronvideo?

tknaller, 2020-03-30:
bitronvideo = ember

cdjackson, 2020-03-30:
Exactly as @tknaller says - the devices are the same...

The issue I linked above improves the way the binding restarts if there is an error. It cannot work around the reset issue so if the NCP gets into a state where it is unsynchronised, as per the original purpose of this issue, then this state is not recoverable. However this should at least improve things.

TheNON75, 2020-03-31:
Thanks you for the confirmation, as soon as the next stable version is released, I will go for it