openhab/org.openhab.binding.zigbee #441: NPE in ZigBeeDongleEzsp when the ZigBee binding starts

TomTravaglino, 2019-05-17:
I'm not sure if this issue belongs here or in zsmartsystems. I'm opening it here as the NPE happens when I start the binding bundles.

This is the stacktrace of the NPE
```
Exception in thread "pool-116-thread-15" java.lang.NullPointerException
	at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.sendCommand(ZigBeeDongleEzsp.java:570)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendCommand(ZigBeeNetworkManager.java:733)
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.send(ZigBeeTransactionManager.java:441)
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendNextTransaction(ZigBeeTransactionManager.java:667)
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.queueTransaction(ZigBeeTransactionManager.java:370)
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendTransaction(ZigBeeTransactionManager.java:354)
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendTransaction(ZigBeeTransactionManager.java:332)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendTransaction(ZigBeeNetworkManager.java:1512)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.permitJoin(ZigBeeNetworkManager.java:1105)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.permitJoin(ZigBeeNetworkManager.java:1077)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.setNetworkStateRunnable(ZigBeeNetworkManager.java:1033)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.access$000(ZigBeeNetworkManager.java:135)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager$2.run(ZigBeeNetworkManager.java:1007)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
```

The NPE is thrown because in `ZigBeeDongleEzsp.sendCommand` the statement `executorService.execute(...)` is executed when `executorService` is still `null`. I debugged the code and could observe that `ZigBeeDongleEzsp.sendCommand` is called before `ZigBeeDongleEzsp.initialize` and thus before `executorService` is instantiated.
Unfortunately I can't provide useful logs because the NPE only shows up in the console but not in the logs. However I have copied the following stacks from the debug view of Eclipse. Maybe they are helpful to understand what happens.

This is the thread where `ZigBeeDongleEzsp.sendCommand` is called
```
Thread [pool-116-thread-426] (Suspended (breakpoint at line 570 in ZigBeeDongleEzsp))	
	owns: ZigBeeTransactionManager  (id=20200)	
	owns: ZigBeeNetworkManager  (id=20201)	
	ZigBeeDongleEzsp.sendCommand(int, ZigBeeApsFrame) line: 570	
	ZigBeeNetworkManager.sendCommand(ZigBeeCommand) line: 733	
	ZigBeeTransactionManager.send(ZigBeeTransaction) line: 441	
	ZigBeeTransactionManager.sendNextTransaction() line: 667	
	ZigBeeTransactionManager.queueTransaction(ZigBeeTransactionQueue, ZigBeeTransaction) line: 370	
	ZigBeeTransactionManager.sendTransaction(ZigBeeCommand, ZigBeeTransactionMatcher) line: 354	
	ZigBeeTransactionManager.sendTransaction(ZigBeeCommand) line: 332	
	ZigBeeNetworkManager.sendTransaction(ZigBeeCommand) line: 1512	
	ZigBeeNetworkManager.permitJoin(ZigBeeEndpointAddress, int) line: 1105	
	ZigBeeNetworkManager.permitJoin(int) line: 1077	
	ZigBeeNetworkManager.setNetworkStateRunnable(ZigBeeNetworkState) line: 1033	
	ZigBeeNetworkManager.access$000(ZigBeeNetworkManager, ZigBeeNetworkState) line: 135	
	ZigBeeNetworkManager$2.run() line: 1007	
	ThreadPoolExecutor.runWorker(ThreadPoolExecutor$Worker) line: not available	
	ThreadPoolExecutor$Worker.run() line: not available	
	Thread.run() line: not available	
```

This is the thread where `ZigBeeDongleEzsp.initialize`
```
Daemon Thread [ESH-thingHandler-3000] (Suspended (breakpoint at line 368 in ZigBeeDongleEzsp))	
	owns: EmberHandler  (id=20202)	
	ZigBeeDongleEzsp.initialize() line: 368	
	ZigBeeNetworkManager.initialize() line: 334	
	EmberHandler(ZigBeeCoordinatorHandler).initialiseZigBee() line: 423	
	EmberHandler(ZigBeeCoordinatorHandler).lambda$0() line: 504	
	272210.run() line: not available	
	Executors$RunnableAdapter<T>.call() line: not available	
	ScheduledThreadPoolExecutor$ScheduledFutureTask<V>(FutureTask<V>).run() line: not available	
	ScheduledThreadPoolExecutor$ScheduledFutureTask<V>.access$201(ScheduledThreadPoolExecutor$ScheduledFutureTask) line: not available	
	ScheduledThreadPoolExecutor$ScheduledFutureTask<V>.run() line: not available	
	WrappedScheduledExecutorService(ThreadPoolExecutor).runWorker(ThreadPoolExecutor$Worker) line: not available	
	ThreadPoolExecutor$Worker.run() line: not available	
	Thread.run() line: not available	
```

cdjackson, 2019-05-17:
I have seen this and think I’ve fixed this, but it might only be in one of my feature branches. I will take a look, but I changed this to instantiate the executor in the constructor rather that in the initialise method. 

I’d suggest to close here and open an issue in the ZSS repo.

TomTravaglino, 2019-05-20:
As you suggested I close this one and open an issue in ZSS.

closed 2019-05-20 by TomTravaglino