openhab/org.openhab.binding.zigbee #451: Lowes IRIS door sensor will not connect

tguless, 2019-06-22:
Currently Lowes IRIS door sensor does not fully connect. 

In Paper UI it states "Node has not completed discovery"

[image]

Here is what I have in the logs:
2019-06-22 01:40:29.190 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zigbee:device:22d564ee:000d6f000398cae0' to inbox.

There seems to be some special negotiation that needs to be done to connect these devices as described here: 
https://github.com/jamesleesaunders/PyAlertMe

It is derived off the work done here:
http://www.desert-home.com/2014/02/raspberry-pi-and-lowes-iris-smart-switch.html

Lowes discontinued their service, so there are many of these sensors out there that are completely orphaned and unusable:

https://www.ebay.com/sch/i.html?_from=R40&_trksid=m570.l1313&_nkw=lowes+door+sensor&_sacat=0

It refuses to pair with my Securifi home automation capable router or anything else.   

I was hopeful that I would have better luck with openhab, but unfortunately I have run into a similar issue.

cdjackson, 2019-06-22:
Please provide a debug log - it's impossible to know what is happening otherwise.

The ZS libraries will perform the required negotiation for a standard IAS device if that's the "special negotiation" you refer to? If it's non-standard, then that's another matter?

openhab-5iver, 2019-06-22:
The first generation Iris door sensors used a proprietary protocol. The second gen (purple box, rounded corners) were Zigbee certified, but they dropped off the network often, needed to be reset to rejoin, and would eat through a battery in a few months. Which version do you have?

tguless, 2019-06-23:
Logged into karaf console as stated here:
https://www.openhab.org/docs/administration/console.html

ssh -p 8101 openhab@localhost

I think I bumped up logs as stated here:
https://www.openhab.org/docs/administration/logging.html

log:set DEBUG ome.event.ThingStatusInfoChangedEvent
log:set DEBUG gbee.discovery.ZigBeeDiscoveryService
log:set DEBUG ig.discovery.internal.PersistentInbox
log:set DEBUG smarthome.event.ThingUpdatedEvent
log:set DEBUG org.openhab.bindin.zigbee

02:01:42.227 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 000D6F000398CAE0: Starting ZigBee device discovery
02:01:42.253 [INFO ] [ig.discovery.internal.PersistentInbox] - Added new thing 'zigbee:device:22d564ee:000d6f000398cae0' to inbox.
02:01:42.255 [INFO ] [smarthome.event.InboxAddedEvent      ] - Discovery Result with UID 'zigbee:device:22d564ee:000d6f000398cae0' has been added.
02:01:42.276 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 000D6F000398CAE0: Starting ZigBee device discovery
02:01:42.417 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 000D6F000398CAE0: Starting ZigBee device discovery
02:01:47.818 [INFO ] [smarthome.event.InboxRemovedEvent    ] - Discovery Result with UID 'zigbee:device:22d564ee:000d6f000398cae0' has been removed.
02:01:47.901 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:device:22d564ee:000d6f000398cae0' changed from UNINITIALIZED to INITIALIZING
02:01:47.919 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:device:22d564ee:000d6f000398cae0' changed from INITIALIZING to UNKNOWN
02:01:47.933 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zigbee:device:22d564ee:000d6f000398cae0' changed from UNKNOWN to OFFLINE: Node has not completed discovery
02:01:47.948 [INFO ] [arthome.event.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:22d564ee:000d6f000398cae0 changed to UNKNOWN.
02:01:48.233 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 000D6F000398CAE0: Starting ZigBee device discovery
02:01:48.235 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:device:22d564ee:000d6f000398cae0' has been updated.
02:01:54.543 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 000D6F000398CAE0: Starting ZigBee device discovery
02:01:54.589 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zigbee:device:22d564ee:000d6f000398cae0' has been updated.
02:01:54.592 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 000D6F000398CAE0: Starting ZigBee device discovery
02:01:54.732 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 000D6F000398CAE0: Starting ZigBee device discovery

tguless, 2019-06-23:
[image]

openhab-5iver, 2019-06-23:
That's a first gen sensor... proprietary Zigbee.

cdjackson, 2019-06-23:
If it’s not a zigbee device, that will make it difficult (or worse) as it’s probably not something I will look to add myself. In any case, the log is clearly too short to be of any use, and most likely not a debug log in that I see no debug information. If you’re unsure how to get debug enabled, please read the documentation as the commands you’ve used are incorrect.

cdjackson, 2019-09-15:
I'm not sure there's much we can do here so closing this. If there is any further information please feel free to reopen.

closed 2019-09-15 by cdjackson