openhab/org.openhab.binding.zigbee #470: Error reading from serial port

openhab-5iver, 2019-08-21:
Every time OH is shut down, I get the following error...

```
2019-08-21 08:46:11.132 [ERROR] [org.openhab.binding.zigbee.handler.ZigBeeSerialPort] - Error reading from serial port 
java.lang.InterruptedException: null
        at java.lang.Object.wait(Native Method) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeSerialPort.read(ZigBeeSerialPort.java:267) [262:org.openhab.binding.zigbee:2.5.0.201908112253]
        at org.openhab.binding.zigbee.handler.ZigBeeSerialPort.read(ZigBeeSerialPort.java:243) [262:org.openhab.binding.zigbee:2.5.0.201908112253]
        at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler.getPacket(AshFrameHandler.java:281) [225:com.zsmartsystems.zigbee.dongle.ember:1.2.1]
        at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler.access$200(AshFrameHandler.java:56) [225:com.zsmartsystems.zigbee.dongle.ember:1.2.1]
        at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler$1.run(AshFrameHandler.java:164) [225:com.zsmartsystems.zigbee.dongle.ember:1.2.
```

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used | HUSBZB-1 |
| openHAB version  | S1658 (but this has been happening since for a few builds now, probably since 1.2 libraries were added  |

```
223 x Active x  80 x 1.2.1                 x com.zsmartsystems.zigbee
224 x Active x  80 x 1.2.1                 x com.zsmartsystems.zigbee.dongle.cc2531
225 x Active x  80 x 1.2.1                 x com.zsmartsystems.zigbee.dongle.ember
226 x Active x  80 x 1.2.1                 x com.zsmartsystems.zigbee.dongle.telegesis
227 x Active x  80 x 1.2.1                 x com.zsmartsystems.zigbee.dongle.xbee
262 x Active x  80 x 2.5.0.201908112253    x org.openhab.binding.zigbee
263 x Active x  80 x 2.5.0.201908112253    x org.openhab.binding.zigbee.cc2531
264 x Active x  80 x 2.5.0.201908112253    x org.openhab.binding.zigbee.ember
265 x Active x  80 x 2.5.0.201908112253    x org.openhab.binding.zigbee.telegesis
266 x Active x  80 x 2.5.0.201908112253    x org.openhab.binding.zigbee.xbee
```

cdjackson, 2019-08-21:
This is just an interrupted exception, so might be fine. It's really hard to tell though if you don't provide the logs as I have no idea when this is happening.

openhab-5iver, 2019-08-21:
I'll be updating OH in the next couple days and will put the binding into debug to get the logs for you.

openhab-5iver, 2019-08-22:
Here is a log...  [zigbee_error_after_oh_stop.log](https://github.com/openhab/org.openhab.binding.zigbee/files/3529050/zigbee_error_after_oh_stop.log)

cdjackson, 2019-08-24:
Thanks. This is normal in this situation. I will remove the logging so that this doesn't appear in the log.

PR #471 (2019-08-25): Remove interrupted exception during port read
Closes #470
closed 2019-08-25 by cdjackson