openhab/org.openhab.binding.zigbee #472: Add support for PowerON attribute in Level Control cluster

curlyel, 2019-08-26:
Hi @cdjackson,
I recently came across the "HUE PowerOn Behavior" feature which came with HUE bulb's firmware by end of 2018.

First I thought, it is something vendor specific but then I found the discussion linked below where @ebaauw figured out that it uses an standard attribute in the Level Control Cluster:

(taken from: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1014#issuecomment-445455472 )

_"... Found the PowerOn attribute for the Level Control cluster: 0x0008/0x4000. A uint8, also standard. Managed to set it through the deCONZ GUI as well. ..."_

According to him, it's kind of standard accross (some) manufacturers:

_"...More fun: the IKEA Trådfri and ubisys D1 seem to support the PowerOn OnOff and PowerOn Level attributes. **They really must be standard, then**. The Trådfri colour temperature bulb does not support PowerOn Color Temperature, however._
_My innr, OSRAM, and GLEDOPTO lights don't support any of the PowerOn attributes. Of course, OSRAM have their own way of setting the power-on behaviour. ..."_

I know, we had a similar conversation regarding the power-on-behavior of the OSRAM Lightify bulbs which was assumed beeing vendor specific (as mentioned by Eric too).

Would be good to get the ability to configure the power-on-behavior at least for "standard compliant" devices in the ZigBEE binding.
( Sure: Would be great having something similar for the OSRAM devices too ;-) )

All use-cases where the HUE-bulbs (and others supporting these attributes) are connected to a switched power circuit (desk lamp, floor lamp, bulbs used in conventional installations with physical wall switches ) will benefit from this setting.

cdjackson, 2019-08-26:
These attributes aren't listed in the current public specification. I'm looking at getting ZBA membership which would give me access to the new spec.

Once I work out exactly what they do, then we can add them. The way the binding configuration works is to check what attributes are supported, and only add configuration for those that are supported in any device, so this should work dynamically.

cdjackson, 2019-08-26:
Level control cluster only appears to add support for attribute 0x4000 - startup level
On Off cluster adds support for attribute 0x4003 - startup state

There may be others, but I will add these as I've found a reference to them at least in some other docs I have.

curlyel, 2019-08-26:
> Once I work out exactly what they do, then we can add them

Great! Meanwhile I was reading in the public available ZCL specification...

If I understand the ZCL correctly, then at least the PowerON-_Level_ is yet there as an optional attribute:

[image]

The default is set to 0xfe, which means the "maximum level". If I got it right, any value between 0 and 0xfe defines the level to be set on power-on. 0xff would be ignored, same as if the attribute is not implemented.

So if the OnLevel attribute is readable (==implemented?) could you try writing a value within the supported range to provide support the OnLevel feature?

That's the original description (from: http://www.zigbee.org/wp-content/uploads/2014/11/docs-07-5123-04-zigbee-cluster-library-specification.pdf ):

**_3.10.2.2.4 OnLevel Attribute_**
_The OnLevel attribute determines the value that the CurrentLevel attribute is set to
when the OnOff attribute of an On/Off cluster on the same endpoint is set to On. If
the OnLevel attribute is not implemented, or is set to 0xff, it has no effect. For
more details see 3.10.2.1.1._

cdjackson, 2019-08-26:
Yes, but these are already implemented as far as I know.

cdjackson, 2019-08-26:
Just to clarify, if these aren't provided as configuration options in your system, then it probably means the device doesn't support them. As I said - the binding will detect what attributes are supported and only present the ones that can be configured to avoid confusion (although it might be causing some here I think).

cdjackson, 2019-08-26:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/725

cdjackson, 2019-08-26:
And I just confirmed that the binding already supports these attributes.

ebaauw, 2019-08-26:
This is the level when turning the light on over ZigBee.  Hue lights don’t support this, most other lights do, in volatile memory (so it’s lost when power-cycling the light).  The level when powering the light is a different attribute, 0x4000, which is stored in non-volatile memory.

curlyel, 2019-08-26:
Thank's Erik for clarifying 👍

curlyel, 2019-08-26:
> probably means the device doesn't support them

Firmware to old? So I brought up my dusty HUE bridge to see if there are updates for my bulbs...
There were indeed firmware updates available and got installed. Now running:

```
LCT001 Version 5.127.1.26581
LCT007 Version 5.127.1.26420
```
PowerOn behavior setting is available for both generation bulbs in the HUE app...

_Back to the Zigbee binding:_
The only shown configuration parameter in Habmin is the _Default Transition Time_ (tested with one of the newer bulbs LCT007):

[image]

Anyway - I've seen your recent add to the library you have linked above. Happy to try this once merged. Let me know ;-)

cdjackson, 2019-08-26:
> Firmware to old?

Maybe, or it may just be that it's not supported. Just because it's in the standard doesn't mean that all devices are required to support these things - even if they are new.

The binding detects what attributes are supported and only presents configuration parameters to the user for supported attributes. So, either the detection of the attributes failed, or they are just not supported.

cdjackson, 2019-08-27:
Yes, I need to update the binding to make this configurable. Currently I'm having problems with the IDE which I spent a few hours last night trying to resolve. I will continue to try and get the IDE working tonight and once I do it's not too hard to add this configuration - the big problem these days is fighting with the development environment.

https://github.com/openhab/openhab-distro/issues/927#issuecomment-525048172

curlyel, 2019-08-27:
Some of my HUE bulbs had indeed a ancient firmware. They haven't seen their HUE bridge for years ;-)
So they **did not support** the PowerON behavior setting even in the HUE app..

Now I have updated them all and can confirm, that the PowerON -_level/color/colortemp_ **is working** as expected **in the HUE app now**. So basically, the feature is there in the current firmware.

For having the PowerON-setting(s) shown by the Zigbee binding too, I should probably wait for your recent adds to the lib (attrs 0x4000 and 0x4003) getting available - right?

curlyel, 2019-08-27:
> problems with the IDE

I know. I follow the linked issue too since weeks ... :-)
Thanks a lot for your persistence! I hope you can keep it. 👍

cdjackson, 2021-11-11:
From what I can see the OnLevel attribute is now available in the binding and included in the LevelControl configuration so I'll close this.

closed 2021-11-11 by cdjackson