openhab/org.openhab.binding.zigbee #473: Support Innr SP 120 Smart Plug

weakfl, 2019-08-29:
### Outline

Hi Chris,

I am trying to add an [Innr SP 120](https://shop.innrlighting.com/en/shop/134/smart-plug-) smart plug to the binding.
It is discovered as `Unkown Device` and adds a dimmer channel, which is not functional.

An openhab debug log and a device.xml are attached below.
I have also captured a sniffer log, lmk if you need it and I will send it per mail.

These plugs are really nice, so it would be great if support could be added.

Thanks!

refs #268

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |  Telegesis                                                                          |
| openHAB version  | 2.5.0~S1668-1    |
| Devices          | Node 18128 46D0 00158D0003442C5D                                  |

### Logs

[innr_sp120_openhab_20190829.log.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/3554393/innr_sp120_openhab_20190829.log.zip)

[00158D0003442C5D.xml.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/3554449/00158D0003442C5D.xml.zip)

cdjackson, 2019-08-29:
Looking at the XML file for this device it is using the ZLL profile which the binding is not currently supporting.

weakfl, 2019-08-29:
I see, thanks Chris. Do you have plans to add ZLL support?

I'm looking for options to replace my bulky 433mhz plugs, and the Innr ones are the smallest I could find ;)

If not, no big deal. So feel free to close as you see fit…

cdjackson, 2019-08-29:
The change to support this is very simple - the not so simple part is working out how to work with all the different profiles at the same time. I did start to look at this as it may be needed for another project that uses Smart Energy profile so let me have a think about it.

weakfl, 2019-09-04:
@cdjackson 

Chris, today I've migrated my network from the Telegesis coordinator to the the Ember EM3588-LR and the SP 120 is now working perfectly.

It was immediately discovered with the correct device name (and not as _Unknown Zigbee Device_) and provides 4 channels (dimmer, activepower, current and voltage), which are all working.

If you're ok, I'll open a PR to update the readme and add it to the list of supported devices (with a note that it doesn't work with the Telegesis stick).

cdjackson, 2019-09-04:
Thanks for the info - yes - a PR to reflect that in the docs would be great - thanks.

closed 2019-09-04 by cdjackson

demichve, 2019-10-24:
@weakfl 
You are sure, that dimming is supported by SP 120? I also use an Ember Stick and the SP 120 has been discovered with all 4 channels. But dimming does not work. I guess, this is not supported by the Hardware. The other channels work.

weakfl, 2019-10-24:
@demichve I didn't mean to say that dimming is supported, just that the device has a dimmer channel. I didn't even try to dimm it, I'm just sending ON/OFF commands.

PR #476 (2019-09-04): Add 'Innr SP 120' to README
add 'Innr SP 120' to list of supported devices and improve notes style, closes #473
commit 54fe68ce (2019-09-04): add 'Innr SP 120' to list of supported devices and improve notes style, closes #473 (#476)