openhab/org.openhab.binding.zigbee #478: Openhab container crashing when accessing Dongle

vKnmnn, 2019-09-06:
---
### Outline
I'm having a hard time getting my CC2531 USB dongle to work. 
I'm running the openhab:latest-alpine image on an archlinux x86_64 box.
Plugging in the USB dongle, the startup of the container never completes. I'm able to pull a crashlog from the container.
AFAIK the config should be correct: the Dgnu.io serial port option is set via ENV, container running in privileged mode, and the device attached via --device=/dev/ttyACM0.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |     CC2531                                            |
| openHAB version  | 2.4     |
| Hardware         |      x86_64  1,6Ghz                                      |
| Memory           |       4GB RAM          |
| Java version     |         i guess java 8, since it's in the container                        |
| Devices          | CC2531 on /dev/ttyACM0                                  |
| Zigbee dongle FW| CC2531ZNP-Pro-Secure_Standard.hex|

coordinator.things
```
Thing zigbee:coordinator_cc2531:stick1 "Zigbee USB Stick" [zigbee_port="/dev/ttyACM0", zigbee_baud=115200, zigbee_flowcontrol=0, zigbee_childtimeout=864000, zigbee_concentrator=0, zigbee_meshupdateperiod=86400, zigbee_panid=83126, zigbee_extendedpanid="237f25da51631c6f", zigbee_networkkey="36 f5 a8 d6 cc 93 ae 63 7a 7b 2c 40 cd da 5b 87", zigbee_powermode=1, zigbee_txpower=8]
```
### Logs
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00000000000036f6, pid=17, tid=0x00007f26d724ab20
#
# JRE version: OpenJDK Runtime Environment (8.0_212-b04) (build 1.8.0_212-b04)
# Java VM: OpenJDK 64-Bit Server VM (25.212-b04 mixed mode linux-amd64 compressed oops)
# Derivative: IcedTea 3.12.0
# Distribution: Custom build (Mon Aug 12 12:54:46 UTC 2019)
# Problematic frame:
# C  0x00000000000036f6
#
# Core dump written. Default location: /openhab/userdata/core or core.17
#
# If you would like to submit a bug report, please include
# instructions on how to reproduce the bug and visit:
#   https://icedtea.classpath.org/bugzilla
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x0000560f2fab8800):  JavaThread "ESH-thingHandler-4" daemon [_thread_in_native, id=307, stack(0x00007f26d714a000,0x00007f26d724aae0)]

siginfo: si_signo: 11 (SIGSEGV), si_code: 1 (SEGV_MAPERR), si_addr: 0x00000000000036f6

Registers:
RAX=0x0000000000000000, RBX=0x0000560f2fab89e0, RCX=0x00007f26fc0fe010, RDX=0x00007f26fc05d42c
RSP=0x00007f26d7248ca8, RBP=0x0000560f30724b20, RSI=0x0000000000000900, RDI=0x0000560f30724b20
R8 =0x0000560f30724b2c, R9 =0x0000560f30724b20, R10=0x00000000c4021658, R11=0x0000000000000206
R12=0x0000000000000000, R13=0x00007f26d7248e28, R14=0x0000000000000001, R15=0x0000000000000011
RIP=0x00000000000036f6, EFLAGS=0x0000000000010202, CSGSFS=0x002b000000000033, ERR=0x0000000000000014
  TRAPNO=0x000000000000000e

Top of Stack: (sp=0x00007f26d7248ca8)
0x00007f26d7248ca8:   00007f26d7dab8c9 00007f26d7248d50
0x00007f26d7248cb8:   00007f2600000001 0000560f308adce0
0x00007f26d7248cc8:   00007f26d9d48408 0000560f2fab8800
0x00007f26d7248cd8:   0000000000000004 00007f26d7248d50
0x00007f26d7248ce8:   00007f26fbaa1286 0000560f308adce0
0x00007f26d7248cf8:   00007f26d7248d30 0000560f2fab8800
0x00007f26d7248d08:   00007f26fbaa12f2 00007f26d7248d50
0x00007f26d7248d18:   00007f26fbbe0f8e 0000560f2fab8800
0x00007f26d7248d28:   00007f26fbccf706 0000560f2fab8800
[... 1585 lines cut ...]
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt topoext perfctr_nb bpext perfctr_llc hw_pstate proc_feedback ssbd vmmcall bmi1 xsaveopt arat npt lbrv svm_lock nrip_save tsc_scale flushbyasid decodeassists pausefilter pfthreshold overflow_recov
bugs		: fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 3295.12
TLB size	: 1024 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate proc_feedback

Memory: 4k page, physical 3930880k(505240k free), swap 0k(0k free)

vm_info: OpenJDK 64-Bit Server VM (25.212-b04) for linux-amd64 JRE (1.8.0_212-b04), built on Aug 12 2019 13:11:10 by "buildozer" with gcc 8.3.0

time: Fri Sep  6 14:30:23 2019
timezone: CEST
elapsed time: 47 seconds (0d 0h 0m 47s)

```

cdjackson, 2019-09-08:
I'm afraid I've got no idea about this - sorry. I would suggest to use a newer binding that 2.4 which is now very old. I doubt it will really help though since presumably this issue is not really related to the binding since the serial port access is all done in other bundles.

I don't think there's a lot that we can do here.

vKnmnn, 2019-09-08:
i switched to openhab/openhab:latest and i don't get any crashes anymore.
it seems that openjdk does something unexpected.

cdjackson, 2019-09-08:
Ok, thanks for the feedback. I guess we will close this here.

closed 2019-09-08 by cdjackson