openhab/org.openhab.binding.zigbee #479: Ikea Trådfri motion sensor (E1745) not working

ascona78, 2019-09-10:
# Outline

New Ikea Trådfri motion sensors (E1745) not working. Discovery works and things can be added through PaperUI.

Debug show OnWithTimedOffCommand, when sensor is triggered, but channels are not triggering.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |   Telegesis Zigbee USB Stick                                                                             |
| openHAB version  | 2.5.0-M3    |
| Hardware         |   Orangepi PC+                                                                             |
| Memory           |   1GB                                                                             |
| Java version     |   Zulu Embedded 8.25.0.76-linux-aarch32hf (build 1.8.0_152-b76)                                                                             |
| Devices          | 3 Ikea Trådfri motion sensors E1745                                   |

### Logs

`[DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: OnWithTimedOffCommand [On/Off: 4991/1 -> 0/1, cluster=0006, TID=0C, onOffControl=1, onTime=1800, offWaitTime=0]
[DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: OnWithTimedOffCommand [On/Off: 4991/1 -> 0/1, cluster=0006, TID=0C, onOffControl=1, onTime=1800, offWaitTime=0]`

cdjackson, 2019-09-11:
@ascona78 thanks - can I ask you to provide a slightly longer log. I'd like to see the two lines that come just before the one starting with ```RX CMD``` - there should be ```RX ZCL``` and ```RX APS``` from memory.

Thanks.

ascona78, 2019-09-11:
Of course!
I attached slightly longer log from before and after those lines.

By the way.. What do you think about RX Telegesis Data line between RX ZCL and RX CMD. Does it look ok to you? 
I'm not too familiar with zigbee.

[zigbee-ikea-e1745-snippet.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/3602095/zigbee-ikea-e1745-snippet.txt)

cdjackson, 2019-09-11:
Thanks.

I don't see anything wrong with the telegesis data. The telegesis protocol isn't super nice and it doesn't necessarily print well, so maybe that's what you're seeing. If you think something is an error then please post the line and I can comment.

ascona78, 2019-09-11:
Ok.. 
I thought these character in 08: EB part might be ok, because message data looks good.

But thanks for clarifying.

AfromD, 2019-09-13:
IIRC, I had a problem with a Tradfri motion sensor, where there was no solution. The sensor did not provide a motion channel to the network, in stead it used a direct link to the light bulb to control it.

cdjackson, 2019-09-13:
This is not too hard to fix. I have some problems with getting the IDE working (again!!) but once I get that resolved I will take a look at this.

ascona78, 2019-09-14:
> IIRC, I had a problem with a Tradfri motion sensor, where there was no solution. The sensor did not provide a motion channel to the network, in stead it used a direct link to the light bulb to control it.

For me Trådfri also provides only switch channel and not motion. Sometimes also batterylevel and batteryvoltage are provided.

cdjackson, 2019-09-14:
> For me Trådfri also provides only switch channel and not motion.

Is that a problem? Please remember that the binding looks at the services that the device provides - if the device doesn't provide the motion sensor features, then you don't get that. That doesn't make it a bug in the binding does it?

If you think there's a bug, then please be clear as to what the issue is so we can look at it.

ascona78, 2019-09-14:
No that's not a problem at all. I just made note to @AfromD reply.

ascona78, 2019-09-15:
Binding is still not triggering channel.
Attached log when motion sensor was triggered.

[zigbee-e1745-after-pr481-1-snippet.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/3614104/zigbee-e1745-after-pr481-1-snippet.txt)

cdjackson, 2019-09-15:
There has not been any change to the binding, so I don't really expect to see anything different.

ascona78, 2019-09-15:
Little update:
I have 3 E1745 motion sensors and now only one has 4 channels (switch, battery level, battery voltage and battery alarm). Other two have only switch.

If I restart that "4 channel" motion sensor by pressing down two of three buttons on back, it makes channel to update OFF. But movement doesn't make channel to trigger.

Those two other motion sensors do nothing even when restarting by pressing buttons on back.

In the very beginning of log I restarted sensor by pressing buttons.
[zigbee-e1745-after-pr481-2-restarted-sensor-snippet.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/3614283/zigbee-e1745-after-pr481-2-restarted-sensor-snippet.txt)

cdjackson, 2019-09-15:
As I said, there hasn’t been any change with this yet. The PR will hopefully be merged in a few days.

ascona78, 2019-09-15:
Yes, sorry.. I should have mentioned I tested PR 481.

cdjackson, 2019-09-15:
#481 won't change the way the channels are detected - it just handled the timed command. I think your issues are likely just problems with the discovery. I'll try and look at your log, but it won't be tonight as I'm trying to get some ZWave stuff done.

cdjackson, 2019-09-15:
I managed a quick look at the log.

Is ```54603``` the sensor? It's the only device where any commands are received in the log.  The log isn't an initialisation log so I can't commend on why channels may not be available.

Given the binding doesn't seem to be receiving ```OnWithTimedOffCommand```, this isn't related to #481 since that simply adds processing of that command.

cdjackson, 2019-09-21:
Support for ```OnWithTimedOffCommand``` is merged in #481 so closing this.

closed 2019-09-21 by cdjackson