openhab/org.openhab.binding.zigbee #501: Lupus small ZigBee temperature sensor

weakfl, 2019-10-08:
### Outline

@chris, I've successfully added a [Lupus small ZigBee temperature sensor](https://www.lupus-electronics.de/shop/en/Alarm-Smarthome/Smarthome-Home-Automation/LUPUS-Small-ZigBee-Temperatursensor-p.html) with one minor issue.

The device provides the following channels:
- batterylevel
- temperature
- batteryalarm
- humidity
- batteryvoltage

All of the channels are working, except `batteryalarm`.

When adding the device, the following showed up in the log:
```
09:45:54.019 [ERROR] [nverter.ZigBeeConverterBatteryPercent] - 000D6FFFFE653392: Error 0xffff setting server binding
09:46:18.032 [INFO ] [ing.zigbee.handler.ZigBeeThingHandler] - 000D6FFFFE653392: Channel zigbee:device:ba8eb59b:000d6ffffe653392:000D6FFFFE653392_1_batteryalarm failed to initialise device
```

Before capturing debug/sniffer logs, I wanted to make sure that there isn't a known issue with `batteryalarm` channels. Should it work and should I capture additional logs?

As I said, it's just a minor issue and I will submit a PR to add the sensor to the list of supported devices if that's ok with you?

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |  Ember EM3588-LR |
| openHAB version  | 2.5 S1709 |
| Devices          | 000d6ffffe653392 |

weakfl, 2019-10-08:
@cdjackson After noticing that the battery of the device is rapidly depleting (8% in 2 hours), I checked the device properties in HABmin and noticed the following:

[image]

I don't know why the device is reporting a wrong powersource, but sniffing the traffic with this device confirms that there is permanent communication going on.

So I guess the device never goes to sleep and hence the high power consumption?

Is there anything I can do to prevent this?

weakfl, 2019-10-08:
@cdjackson disregard my previous comment, I was filtering by the wrong network address in wireshark 🤦‍♂ 

Traffic is fine, reports are coming every 5 minutes. So it's probably just a bad battery...

cdjackson, 2019-10-08:
If the device thinks it’s on mains, then that is up to the device - there’s not a lot I can do to change that. 

Does it say if it’s a router or end device?

weakfl, 2019-10-08:
> Does it say if it’s a router or end device?

It says it's an end device. Maybe it just needed some time to work out the battery level, steady at 88% since 2 hours now.

Lmk if you need logs regarding the `battery_alarm` channel. Maybe it's just part of the cluster but not implemented since the device has `batterylevel` and `batteryvoltage` channels anyway?

Overall it's working fine.

cdjackson, 2019-10-08:
Battery monitors on these sort of devices tend to be pretty rubbish - they often use the battery voltage to estimate the energy in the battery and voltage may change due to load, or temperature etc. Maybe this dropped after first use and that caused the battery voltage to drop...

I don't suppose you can get a debug log showing the original bind error?

Yes - please feel free to update the supported device list - thanks.

weakfl, 2019-10-08:
> Maybe this dropped after first use and that caused the battery voltage to drop...

Yes, that seems very plausible.

> I don't suppose you can get a debug log showing the original bind error?

Sure, just had to remove it and make it join again. First error is around L294, but I noticed just now that it says `ZigBeeConverterBatteryPercent`, so it's probably unrelated to the `batteryalarm` channel. 
```
2019-10-08 16:55:04.237 [ERROR] [verter.ZigBeeConverterBatteryPercent] - 000D6FFFFE653392: Error 0xffff setting server binding
```

The second one from my initial post is around L465:
```
2019-10-08 16:55:28.339 [INFO ] [ng.zigbee.handler.ZigBeeThingHandler] - 000D6FFFFE653392: Channel zigbee:device:ba8eb59b:000d6ffffe653392:000D6FFFFE653392_1_batteryalarm failed to initialise device
```

I've also attached a sniffer log, filtered by old and new network address of the device and transport key removed :)
I guess you don't need all the traffic?

**OLD** network address: 0xACD4
**NEW** network address: 0x6C64
mac address: 000D6FFFFE653392

[zigbee_debug_log_20191008.txt.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/3703183/zigbee_debug_log_20191008.txt.zip)

[zigbee_wireshark_log_20191008.pcapng.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/3703221/zigbee_wireshark_log_20191008.pcapng.zip)

Ka0o0, 2020-09-21:
@weakfl what is your experience with the battery? I ordered one in June and had to replace the battery roughly after one month. I now got a packet sniffer and it seems that the sensor transmits a lot of `Data Request` packages as soon was it gets online, which is btw. 5 minutes and not 10 minutes.

weakfl, 2020-09-21:
@Ka0o0 I have two of them running and got 6-8 months of battery life out of them. Battery readings are quite off though, depending on location. I have one in a box where i store filaments for 3d printing and it showed 2% for months without dying.

Ka0o0, 2020-09-21:
Okay thanks for the info. So maybe it were my two batteries. I use the ikea 2032 batteries and I will try with another one.

mjhas, 2020-09-22:
I have the same issue with newly bought sensors. Battery is almost gone after 2 weeks (starting out at 90% and more) on several sensors, so I am pretty sure this is software/firmware related.

cdjackson, 2020-09-22:
> I am pretty sure this is software/firmware related

Why? What can you provide to help us solve your issue? It's pretty much impossible to help you if you don't provide information about your problem!

mjhas, 2020-09-23:
Sorry for not providing something here the log file from today with debug on
[openhab.log.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/5271565/openhab.log.txt)

And a view of the quick drop in battery level: 
[image]

cdjackson, 2020-09-23:
Can you tell me what device this is please so I can try to work out what device it is in the log.

mjhas, 2020-09-23:
Hope this helps: 
```
zigbee (Type=Bridge, Label=BV 2010/10 (ZigBee USB dongle), Bridge=null)
zigbee:device:0118E7EC:00158d000320af65 (Type=Thing, Label=LUMI lumi.weather, Bridge=zigbee)
zigbee:device:0118E7EC:086bd7fffef073ef (Type=Thing, Label=lupus5, Bridge=zigbee)
zigbee:device:0118E7EC:086bd7fffef06ed2 (Type=Thing, Label=lupus8, Bridge=zigbee)
zigbee:device:0118E7EC:00124b001c0cfd62 (Type=Thing, Label=powerswitch2, Bridge=zigbee)
zigbee:device:0118E7EC:14b457fffedf1564 (Type=Thing, Label=lupus9, Bridge=zigbee)
zigbee:device:0118E7EC:7cb03eaa0a0a366a (Type=Thing, Label=OSRAM Plug 01, Bridge=zigbee)
zigbee:device:0118E7EC:086bd7fffef086d1 (Type=Thing, Label=lupus6, Bridge=zigbee)
hue:group:ecb5fa07618b:1 (Type=Thing, Label=sleeping room (Room), Bridge=hue)
zigbee:device:0118E7EC:14b457fffedf156b (Type=Thing, Label=lupus4, Bridge=zigbee)
zigbee:device:0118E7EC:086bd7fffef08e26 (Type=Thing, Label=lupus3, Bridge=zigbee)
hue:0200:ecb5fa07618b:1 (Type=Thing, Label=Hue iris 1, Bridge=hue)
zigbee:device:0118E7EC:086bd7fffef073ca (Type=Thing, Label=lupus7, Bridge=zigbee)
zigbee:device:0118E7EC:086bd7fffef09e5b (Type=Thing, Label=lupus1, Bridge=zigbee)
hue:bridge:ecb5fa07618b (Type=Bridge, Label=Philips hue (10.10.10.3), Bridge=null)
zigbee:device:0118E7EC:086bd7fffef1c317 (Type=Thing, Label=lupus2, Bridge=zigbee)
```

cdjackson, 2020-09-23:
Not really. I want to know what devices you're having problems with the battery on. You've not said what they are - you just said "I have the same issue with newly bought sensors". I don't know what devices these are though.

mjhas, 2020-09-23:
The devices are the [Lupus small temperature sensors ](https://www.lupus-electronics.de/shop/en/advanced_search_result.html?keywords=12134). Seen in the above device list as lupus1-9, they all show the same behavior regarding battery level as seen in the plot above.