openhab/org.openhab.binding.zigbee #506: [Telegesis] Baud rate cannot be configured

petero-dk, 2019-10-12:
I have a Telegesis usb device where I have had the Baud rate on the device configured to a non-standard 115200 and not the standard 19200. When I change the baud rate on the device in openHAB the settings wont stick. It cannot connect and when I open the device settings again the baudrate is 19200 again. (When I reconfigure my device to 19200 baudrate via the Telegesis Terminal utility I can get it to work in openHAB

When I look at the source code it seems the device is hard configured with the default 19200. 

https://github.com/openhab/org.openhab.binding.zigbee/blob/c768e658b282eea32651b246ecf0856e4daf0f4d/org.openhab.binding.zigbee.telegesis/src/main/java/org/openhab/binding/zigbee/telegesis/internal/discovery/ZigBeeTelegesisUsbSerialDiscoveryParticipant.java#L76-L77

I am correct in that this is the place where the device is initiated with the default baud rate?

cdjackson, 2019-10-12:
The code that you reference is not related to opening the device. This is only related to automatic discovery of dongles.

The configuration of the Telegesis dongle uses the baud rate that you configure in the UI.

https://github.com/openhab/org.openhab.binding.zigbee/blob/c768e658b282eea32651b246ecf0856e4daf0f4d/org.openhab.binding.zigbee.telegesis/src/main/java/org/openhab/binding/zigbee/telegesis/handler/TelegesisHandler.java#L68

cdjackson, 2019-11-05:
@petero-dk shall we close this or is there further information available?

petero-dk, 2019-11-05:
Give me one week then I will do a little more testing, ok?

petero-dk, 2019-11-22:
Ok, i finally got to test a little more. 

(The starting point is a fully working openHAB with a standard telegesis usb stick.)

First I checked the baud rate on the usb device via the telegesis terminal:
```
Command: ATS12?
Response: 0510
```
This means 19200 BAUD and no command response.

Then I set the new BAUD rate
```
Command: ATS12=0C10
Response (closed connection)
```
This sounds good, reconnecting to the device with 115200 connects and works now ATS12? gives 0C10.

Before closing openhab I set the BAUD rate of the device to 115200 then shutdown openHAB, upon restarting the device I only get the following error in the zigbee log (will logging down to info)
```

22-Nov-2019 10:57:09.669 [ERROR] [rg.openhab.binding.zigbee.handler.ZigBeeSerialPort] - Serial Error: Port /dev/ttyUSB0 in use.
22-Nov-2019 10:57:09.670 [ERROR] [tems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis] - Unable to open Telegesis serial port
22-Nov-2019 10:59:13.987 [ERROR] [tems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis] - Unable to configure Telegesis dongle
```
Checking the configuration for the dongle in openHAB it says BAUD: 19200, then I change it again to 115200, and now the dongle goes online and connects the devices.

Fully working openHAB with 115200 dongle BAUD.

Restarting openHAB again. Zigbee fails to connect. 

The dongle thing has the BAUD rate set to 19200 again.

In conclusion. The binding works just fine with a baud rate of 115200 but it either resets it to 19200 on startup or does not save the configured value in paper ui?

petero-dk, 2019-11-22:
Ok, checked the jsondb after I have set the baud rate in paperui, it contains the 115200 value
```json
  "zigbee:coordinator_telegesis:04000BCC": {
    "class": "org.eclipse.smarthome.core.thing.internal.BridgeImpl",
    "value": {
      "label": "Telegesis USB Device (ZigBee USB dongle)",
      "channels": [],
      "configuration": {
        "properties": {
          "zigbee_port": "/dev/ttyUSB0",
          "zigbee_baud": 115200,
```

petero-dk, 2019-11-22:
For a temporary fix it is possible to manually set the configuration using a rule and the rest interface:

```
rule "Fix: Zigbee Device"
  when 
      System started 
  then
  val String json = '{"zigbee_baud": 115200}'
  logWarn("zigbee fix", json)
  sendHttpPutRequest("http://localhost:8080/rest/things/zigbee:coordinator_telegesis:04000BCC/config", "application/json", json)
end
```

cdjackson, 2019-11-22:
The other workaround of course is just to use 19k2 - this will have no significant impact on performance and is what most users of these dongles use.

petero-dk, 2019-11-22:
Right ;-) 

I have been reading through the code of the binding (again), and the only place I can see that 19k2 is set is in discovery result? Can that override an existing thing if it is rediscovered?

cdjackson, 2019-11-22:
This gets set when trying to disable echo which is done as part of the critical commanding.

My suggestion is to use 19k2 - it's what the manufacturer recommends and almost all users I know of are using (and I don't mean within OH).

petero-dk, 2019-11-22:
Then why have it as a settable configuration? If it is overriden anyway then it is just confusing?

cdjackson, 2019-11-22:
Sure, and it’s a bug that was introduced when fixing something else. My point is that there’s no major need for this and my suggestion is to leave it set to 19k2 to avoid problems. Changing will not help performance so there’s no real need for this in my view.

cdjackson, 2019-11-23:
I had another look at this, and it seems that the driver handles the bit setting correctly. Please can you provide a log showing your problem.

I also had a look at what you stated above showed the error - you said this was the problem -:

```
22-Nov-2019 10:57:09.669 [ERROR] [rg.openhab.binding.zigbee.handler.ZigBeeSerialPort] - Serial Error: Port /dev/ttyUSB0 in use.
```

This simply means that the serial port was not opened - it doesn't mean that the baud rate is wrong at all, so is not related to this issue.

If you provide a log showing the startup with 115k2 after the dongle was set to 115k2 I will take a look, but at the moment I can't see anywhere that the driver will change the baud rate.

petero-dk, 2019-11-23:
That was the only error I received. I have configured the logging like this:

```
# Zigbee
#
log4j2.logger.Zigbee.name = org.openhab.binding.zigbee
log4j2.logger.Zigbee.level = INFO
log4j2.logger.Zigbee.additivity = false
log4j2.logger.Zigbee.appenderRefs = Zigbee
log4j2.logger.Zigbee.appenderRef.Zigbee.ref = Zigbee

log4j2.logger.com_zsmartsystems_zigbee.name = com.zsmartsystems.zigbee
log4j2.logger.com_zsmartsystems_zigbee.level = INFO
log4j2.logger.com_zsmartsystems_zigbee.additivity = false
log4j2.logger.com_zsmartsystems_zigbee.appenderRefs = Zigbee
log4j2.logger.com_zsmartsystems_zigbee.appenderRef.Zigbee.ref = Zigbee

log4j2.logger.com_zsmartsystems_zigbee_dongle_telegesis_internal_ash.name = com.zsmartsystems.zigbee.dongle.telegesis.internal.ash
log4j2.logger.com_zsmartsystems_zigbee_dongle_telegesis_internal_ash.level = INFO
log4j2.logger.com_zsmartsystems_zigbee_dongle_telegesis_internal_ash.additivity = false
log4j2.logger.com_zsmartsystems_zigbee_dongle_telegesis_internal_ash.appenderRefs = Zigbee
log4j2.logger.com_zsmartsystems_zigbee_dongle_telegesis_internal_ash.appenderRef.Zigbee.ref = Zigbee
```

cdjackson, 2019-11-23:
I wasn’t looking for errors - I wanted to see the full debug log so I can see what is happening.

The logging configuration looks strange - I’m not sure why there are underscores used? Why not configure logging as per the binding docs?

petero-dk, 2019-11-23:
I am logging to a specific zigbee file, the underscores are actually made by openHAB when I configure the logging as per the binding docs (in the karaf console) then it adds the logging configuration with underscores in the name. (When I did was then to move it to its own file)

Any who, this is the full log from a restart to a start:

```
3-Nov-2019 16:15:43.854 [DEBUG] [stems.zigbee.database.ZigBeeNetworkDatabaseManager] - Data store: shutting down.
23-Nov-2019 16:15:43.946 [DEBUG] [rg.openhab.binding.zigbee.handler.ZigBeeSerialPort] - Serial port '/dev/ttyUSB0' closed.
23-Nov-2019 16:15:43.946 [DEBUG] [ee.dongle.telegesis.internal.TelegesisFrameHandler] - TelegesisFrameHandler thread exited.
23-Nov-2019 16:15:43.950 [DEBUG] [ee.dongle.telegesis.internal.TelegesisFrameHandler] - TelegesisFrameHandler closed.
23-Nov-2019 16:15:43.951 [DEBUG] [tems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis] - Telegesis dongle shutdown.
23-Nov-2019 16:15:43.952 [DEBUG] [ystems.zigbee.transaction.ZigBeeTransactionManager] - Shutting down transaction manager
23-Nov-2019 16:15:43.953 [DEBUG] [tsystems.zigbee.transaction.ZigBeeTransactionQueue] - Broadcast: Queue shutdown
23-Nov-2019 16:15:43.954 [DEBUG] [tsystems.zigbee.transaction.ZigBeeTransactionQueue] - Multicast: Queue shutdown
23-Nov-2019 16:15:43.955 [DEBUG] [ab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Initialising ZigBee coordinator
23-Nov-2019 16:15:43.957 [DEBUG] [tsystems.zigbee.transaction.ZigBeeTransactionQueue] - Default: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=2, interTransactionDelay=50, maxRetries=2]
23-Nov-2019 16:15:43.957 [DEBUG] [tsystems.zigbee.transaction.ZigBeeTransactionQueue] - Broadcast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=3, interTransactionDelay=1200, maxRetries=0]
23-Nov-2019 16:15:43.958 [DEBUG] [tsystems.zigbee.transaction.ZigBeeTransactionQueue] - Multicast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=3, interTransactionDelay=1200, maxRetries=0]
23-Nov-2019 16:15:43.959 [DEBUG] [ab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - Mesh Update Period 86400
23-Nov-2019 16:15:43.960 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - Adding supported cluster 1280
23-Nov-2019 16:15:43.961 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNode               ] - ClusterMatcher starting
23-Nov-2019 16:15:43.962 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNode               ] - ClusterMatcher adding cluster 1280
23-Nov-2019 16:15:43.963 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - Adding supported cluster 25
23-Nov-2019 16:15:43.964 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNode               ] - ClusterMatcher adding cluster 25
23-Nov-2019 16:15:43.964 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - ZigBeeNetworkManager initialize: networkState=UNINITIALISED
23-Nov-2019 16:15:43.965 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - Network state is updated to INITIALISING
23-Nov-2019 16:15:43.967 [DEBUG] [ab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - null: networkStateUpdated called with state=INITIALISING
23-Nov-2019 16:15:44.104 [DEBUG] [rg.openhab.binding.zigbee.internal.ZigBeeDataStore] - 00178801045100D4: ZigBee reading network state complete.
23-Nov-2019 16:15:44.106 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeEndpoint           ] - 5299/11: Unsupported cluster 1000 - using ZclCustomCluster
23-Nov-2019 16:15:44.107 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeEndpoint           ] - 5299/11: Unsupported cluster FC01 - using ZclCustomCluster
23-Nov-2019 16:15:44.108 [DEBUG] [stems.zigbee.database.ZigBeeNetworkDatabaseManager] - 00178801045100D4: Data store: Node was restored.
23-Nov-2019 16:15:44.109 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - 00178801045100D4: Updating node NWK=5299
23-Nov-2019 16:15:44.158 [DEBUG] [rg.openhab.binding.zigbee.internal.ZigBeeDataStore] - 001788010450C4ED: ZigBee reading network state complete.
23-Nov-2019 16:15:44.159 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeEndpoint           ] - 3700/11: Unsupported cluster 1000 - using ZclCustomCluster
23-Nov-2019 16:15:44.160 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeEndpoint           ] - 3700/11: Unsupported cluster FC01 - using ZclCustomCluster
23-Nov-2019 16:15:44.161 [DEBUG] [stems.zigbee.database.ZigBeeNetworkDatabaseManager] - 001788010450C4ED: Data store: Node was restored.
23-Nov-2019 16:15:44.162 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - 001788010450C4ED: Updating node NWK=3700
23-Nov-2019 16:15:44.199 [DEBUG] [rg.openhab.binding.zigbee.internal.ZigBeeDataStore] - 0017880106348B0B: ZigBee reading network state complete.
23-Nov-2019 16:15:44.201 [DEBUG] [stems.zigbee.database.ZigBeeNetworkDatabaseManager] - 0017880106348B0B: Data store: Node was restored.
23-Nov-2019 16:15:44.202 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNetworkManager     ] - 0017880106348B0B: Updating node NWK=51173
23-Nov-2019 16:15:44.247 [DEBUG] [rg.openhab.binding.zigbee.internal.ZigBeeDataStore] - 001788010450FDB0: ZigBee reading network state complete.
23-Nov-2019 16:15:44.249 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeEndpoint           ] - 32754/11: Unsupported cluster 1000 - using ZclCustomCluster
23-Nov-2019 16:15:44.250 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeEndpoint           ] - 32754/11: Unsupported cluster FC01 - using ZclCustomCluster
[... cut ...]
23-Nov-2019 16:15:52.688 [DEBUG] [ee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1
23-Nov-2019 16:15:52.689 [DEBUG] [ee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSetRegisterBitCommand [register=14, bit=1, state=false, password=null]
23-Nov-2019 16:15:52.691 [DEBUG] [ee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis Data:ATS0E1=0

23-Nov-2019 16:15:55.692 [DEBUG] [ee.dongle.telegesis.internal.TelegesisFrameHandler] - TELEGESIS Timer: Timeout 1
```

cdjackson, 2019-11-23:
In this file it looks like the baud rate is being set to 19k2 - not 115k2. Are you sure this is configured correctly in PaperUI? The binding isn’t doing anything to change this so it’s down to the configuration.

petero-dk, 2019-11-23:
[image]

Also as I said, I have checked the json db to make sure everything is in order. Before shutdown and after shutdown. When openHAB is stopped the json db says 115200 baud. 

Unless there is some specific other place I should configure this I am very sure I have done it correctly.

cdjackson, 2019-11-23:
Well, there's clearly something you are doing wrong. I've spent the last hour setting up a Telegesis system and it is working just fine!

[image]

```
17:12:09.807 [-thingHandler-1] DEBUG o.o.b.z.handler.ZigBeeSerialPort:131 - Connecting to serial port [/dev/tty.SLAB_USBtoUART] at 115200 baud, flow control FLOWCONTROL_OUT_NONE.
17:12:09.833 [-thingHandler-1] DEBUG o.o.b.z.handler.ZigBeeSerialPort:166 - Serial port [/dev/tty.SLAB_USBtoUART] is initialized.
17:12:09.834 [sisFrameHandler] DEBUG c.z.z.d.t.i.TelegesisFrameHandler:157 - TelegesisFrameHandler thread started
17:12:09.836 [ecutor-thread-1] DEBUG c.z.z.d.t.i.TelegesisFrameHandler:378 - TX Telegesis queue: 1
17:12:09.836 [ecutor-thread-1] DEBUG c.z.z.d.t.i.TelegesisFrameHandler:351 - TX Telegesis: TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null]
17:12:09.836 [ecutor-thread-1] DEBUG c.z.z.d.t.i.TelegesisFrameHandler:362 - TX Telegesis Data:ATS124=1

17:12:09.839 [sisFrameHandler] DEBUG c.z.z.d.t.i.TelegesisFrameHandler:177 - RX Telegesis Data:OK
17:12:09.842 [sisFrameHandler] DEBUG c.z.z.d.t.i.TelegesisFrameHandler:392 - RX Telegesis: TelegesisSetRegisterBitCommand [register=18, bit=4, state=true, password=null, status=SUCCESS]
17:12:09.842 [ecutor-thread-1] DEBUG c.z.z.d.t.i.TelegesisFrameHandler:378 - TX Telegesis queue: 1
17:12:09.842 [ecutor-thread-1] DEBUG c.z.z.d.t.i.TelegesisFrameHandler:351 - TX Telegesis: TelegesisSetRegisterBitCommand [register=14, bit=1, state=false, password=null]
17:12:09.843 [ecutor-thread-1] DEBUG c.z.z.d.t.i.TelegesisFrameHandler:362 - TX Telegesis Data:ATS0E1=0
```

[image]

I think we can close this now.

petero-dk, 2019-11-23:
This also works after a restart?

Regards

Peter

[image]<https://user-images.githubusercontent.com/1041475/69482531-97f87a00-0e14-11ea-9817-8c104648213b.png>

[image]<https://user-images.githubusercontent.com/1041475/69482523-8adb8b00-0e14-11ea-8e82-ea3fc8770616.png>

cdjackson, 2019-11-23:
A restart is not relevant since that doesn't change the binding at all.

petero-dk, 2019-11-23:
Ok then, do you have an idea to how I can enable more logging (to see when the thing configuration is changed?) 

That way I can perhaps figure out what is changing the thing configuration? This is a clean setup of openhabian on a rpi4.

cdjackson, 2019-11-23:
Not really - it must be outside of the binding. The binding doesn’t have any influence over the configuration storage.

I also confirmed this works after a restart for completeness.

petero-dk, 2019-11-23:
Ok, I have a possible cause:

When I enable DEBUG for everything, absolutely everything, the only time there is a notification of telegesis thing configuration update (or anything similar) is in these log lines
```
2019-11-23 19:45:11.797 [DEBUG] [g.discovery.internal.PersistentInbox] - Discovery result with thing 'zigbee:coordinator_telegesis:04000BCC' not added as inbox entry. It is already present as thing in the ThingRegistry.
2019-11-23 19:45:11.799 [DEBUG] [g.discovery.internal.PersistentInbox] - The configuration for thing 'zigbee:coordinator_telegesis:04000BCC' is updated...
```

Now when I look at the source for these debug messages (https://github.com/eclipse/smarthome/blob/b8455de15d65512e8fac4e94d42de6ab2fccf1c5/bundles/config/org.eclipse.smarthome.config.discovery/src/main/java/org/eclipse/smarthome/config/discovery/internal/PersistentInbox.java#L215 and 
https://github.com/eclipse/smarthome/blob/b8455de15d65512e8fac4e94d42de6ab2fccf1c5/bundles/config/org.eclipse.smarthome.config.discovery/src/main/java/org/eclipse/smarthome/config/discovery/internal/PersistentInbox.java#L223) 

My guess to what happens is that the zigbee binding returns a telegesis discovery result with a specific port and baud setting
https://github.com/openhab/org.openhab.binding.zigbee/blob/c768e658b282eea32651b246ecf0856e4daf0f4d/org.openhab.binding.zigbee.telegesis/src/main/java/org/openhab/binding/zigbee/telegesis/internal/discovery/ZigBeeTelegesisUsbSerialDiscoveryParticipant.java#L71-L81

and the persistentinbox will look at that information, determine that there is a discrepancy between the discovered inbox item and the configured thing, and then update the configured thing with the discovered result.

petero-dk, 2019-11-23:
I cannot however figure out why you did not reproduce the issue? Was your dongle autodiscovered?

cdjackson, 2019-11-23:
No - the dongle was not auto discovered.

I’m not really sure how to solve this - I think that this is likely a big in the openHAB core. In the past, it was possible for bindings to get a list of things that had been discovered to allow issues like this to be resolved. In the Z Wave binding I used this to update information in auto discovered things but this is no longer possible as it was removed some time ago.

petero-dk, 2019-11-24:
Could this part of the discovery just be removed?
```java
.withProperty(CONFIGURATION_BAUD, QIVICON_DONGLE_BAUD_RATE)
```
https://github.com/openhab/org.openhab.binding.zigbee/blob/c768e658b282eea32651b246ecf0856e4daf0f4d/org.openhab.binding.zigbee.telegesis/src/main/java/org/openhab/binding/zigbee/telegesis/internal/discovery/ZigBeeTelegesisUsbSerialDiscoveryParticipant.java#L77

Since the baud is set with a default value in the xml file?
https://github.com/openhab/org.openhab.binding.zigbee/blob/dafa9c6bdf92594947880e3fc2dec3592830fd9a/org.openhab.binding.zigbee.telegesis/src/main/resources/ESH-INF/thing/controller_telegesis.xml#L26

petero-dk, 2019-11-24:
The problem is just not at startup, this also happens when doing a discovery for new things (which obviously kills the connection to the dongle and nothing is found)

cdjackson, 2019-11-24:
If you confirm that it works ok without this at the standard 19k2, then I’m happy to remove it.

petero-dk, 2019-11-24:
I am having some issues compiling bindings, but let me see what I can do. Whenever I try to set up a dev environment it only works for a little while and that gets tiresome. Maybe I am not that smart. Give me a couple of days

cdjackson, 2019-11-24:
Ok, thanks - I know the development environment is still not great unfortunately. I’m also having problems with it at the moment as I need to get it to run with updated libraries :(

petero-dk, 2019-11-30:
I got everything building now, but I cannot find the combined jar to upload to add-ons (I have created a whole new openhab system just for this test so I don't mess with my production) 

I have tried to upload just the telegesis part but that fails

cdjackson, 2019-11-30:
I don’t know what you mean by “combined JAR”? The JAR will be in the target folder, but there is no sort of “combined” JAR.

petero-dk, 2019-11-30:
Got it to work, uninstalled the telegesis bundler but that removed everything. So just disabled it and dropped the new one in the addons folder (quite simple actually) 

Commencing testing

petero-dk, 2019-11-30:
Results from removing that part of the code:

Starting with a default dongle with BAUD 19200 it will be autodiscovered with BAUD 19200 and just work. 

Changing the thing settings to BAUD 115200 and changing the dongle to 115200 will also now work. The thing configuration is no longer reset to 19200. So the line is unnecessary :-) 

Would you like a PR or just change it yourself?

cdjackson, 2019-11-30:
Thanks for testing. If you can create a PR it would be appreciated.

petero-dk, 2019-11-30:
I will have that in a little while, in the meanwhile I will be running a longer test on my production system now. I do not expect any issues.

cdjackson, 2019-11-30:
I'l close this based on #524 which removes the reconfiguration of the baud rate when the binding starts.

closed 2019-11-30 by cdjackson

petero-dk, 2019-12-01:
Does the binding reset the dongle when you add it? After I did the testing on the new system (I used my only telegesis dongle the production system (with the original binding not the demo one I created to make the testing) will have the dongle online and all devices, but none are responding to commands and when I try to use them they go offline.

petero-dk, 2019-12-01:
I can see that the PANID and the extended PANID are different from my backups

cdjackson, 2019-12-01:
It shouldn’t unless you tell it to reset.