openhab/org.openhab.binding.zigbee #507: State update when mains power is off

weakfl, 2019-10-13:
When a mains powered device is switched off (eg. a Tradfri bulb), the connected items will keep their last state ad infinitum. For example the level will remain at 100 if that was the last known state.

@cdjackson I'm not sure if the binding should or even could handle this, but it would be nice if there would be some kind of state update after a device has been powered off. Obviously it's impossible to update the state right away, but maybe it's possible at the next polling cycle?

Even setting the item/channel states to `UNDEF` should be enough to be able to handle this within an OH rule.

cdjackson, 2019-10-13:
I’m not sure what the “standard” way to do this is - I don’t think it is recommended to change the state, but I could be wrong.

The binding will set the thing itself OFFLINE, but I don’t think it should go and change the channel states as the system should manage the state once the device is OFFLINE.

weakfl, 2019-10-13:
> The binding will set the thing itself OFFLINE, but I don’t think it should go and change the channel states as the system should manage the state once the device is OFFLINE.

You are right, the thing's state indeed changes to OFFLINE. And that should be enough to update the item states through a thing-based trigger (as the system doesn't change the item states).

Thanks @cdjackson!

closed 2019-10-13 by weakfl

soyxan, 2020-01-02:
An why not update the status of the item to UNDEF, as suggested in this issue #544 ?