openhab/org.openhab.binding.zigbee #523: OTA firmware update via OH console

weakfl, 2019-11-24:
Add functionality to perform OTA firmware updates to the OH console, as discussed [here](https://community.openhab.org/t/is-it-possible-to-upgrade-tradfri-or-hue-hardware-firmware-without-their-hubs/85970/21).

cdjackson, 2020-01-02:
I'm in the process of creating the full system to work with OH. See the firmware provider here -:

https://github.com/zsmartsystems/com.zsmartsystems.openhab.firmwareprovider

cdjackson, 2020-01-02:
And the OTA CLI command enhancements are here -:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/945

weakfl, 2020-01-02:
Very nice, looking forward to this.

@cdjackson once the improvements are done, could you maybe also provide a new version of the console jar on your website?

cdjackson, 2020-01-02:
I can, but note that this should not be needed. It will be possible to use the Karaf console instead for most commands. (So long as the appropriate bundles are installed).

OTA will be available through PaperUI.

weakfl, 2020-01-02:
I see, perfect. Even easier to use then.

cdjackson, 2020-01-02:
Yes - that’s the plan. I’ve written this so that I can easily distribute firmware files and they should just need to be dropped in a folder :)

cdjackson, 2020-01-19:
The OTA should be fixed with https://github.com/openhab/org.openhab.binding.zigbee/pull/553.

This will allow you to use the console with a command such as -:

```
zigbee otaupgrade start 32790/3 firmware/OTA_CLA60_RGBW_MasterKey_01020307.ota
```

I also have a firmware provider written which we can discuss if we want to make this available through PaperUI.

CodeFinder2, 2020-03-02:
Sorry to ask a rather semi-related question but it seems that OpenHAB already supports OTA updates for Bitron zigbee smart plugs. So, would you mind telling me where I can find / download the firmware files? Already searched the internet and this repo but without success...

Thanks!

cdjackson, 2020-03-02:
Sorry - I’ve no idea. I would suggest you contact the manufacturer. Often they do not make them available.

CodeFinder2, 2020-03-02:
Hm, I thought openhab is downloading them when an OTA update is triggered? 🙈

cdjackson, 2020-03-02:
No, it’s not.

cdjackson, 2020-03-29:
Closing this as this is implemented.

closed 2020-03-29 by cdjackson