openhab/org.openhab.binding.zigbee #525: Add support for Develco Products External Meter Interface

petero-dk, 2019-12-05:
The device connects but creates no channels.

I would have expected it to be very similar to the #164 but when connected it just shows "unkown device" and "OFFLINE Node has not completed discovery"

Technical manual:
https://www.develcoproducts.com/media/1691/external-meter-interface-technical-manual.pdf

Installation Manual:
https://www.develcoproducts.com/media/3815/emi-installation-manual-v15-t00000042.pdf

I received the following error when the device joined:

```
2019-12-05 21:24:43.652 [WARN ] [mmon.WrappedScheduledExecutorService] - Scheduled runnable ended with an exception: 
java.util.NoSuchElementException: null
	at java.util.HashMap$HashIterator.nextNode(HashMap.java:1447) ~[?:1.8.0_222]
	at java.util.HashMap$KeyIterator.next(HashMap.java:1469) ~[?:1.8.0_222]
	at java.util.Collections.min(Collections.java:598) ~[?:1.8.0_222]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.getExpectedUpdatePeriod(ZigBeeThingHandler.java:461) ~[?:?]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.doNodeInitialisation(ZigBeeThingHandler.java:428) ~[?:?]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.access$0(ZigBeeThingHandler.java:238) ~[?:?]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$1.call(ZigBeeThingHandler.java:232) ~[?:?]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$1.call(ZigBeeThingHandler.java:1) ~[?:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) ~[?:1.8.0_222]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) ~[?:1.8.0_222]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) ~[?:1.8.0_222]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:1.8.0_222]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:1.8.0_222]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_222]
```
Unfortunately I did not have debug logs enabled at that time, but I can redo the process if necessary? 

Additionally the device reports the following properties:
modelId	ZHEMI101
vendor	Develco
zigbee_device_initialised	true
zigbee_devices	[]
zigbee_lastupdate	
zigbee_logicaltype	END_DEVICE
zigbee_manufacturercode	0x0000
zigbee_neighbors	[]
zigbee_networkaddress	18218
zigbee_powerlevel	CRITICAL
zigbee_powermode	RECEIVER_ON_IDLE
zigbee_powersource	UNKNOWN
zigbee_powersources	[]
zigbee_routes	[]
zigbee_stkcompliance	0
zigbee_zclversion	3

What should I do to assist?

cdjackson, 2019-12-05:
I can’t really comment without debug logs etc. Please read the binding documentation section about what to do when things don’t go right.

TheDiverM, 2019-12-20:
Hi.
The develco comes up but both voltage and amp has value of 'maxint' 65535.

TheDiverM, 2020-01-12:
Is there anybody looking at this or?

cdjackson, 2021-08-09:
Closing this as it's quite old, and there have been a lot of updates since. If the problem persists, then please provide a debug log showing the issue to allow it to be debugged (as requested earlier).

closed 2021-08-09 by cdjackson

petero-dk, 2022-01-28:
I will just be adding some more information to this issue including a debug log:

Firstly the status of the device is that I can add it to openHAB now, it connects and states as being ONLINE. However some weirdness happens.

First it does not continue to send updates, per the manual it should send updates every 5 seconds, and once attached it will do that for a little while, but then just stop (still being online)

Also the numbers seem to always be going down, in this case
[image]
it started at around 400 in demand but then just slowly went down (never up)

When I try to reinitialize the device I get the following error: 

```
java.lang.ClassCastException: class java.lang.Integer cannot be cast to class java.lang.Long (java.lang.Integer and java.lang.Long are in module java.base of loader 'bootstrap')
	at com.zsmartsystems.zigbee.serialization.DefaultSerializer.appendZigBeeType(DefaultSerializer.java:209) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.field.AttributeReportingConfigurationRecord.serialize(AttributeReportingConfigurationRecord.java:361) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.ZclFieldSerializer.serialize(ZclFieldSerializer.java:38) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.clusters.general.ConfigureReportingCommand.serialize(ConfigureReportingCommand.java:107) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendCommand(ZigBeeNetworkManager.java:905) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.send(ZigBeeTransactionManager.java:443) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendNextTransaction(ZigBeeTransactionManager.java:681) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.queueTransaction(ZigBeeTransactionManager.java:372) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendTransaction(ZigBeeTransactionManager.java:351) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendTransaction(ZigBeeNetworkManager.java:2030) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.ZigBeeNode.sendTransaction(ZigBeeNode.java:926) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.ZigBeeEndpoint.sendTransaction(ZigBeeEndpoint.java:596) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.ZclCluster.sendCommand(ZclCluster.java:294) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.ZclCluster.sendCommand(ZclCluster.java:305) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.ZclCluster.setReporting(ZclCluster.java:1767) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.ZclCluster.setReporting(ZclCluster.java:589) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.ZclAttribute.setReporting(ZclAttribute.java:431) ~[bundleFile:?]
	at org.openhab.binding.zigbee.internal.converter.ZigBeeConverterMeteringSummationReceived.initializeDevice(ZigBeeConverterMeteringSummationReceived.java:81) ~[bundleFile:?]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.initializeDevice(ZigBeeThingHandler.java:514) ~[bundleFile:?]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.doNodeInitialisation(ZigBeeThingHandler.java:377) [bundleFile:?]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$1.call(ZigBeeThingHandler.java:227) [bundleFile:?]
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler$1.call(ZigBeeThingHandler.java:1) [bundleFile:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) [?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
	at java.lang.Thread.run(Thread.java:829) [?:?]
```

A full debug log will come ASAP, as I just redid the entire system, but for now here is the detected file.
[0015BC001B021B28.xml.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/7960257/0015BC001B021B28.xml.txt)
.

cdjackson, 2022-01-28:
As explained above, without a debug log I've no way to know what caused this exception, so please provide this once you have it.

petero-dk, 2022-01-28:
[zigbee.log](https://github.com/openhab/org.openhab.binding.zigbee/files/7961358/zigbee.log)

During this debug log I did two things (for that device)

I ran the zigbee fingerprint command in karaf

and I did a initialize=true from the webui

cdjackson, 2022-01-28:
What version of the binding are you using? I'd suggest to ensure that you have the latest as I think this exception is already resolved with https://github.com/openhab/org.openhab.binding.zigbee/pull/732.

I'm not sure that hte numbers going down is a problem with the binding. We use this channel for a commercial (and zigbee alliance certified) smart meter product and it receives the data file. However again I would need to see a debug log showing this issue - the log you provide here always reports 0. I would also say though that for something to count down - it must have previously gone up ;) . The binding will however only report what the device sends.

petero-dk, 2022-01-28:
I run unstable - latest main

cdjackson, 2022-01-28:
Ok, it's the same error in a different channel. #734 should fix this.

petero-dk, 2022-01-28:
Cool, can I do anything to assist more? How can I test this PR?

cdjackson, 2022-01-28:
This should be available in the next snapshot which I guess will be generated tonight EU time - so if you grab the next update in the morning, hopefully it will be included.

petero-dk, 2022-01-28:
Thanks, have a wonderful weekend.

cdjackson, 2022-01-28:
Thanks - you also. Let me know if this is resolved :)

petero-dk, 2022-01-29:
Ok, the error no longer shows up so it seems to fix that bug. The steady decline of instantaneous demand still shows up.

First it reports 0, then a value, and then all subsequent values are lower and lower until it stops, t

his is definitly now how my power usage is:
[image]

Log files:
[zigbee7.log](https://github.com/openhab/org.openhab.binding.zigbee/files/7963327/zigbee7.log)
[zigbee.log](https://github.com/openhab/org.openhab.binding.zigbee/files/7963323/zigbee.log)

cdjackson, 2022-01-29:
It looks like this is correct - as reported by the device.

petero-dk, 2022-01-29:
hmm, maybe there is a configuration error on the device it self, this type of device has a menu that can be controlled via a button. Or perhaps the sensor is wrongly configured. It has a bunch of settings according to the manual. :-( 

Thanks for fixing that error, but I do not think this device is working.