openhab/org.openhab.binding.zigbee #531: Reminder for newer Z-Smart systems versions

triller-telekom, 2019-12-20:
In the `ZigBeeCoordinatorHandler` the order of initializing the `ZigbeeNetworkManager` and adding extensions to it has to be changed, because is is now checked that the network is in `INITIALIZING` state before adding extensions:

https://github.com/openhab/org.openhab.binding.zigbee/blob/9d4c1ac047d26abac5c4a06fc8dd2da1ae1f421f/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/handler/ZigBeeCoordinatorHandler.java#L407-L440

The same applies if we want to add all supported clusters to the network manager, this has to be done AFTER initializing the network manager as well.

cdjackson, 2019-12-20:
Thanks Stefan,
Yes, this is understood and I will mop this up over Christmas when I migrate to the new ZSS libraries. 

Have a good Christmas.

Cheers
Chris

cdjackson, 2020-01-03:
Closed by #517

closed 2020-01-03 by cdjackson