openhab/org.openhab.binding.zigbee #544: Show bulb as "off" when the linked Thing is offline

soyxan, 2020-01-02:
Hi,

I am using this binding in openhab 2.5.0 and I wanted to show the Zigbee bulbs as off when they are switched off from the wall switch (some time after the power cut to the bulb the Thing will appear as offline)

```
rule "ZB_Sofa"
when
	Thing "zigbee:device:4b531422:7cb03eaa0a05ce9b:7CB03EAA0A05CE9B_3_dimmer" changed to OFFLINE
then
	ZB_Sofa.postUpdate(OFF)
end
```

Do you think that this should be implemented in the binding? Other bindings (for instance Samsung TV), show the item as "OFF" when the TV is offline.

Thanks for your great work!

weakfl, 2020-01-02:
Already answered in [507](https://github.com/openhab/org.openhab.binding.zigbee/issues/507).

closed 2020-01-02 by soyxan

cdjackson, 2020-01-02:
> show the item as "OFF" when the TV is offline.

How do I know it is off? It is OFFLINE, so that means we cannot communicate with the device.

soyxan, 2020-01-02:
> > show the item as "OFF" when the TV is offline.
> 
> How do I know it is off? It is OFFLINE, so that means we cannot communicate with the device.

Yes it is an assumption, but at least is a way to know it. For bulbs I think is more interesting to know it, so I will keep using the Rule I posted before.

cdjackson, 2020-01-02:
I would argue that if this is considered the correct implementation, then this should be handled by the core so that it is consistent for all bindings/things. The core knows when a device is OFFLINE, so it could set the state to OFF, or probably more appropriate would be UNDEF to show that we just don't know.

Maybe it's worth opening such a request in the openhab-core repository.

soyxan, 2020-01-02:
Yes I agree with you, in fact I have modifies my rule this way:

rule "ZB_Sofa"
when
	Thing "zigbee:device:4b531422:7cb03eaa0a05ce9b:7CB03EAA0A05CE9B_3_dimmer" changed to OFFLINE
then
	ZB_Sofa.postUpdate(UNDEF)
end

Regarding the open of the request, I will do it, but why in the openhab-core repository intead of in this one (noob question)?

cdjackson, 2020-01-02:
This repository is for ZigBee - so you are right, I can fix it here. However, if we want consistent behaviour for all bindings, then it should be done in the core. Then all bindings will work the same - otherwise we have to update all 200+ bindings to do this, and we will no doubt have different bindings doing different things which makes the system messy from a user perspective when things are not consistent.

(Does that make sense?)