openhab/org.openhab.binding.zigbee #554: [Eurotronic Spirit Zigbee] Add support for further channels

iot-openhab, 2020-01-22:
Possible Cluster "Thermostat" Channels can be found in
https://eurotronic.org/wp-content/uploads/2019/11/Spirit_ZigBee_BAL_web_DE_Okt.-2019.pdf
-> Page 16

The Cluster provides further channels:

- TRV Mode (Attibute ID 0x4000)
- Set Valve Position (Attribute ID 0x4001)
- Current Temperature Set point (Attribute ID 0x4003)

Also the Power configuration Cluster is intresting
- Battery Percentage Remaining (Attribute ID 0x0021)

Requested from https://community.openhab.org/t/cant-connect-eurotronic-spirit-zigbee-to-oh-using-bitronvideo-coordinator/88244/39

nailyk-fr, 2020-11-05:
I am new with openhab2 but it is also something I would like to have working. 
If someone can, at least, give me some hints where to start, I would gladly try to make some PR :smiley:

nailyk-fr, 2020-11-13:
Hello ! 

If someone could point me to an already existing channel I would gladely appreciate it as an entry point. 
From the search I made, the only issue is those eTRV are missing the channel 0x00FF to be able to work. 
I am far to be a good developper but I would be able to use an existing one as a 'source' and copy it to make it work. 

And, as it would be submitted through pull-request, you would be able to reject until it is matching your requirements :wink:

cdjackson, 2021-11-11:
Battery Percent remaining is supported.

The above link no longer works so pasting in the table from the manual.

[image]

rs22, 2022-10-07:
I'm looking for a way how to turn off my Eurotronic Spirit Zigbee when an open window is detected in OpenHAB. The device does not react to setting the System Mode channel to 'off' (attribute id 0x001C).

So it looks like this can only be done using the vendor-specific TRV mode field (0x4000).
Is this correct? 

Does that mean a change is required here?
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.autocode/src/main/resources/0201_Thermostat.xml

For reference, here's a link to the now offline product documentation:
http://web.archive.org/web/20191127152422/https://eurotronic.org/wp-content/uploads/2019/11/Spirit_ZigBee_BAL_web_DE_Okt.-2019.pdf

rs22, 2022-10-08:
My thermostats (tested with firmware versions 20190408 and 20191014) can only be put into 'off' mode by using the host flags (0x4008), as described here: https://www.zigbee2mqtt.io/devices/SPZB0001.html

- writing 32 -> thermostat shows 'off'
- writing 16 -> thermostat recalls occupied heating setpoint

Changing the System Mode (0x001C) has no effect. So far, I have tested with de deCONZ software and would like to control the thermostats directly using the OpenHAB binding.

How can I add the necessary channels to the binding?

cdjackson, 2022-10-10:
You'd need to create a device definition. If you look in the code, you will find some XML files which define the devices, and in these it's possible to add custom attributes. Hopefully that will allow you to create what you are looking for - if it works ok, then please provide a PR into the repository so others also get to use this.