openhab/org.openhab.binding.zigbee #565: Adding enhanced support for Xiaomi/Aqara Curtain Motor

robotronnn, 2020-03-31:
So far it's only possible to toggle the motor to either close or open the curtain.
Would be lovely if you could integrate the missing channels.
Thanks for the binding!

[image]

cdjackson, 2020-08-11:
> Would be lovely if you could integrate the missing channels.

What channels are missing though? I don't have the device, so I've no idea what this supports. If you are able to add the functions, then please do. If you want to send me a device then I will take a look.

cdjackson, 2021-04-25:
Closing this. If there is further information available describing what is required, then please provide as requested earlier and re-open.

closed 2021-04-25 by cdjackson