openhab/org.openhab.binding.zigbee #573: Change value calculation of Xiaomi Aqara Motion Sensor illuminance channel

guy-inkognito, 2020-05-11:
### Summary

I propose to change the value calculation for the Aqara Motion Sensor illuminance channel because I think it's wrong. I believe it adapts the calc from the Hue Motion Sensor specification which requires to apply a logarithmic scale. This tranformation shall not be used for the Aqara device.

### Details

I've played around with the Aqara devices for a while now and have also seen the very small values for the illuminance channel. People in the community suggest to just multiply it by 1000 to get the lux value. I believe that's wrong. You rather need to revert the transformation done [here](https://github.com/openhab/org.openhab.binding.zigbee/blob/f746d4686138acb51fd3cd12c92e72422bc05bbf/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/converter/ZigBeeConverterIlluminance.java#L175) with:

`var lux = 10000 * log10(illumination+1);`

The reported values, after applying the reverse formula, always transform nicely into full numbers without any decimal places. I currently do this in a custom JS transformation. However, I believe this should be included in the binding directly.

### Misc

That's my first suggestion and please feel free to correct me if I'm doing something wrong. I'm thankful for all guidance. Thx to the community and all the contributors, you're great and I'm really enjoying OpenHAB!

cdjackson, 2020-05-11:
The binding uses the standard calculation for luminance - as per the ZigBee standards. If the device isn't following the standard then that makes things a bit problematic and I'd suggest to handle this through a transformation rule.

[image]

guy-inkognito, 2020-05-11:
I see. If it's handled incorrectly by the device itself, that's unfortunate indeed. Is there an option to maybe put these special behaviors into the documentation then?

I'll put it into the community forum to have it archived, though.

Thx and cheers!

cdjackson, 2020-05-11:
> Is there an option to maybe put these special behaviors into the documentation then?

What do you mean? Do you mean can we document that the illuminance is different than standard? I guess this could go into the readme in the supported devices - we could add a note that something is non-standard.

guy-inkognito, 2020-05-11:
> I guess this could go into the readme in the supported devices - we could add a note that something is non-standard.

That's exactly what I have in my mind. I'll do a proposal.

cdjackson, 2020-05-11:
Perfect - thanks.

guy-inkognito, 2020-05-11:
For reference: https://community.openhab.org/t/correctly-calculate-lux-values-at-xiaomi-aqara-motion-sensors-illuminance-channel/98603

cdjackson, 2020-05-11:
I just wanted to clarify before merging your PR. You state -:

<a name="note3"></a> *Note 3: The illuminance channel value is being reported incorrectly. Transform by 10000\*log10(measuredValue+1) to get the correct illuminance in lux.*

 Currently the formula that is applied is this formula -:

```
            updateChannelState(new DecimalType(Math.pow(10.0, (Integer) val / 10000.0) - 1));
```

So it seems that the Aqara device is reversing this, and therefore the Aqara is reporting directly in lux.

For my understanding, is that correct?

Thanks.

guy-inkognito, 2020-05-12:
That is exactly correct. Thanks!

cdjackson, 2020-05-12:
Great - thanks. Just wanted to ensure I had the right understanding and there wasn't a bug somewhere :)

I'll close this now and merge the other PR - thanks.

closed 2020-05-12 by cdjackson

guy-inkognito, 2020-05-12:
Thanks for the nice and quick interaction. Really appreciate your work!